betaflight/betaflight #12402: latest master have wrong target config

TheIsotopes, 2023-02-21:
### Describe the bug

since #12353 with cloud build the fc have wrong config setup

### To Reproduce

use latest master and check `dump` output with on older `dump` output from same target

here a sreenshot: (on right latest master, on left `dump` output before `#12353` merged)
[image]

### Expected behavior

correct target config

### Support ID

```txt
Id: "e805d4ee-f7c8-4c84-bf69-7d9c05d26f5f"
```

### Flight controller

MATEKF722MINI

TheIsotopes, 2023-02-21:
here the `dump` output before `#12353` was merged:

```
# dump

# version
# Betaflight / STM32F7X2 (S7X2) 4.5.0 Feb 21 2023 / 17:12:20 (a7e4bf41e) MSP API: 1.45

# config: YES

# start the command batch
batch start

board_name MATEKF722MINI
manufacturer_id MTKS

# name: -

# resources
resource BEEPER 1 C13
resource MOTOR 1 B04
resource MOTOR 2 B05
resource MOTOR 3 B00
resource MOTOR 4 B01
resource MOTOR 5 A15
resource MOTOR 6 B03
resource MOTOR 7 B06
resource MOTOR 8 B07
resource LED_STRIP 1 A08
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 A01
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 C07
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 NONE
resource SERIAL_RX 12 NONE
resource I2C_SCL 1 B08
resource I2C_SCL 2 B10
resource I2C_SCL 3 A08
resource I2C_SCL 4 NONE
resource I2C_SDA 1 B09
resource I2C_SDA 2 B11
resource I2C_SDA 3 B04
resource I2C_SDA 4 NONE
resource LED 1 A14
resource LED 2 A13
resource LED 3 NONE
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 3 C10
resource SPI_SCK 4 NONE
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
[... 1735 lines cut ...]
set thr_expo = 0
set rates_type = ACTUAL
set quickrates_rc_expo = OFF
set roll_rc_rate = 7
set pitch_rc_rate = 7
set yaw_rc_rate = 7
set roll_expo = 0
set pitch_expo = 0
set yaw_expo = 0
set roll_srate = 67
set pitch_srate = 67
set yaw_srate = 67
set throttle_limit_type = OFF
set throttle_limit_percent = 100
set roll_rate_limit = 1998
set pitch_rate_limit = 1998
set yaw_rate_limit = 1998
set roll_level_expo = 0
set pitch_level_expo = 0

# end the command batch
batch end

# 
```

haslinghuis, 2023-02-21:
Resource LED seems to have a one-of-bug
Resource ADC is not assigned ???
Resource OSD has been removed as it's not added as build option

https://build.betaflight.com/api/builds/c758922f29a292ded877ce308d5d89f2/json

haslinghuis, 2023-02-22:
@TheIsotopes try flashing with commit `#12397` as custom defaults are not applied on current master.

TheIsotopes, 2023-02-22:
@haslinghuis did you also compare the two dumps i added? there are seven other differences.

I also noticed that the request to load the custom defaults no longer appears in the configurator.

TheIsotopes, 2023-02-22:
> try flashing with commit #12397 as custom defaults are not applied on current master.

I will test later.

TheIsotopes, 2023-02-22:
> try flashing with commit `#12397` as custom defaults are not applied on current master.

seems not to work ... using `make MATEKF722MINI EXTRA_FLAGS="-DCLOUD_BUILD -DUSE_DSHOT -DUSE_FLASH -DUSE_GPS -DUSE_GPS_PLUS_CODES -DUSE_LED_STRIP -DUSE_PINIO -DUSE_SERIALRX -DUSE_SERIALRX_CRSF -DUSE_SERIALRX_DEFAULT -DUSE_SERIALRX_GHST -DUSE_SERIALRX_SBUS -DUSE_TELEMETRY -DUSE_TELEMETRY_CRSF -DUSE_TELEMETRY_GHST -DUSE_TELEMETRY_SMARTPORT -DUSE_VTX"`

TheIsotopes, 2023-02-22:
@haslinghuis I've now tested it on a f405, everything is ok there.

Hmm, seems only affect on F7x2 / F7xx.

haslinghuis, 2023-02-22:
F411SX1280, MAMBAG4, KAKUTEH7MINI (H743) are affected as well.

So only F405 (MATEKF405TE) is ok, don't have a normal F411

blckmn, 2023-02-22:
https://github.com/betaflight/betaflight/pull/12405

TheIsotopes, 2023-02-23:
@blckmn @haslinghuis tested again latest master with merged `#12405` 

no request to load custom defaults appears :-(
Resource LED, ADC and OSD now ok.

blckmn, 2023-02-23:
> @blckmn @haslinghuis tested again latest master with merged `#12405` 
> 
> 
> 
> no request to load custom defaults appears :-(
> 
> Resource LED, ADC and OSD now ok.

#12410 fixes that issue.

TheIsotopes, 2023-02-23:
> #12410 fixes that issue.

when will it be assembled to test this.

haslinghuis, 2023-02-23:
@TheIsotopes did some testing, see comment in PR. Add the PR to commit field like so (expert mode enabled):
[image]

haslinghuis, 2023-02-23:
Tested `#12410` okay on F72X and MATEKF405TE.

There seems be another problem (outside this PR):

- BETAFPVF4SX1280 no connection after applying custom defaults (same on master when doing defaults in CLI)
- MATEKH743 no connection after flashing. Only way to recover is DFU (same on master)

Looking at the order of PR's and further testing with MATEKH743 seems there is some configuration issue which does not seem to affect F72X.

Tested with MATEKH743 no communication after flashing master (all leds are steady on):

- FIX: LED is zero based index (variance to resource command), ADC pins incorrectly mapped. (#12405)
- Adding VMA address to config.h (#12386)
- Enable cloud build features at runtime (#12388) 
- REFACTOR: moving timer definitions to MCU locations. (#12397) - Same as below.
- FIX: Multi-gyro enabled but not configured at compile time. (#12404) 
- config.h patch from unified-targets 22-FEB-2023 (#12403) - BUILDS OKAY, no communications from there.
- Auto generation of config.h from unified target repo (#12353) - DOES NOT BUILD
- Add CMS support on TX for ELRS (#12308) - TESTED OKAY

blckmn, 2023-02-23:
All LEDs on is a hard fault. Will investigate.

haslinghuis, 2023-02-25:
#12414 and #12415 solves most issues except MATEKH743 and baro recognition.

haslinghuis, 2023-02-25:
MATEKH743.
Looking at the last 3 PR's in the list above trying to find where it fails but I don't get it. As #12308 works, #12353 does not build and #12403 only updates more config files (but not for MatekH743) - so it should not build either as #12353 was failing.
Using order from nightly builds.

TheIsotopes, 2023-03-04:
The error that caused the issue to be opened has been fixed and MATEKH743 is working again, so I'm closing here.

closed 2023-03-04 by TheIsotopes