betaflight/betaflight #12409: 4.3+ unstable altitude in ACRO

Sekilsgs2, 2023-02-23:
### Describe the bug

Since first 4.3 rc1 released im see that copter in ACRO began to hold the altitude badly - you have to constantly work with throttle and there is no middle - there is either a lot of throttle or low - it's not very pleasant to fly. On 4.2.11 there is no such thing - it flies very clearly. adjusting pid and filters does not help. does anyone have the same problems? three of us in the group also noticed this. 
P.s. im test this on 3 different copters. 7, 6 and 5 inch. 
4.4 have same problem.

### To Reproduce

Simple install bf 4.3+ and compare altitude stability with 4.2.11

### Expected behavior

correct behavior when you do not have to yank the throttle so often to equalize the height.

### Support ID

```txt
Later ;)
```

### Flight controller

Mamba mk4 h743

haslinghuis, 2023-02-23:
Please check https://github.com/betaflight/betaflight/pull/12231

Sekilsgs2, 2023-02-24:
> Please check #12231

Im say about acro mode but your link about angle and horizon?

NoahWL, 2023-03-10:
I've noticed something similar with my BetaFPV Meteor85 (STM32F411SX1280) running 4.3.1.  I have not flown much in angle mode but have observed this in acro/default mode.  Finding the throttle position to hold a hover is challenging, yes, but entirely possible.

However, once that point is held for a few seconds, without any change in TX inputs, the quad behaves as if I lowered the throttle.  I can clearly hear the motors drop slightly in RPM.  If I then raise the throttle, even just barely, the quad quickly climbs back up and I must lower the throttle again to re-hold a hover.  It's very strange.  This is in a fairly open space indoors so wind is not an issue.

I'll be installing an OpenLager tonight if anyone has recommendations for what to log.  I am not new to quads but am new to BetaFlight and customizable FCs in general.

> Please check #12231

Are you suggesting we flash that commit and re-test?  Otherwise, I'm not sure how it is relevant to this issue.  There is no info there regarding changes to behavior in acro mode.

Sekilsgs2, 2023-03-10:
> I've noticed something similar with my BetaFPV Meteor85 (STM32F411SX1280) running 4.3.1. I have not flown much in angle mode but have observed this in acro/default mode. Finding the throttle [...]

What about 4.2.11 - don't have this problem?

limonspb, 2023-03-11:
> ### Describe the bug
> Since first 4.3 rc1 released im see that copter in ACRO began to hold the altitude badly - you have to constantly work with throttle and there is no middle - there is either a lot of throttle or low - it's not very pleasant to fly. On 4.2.11 there is no such thing - it flies very clearly. adjusting pid and filters does not help. does anyone have the same problems? three of us in the group also noticed this. P.s. im test this on 3 different copters. 7, 6 and 5 inch. 4.4 have same problem.
> 
> ### To Reproduce
> Simple install bf 4.3+ and compare altitude stability with 4.2.11
> 
> ### Expected behavior
> correct behavior when you do not have to yank the throttle so often to equalize the height.
> 
> ### Support ID
> ```
> Later ;)
> ```
> 
> ### Flight controller
> Mamba mk4 h743
> 
> ### Other components
> _No response_
> 
> ### How are the different components wired up (including port information)
> _No response_
> 
> ### Add any other context about the problem that you think might be relevant here
> _No response_

Could you plz flash 4.2, record blackbox how it hovers nice and smooth. Then flash 4.3, record blackbox how it doesn't hover good. And share both. Thank you.
This will allow to exclude any user changed in gear / settings, and allow to see what's actually happening.

Sekilsgs2, 2023-03-11:
> > ### Describe the bug
> > Since first 4.3 rc1 released im see that copter in ACRO began to hold the altitude badly - you have to constantly work with throttle and there is no middle - there is either a lot of throttle or low - it's not very pleasant to fly. On 4.2.11 there is no such thing - it flies very clearly. adjusting pid and filters does not help. does anyone have the same problems? three of us in the group also noticed this. P.s. im test this on 3 different copters. 7, 6 and 5 inch. 4.4 have same problem.
> > ### To Reproduce
> > Simple install bf 4.3+ and compare altitude stability with 4.2.11
> > ### Expected behavior
> > correct behavior when you do not have to yank the throttle so often to equalize the height.
> > ### Support ID
> > ```
> > Later ;)
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > ### Flight controller
> > Mamba mk4 h743
> > ### Other components
> > _No response_
> > ### How are the different components wired up (including port information)
> > _No response_
> > ### Add any other context about the problem that you think might be relevant here
> > _No response_
> 
> Could you plz flash 4.2, record blackbox how it hovers nice and smooth. Then flash 4.3, record blackbox how it doesn't hover good. And share both. Thank you. This will allow to double check the claim, exclude any user changed in gear / settings, and allow to see what's actually happening.

Oh.. maybe later - i'm rebuild my copter and wait some parts. 
in my opinion - it's hard to track - and as if there is not enough expo to gas - adding this expo seems to make it a little better - but it's still not like in 4.2.11. Maybe this is a decrease in filtration, added in 4.3+ at low throttle - and because of the noise, the flight controller does not understand much what to do with it .. my friend says he has no difference between what is at 4.2 and what is at 4.4. And the other two say there is such a thing .. maybe this is just due to different vibrations on different frames / propellers / motors ..
But on 4.2.11 it flies great and I don’t even want to go higher.

limonspb, 2023-03-11:
> Oh.. maybe later - i'm rebuild my copter and wait some parts. in my opinion - it's hard to track - and as if there is not enough expo to gas - adding this expo seems to make it a little better - but [...]

If all you saying is true it would be super easy to check. Just blackbox