betaflight/betaflight #12412: FC getting unresponsive after adjusting filters

dweight, 2023-02-24:
### Describe the bug

This issue happened to me twice already. I'm in the process of tuning my quad so I'm messing with the filters to get the best possible combination. After couple of flights with logging changing any filter (using OSD or configurator) is making FC unresponsive (only DFU mode flash is helping). The first time it happened when changing gyro lp 2 freq (using OSD), the second time after changing max dynamic notch cutoff. 

### To Reproduce

Using Foxeer F722 v4 try to change filters couple of times (with saving). It eventually will make the FC hang.

### Expected behavior

Changing filters should not hang the FC :)

### Support ID

```txt
version
# Betaflight / STM32F7X2 (S7X2) 4.4.0 Feb 11 2023 / 14:09:02 (4605309d8) MSP API: 1.45

# config: YES
# board: manufacturer_id: FOXE, board_name: FOXEERF722V4

# status
MCU F722 Clock=216MHz, Vref=3.26V, Core temp=38degC
Stack size: 2048, Stack address: 0x20010000
Configuration: CONFIGURED, size: 3330, max available: 16384
Devices detected: SPI:1, I2C:0
Gyros detected: gyro 1 locked dma
GYRO=ICM42688P, ACC=ICM42688P
OSD: MAX7456 (30 x 13)
BUILD KEY: 17d4116d566dd7a839454e113a1cb110 (4.4.0)
System Uptime: 80 seconds, Current Time: 2023-02-24T14:22:27.707+00:00
CPU:46%, cycle time: 127, GYRO rate: 7874, RX rate: 15, System rate: 9
Voltage: 1 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
FLASH: JEDEC ID=0x00ef4018 16M
Arming disable flags: RXLOSS CLI MSP RPMFILTER

# flash_info
Flash sectors=256, sectorSize=65536, pagesPerSector=256, pageSize=256, totalSize=16777216 JEDEC ID=0x00ef4018
Partitions:
  0: FLASHFS   0 255
FlashFS size=16777216, usedSize=0

# dump hardware

# version
# Betaflight / STM32F7X2 (S7X2) 4.4.0 Feb 11 2023 / 14:09:02 (4605309d8) MSP API: 1.45

# config: YES

# start the command batch
batch start

board_name FOXEERF722V4
manufacturer_id FOXE

# resources
resource BEEPER 1 A04
resource MOTOR 1 A09
resource MOTOR 2 A08
resource MOTOR 3 C09
resource MOTOR 4 C08
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERIAL_TX 1 B06
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 C12
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 B07
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 A01
resource SERIAL_RX 5 D02
resource SERIAL_RX 6 C07
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 NONE
resource SERIAL_RX 12 NONE
resource I2C_SCL 1 B08
resource I2C_SCL 2 NONE
resource I2C_SCL 3 NONE
resource I2C_SCL 4 NONE
resource I2C_SDA 1 B09
resource I2C_SDA 2 NONE
resource I2C_SDA 3 NONE
resource I2C_SDA 4 NONE
resource LED 1 C15
resource LED 2 NONE
resource LED 3 NONE
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 3 C10
resource SPI_SCK 4 NONE
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MISO 3 C11
resource SPI_MISO 4 NONE
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 B15
resource SPI_MOSI 3 B05
resource SPI_MOSI 4 NONE
resource CAMERA_CONTROL 1 B03
resource ADC_BATT 1 C00
resource ADC_RSSI 1 A00
resource ADC_CURR 1 C02
resource ADC_EXT 1 NONE
resource PINIO 1 NONE
resource PINIO 2 NONE
resource PINIO 3 NONE
resource PINIO 4 NONE
resource USB_MSC_PIN 1 NONE
resource FLASH_CS 1 B12
resource OSD_CS 1 C03
resource GYRO_EXTI 1 C04
resource GYRO_EXTI 2 NONE
resource GYRO_CS 1 B02
resource GYRO_CS 2 NONE
resource USB_DETECT 1 NONE
resource PULLUP 1 NONE
resource PULLUP 2 NONE
resource PULLUP 3 NONE
resource PULLUP 4 NONE
resource PULLDOWN 1 NONE
resource PULLDOWN 2 NONE
resource PULLDOWN 3 NONE
resource PULLDOWN 4 NONE

# timer
timer B07 AF2
# pin B07: TIM4 CH2 (AF2)
timer A09 AF1
# pin A09: TIM1 CH2 (AF1)
timer A08 AF1
# pin A08: TIM1 CH1 (AF1)
timer C09 AF3
# pin C09: TIM8 CH4 (AF3)
timer C08 AF3
# pin C08: TIM8 CH3 (AF3)
timer C06 AF3
# pin C06: TIM8 CH1 (AF3)
timer C07 AF3
# pin C07: TIM8 CH2 (AF3)
timer A15 AF1
# pin A15: TIM2 CH1 (AF1)
timer B03 AF1
# pin B03: TIM2 CH2 (AF1)

# dma
dma SPI_MOSI 1 NONE
dma SPI_MOSI 2 NONE
dma SPI_MOSI 3 NONE
dma SPI_MOSI 4 NONE
dma SPI_MISO 1 NONE
dma SPI_MISO 2 NONE
dma SPI_MISO 3 NONE
dma SPI_MISO 4 NONE
dma SPI_TX 1 NONE
dma SPI_TX 2 NONE
dma SPI_TX 3 NONE
dma SPI_TX 4 NONE
dma SPI_RX 1 NONE
dma SPI_RX 2 NONE
dma SPI_RX 3 NONE
dma SPI_RX 4 NONE
dma ADC 1 NONE
dma ADC 2 NONE
dma ADC 3 0
# ADC 3: DMA2 Stream 0 Channel 2
dma UART_TX 1 NONE
dma UART_TX 2 NONE
dma UART_TX 3 NONE
dma UART_TX 4 NONE
dma UART_TX 5 NONE
dma UART_TX 6 NONE
dma UART_TX 7 NONE
dma UART_TX 8 NONE
dma UART_RX 1 NONE
dma UART_RX 2 NONE
dma UART_RX 3 NONE
dma UART_RX 4 NONE
dma UART_RX 5 NONE
dma UART_RX 6 NONE
dma UART_RX 7 NONE
dma UART_RX 8 NONE
dma pin B07 0
# pin B07: DMA1 Stream 3 Channel 2
dma pin A09 0
# pin A09: DMA2 Stream 6 Channel 0
dma pin A08 0
# pin A08: DMA2 Stream 6 Channel 0
dma pin C09 0
# pin C09: DMA2 Stream 7 Channel 7
dma pin C08 0
# pin C08: DMA2 Stream 2 Channel 0
dma pin C06 0
# pin C06: DMA2 Stream 2 Channel 0
dma pin C07 0
# pin C07: DMA2 Stream 2 Channel 0
dma pin A15 0
# pin A15: DMA1 Stream 5 Channel 3
dma pin B03 0
# pin B03: DMA1 Stream 6 Channel 3

# feature
feature -RX_PPM
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -RANGEFINDER
feature -TELEMETRY
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -OSD
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -RX_SPI
feature -ESC_SENSOR
feature -ANTI_GRAVITY
feature RX_SERIAL
feature OSD
feature AIRMODE
feature ANTI_GRAVITY

# serial
serial 20 1 115200 57600 0 115200
serial 0 0 115200 57600 0 115200
serial 1 0 115200 57600 0 115200
serial 2 64 115200 57600 0 115200
serial 3 0 115200 57600 0 115200
serial 4 2048 115200 57600 0 115200
serial 5 0 115200 57600 0 115200

# master
set gyro_to_use = FIRST
set adc_device = 3
set blackbox_device = SPIFLASH
set dshot_burst = ON
set dshot_bitbang = AUTO
set dshot_bitbang_timer = AUTO
set current_meter = ADC
set battery_meter = ADC
set vbat_scale = 110
set ibata_scale = 400
set beeper_inversion = ON
set beeper_od = OFF
set beeper_frequency = 0
set max7456_clock = NOMINAL
set max7456_spi_bus = 3
set max7456_preinit_opu = OFF
set led_inversion = 0
set pinio_config = 1,1,1,1
set pinio_box = 255,255,255,255
set usb_msc_pin_pullup = ON
set flash_spi_bus = 2
set gyro_1_bustype = SPI
set gyro_1_spibus = 1
set gyro_1_i2cBus = 0
set gyro_1_i2c_address = 0
set gyro_1_sensor_align = CW270
set gyro_1_align_roll = 0
set gyro_1_align_pitch = 0
set gyro_1_align_yaw = 2700
set gyro_2_bustype = SPI
set gyro_2_spibus = 0
set gyro_2_i2cBus = 0
set gyro_2_i2c_address = 0
set gyro_2_sensor_align = CW0
set gyro_2_align_roll = 0
set gyro_2_align_pitch = 0
set gyro_2_align_yaw = 0
set i2c1_pullup = OFF
set i2c1_clockspeed_khz = 800
set i2c2_pullup = OFF
set i2c2_clockspeed_khz = 800
set i2c3_pullup = OFF
set i2c3_clockspeed_khz = 800
set mco2_on_pc9 = OFF
set scheduler_relax_rx = 25
set scheduler_relax_osd = 25
set box_user_1_name = -
set box_user_2_name = -
set box_user_3_name = -
set box_user_4_name = -

# end the command batch
batch end
```

### Flight controller

Foxeer F722 v4

haslinghuis, 2023-02-24:
Should be fixed in https://github.com/betaflight/betaflight/pull/12329

haslinghuis, 2023-02-24:
Duplicate of #12254
Duplicate of #12283

closed 2023-02-24 by haslinghuis