betaflight/betaflight #12416: Vbat sag compensation doesn't save

ASDosjani, 2023-02-25:
### Describe the bug

I enable sag compensation but after a save and reboot it's turned off again. I tried to enable it from cli but after reboot it's also turned off.

### To Reproduce

Betaflight 4.4+

### Expected behavior

Saving correctly.

### Configurator version

10.9, 10.10

closed 2023-02-25 by ASDosjani

demvlad, 2024-06-12:
I've cought this issue on my OmniNXT F7 FC. The 'vbat sag compensation' option does not save.
But i've found the condition of this issue: 'Voltage meter source' is the 'Esc sensor' at the 'Power  and battery' tab.
If i set  Voltage meter source as 'Onboard ADC', the 'vbat sag compensation' is saved normal.
The ESC - Airbot Typhoon 32. It has sensors and telemetry. The ESC telemetry works properly - i see voltage and current with installed 'Esc sensor'  option.

demvlad, 2024-06-13:
Please, could you write,
what sensors are better for using in BF4.5 - the onboard ADC or ESC telemetry?

haslinghuis, 2024-06-13:
Good find @demvlad `vbat_sag_compensation` requires onboard ADC

https://github.com/betaflight/betaflight/blob/dc40b8f65526a526383ebdb5aaba755712c3fcae/src/main/config/config.c#L285-L289

demvlad, 2024-06-13:
Ok. Perhaps, it is easy to resolve.
I will try to open a PR for that

haslinghuis, 2024-06-13:
@demvlad it's a feature not a bug:

 _The quad must have an on-board battery voltage sensor. Sag compensation options will not be available when ESC based serial voltage telemetry is active, because the voltage data is delayed too much._

See: d63ba914c606e3361da9d2742a3c2aafdacf71b2

ASDosjani, 2024-06-13:
> @demvlad it's a feature not a bug:
>
> _The quad must have an on-board battery voltage sensor. Sag compensation options will not be available when ESC based serial voltage telemetry is active, because [...]

It might be useful to indicate it in the configurator, disabled switch or something

demvlad, 2024-06-13:
Thank's for information. I've understood.
Probably, i will disable the ESC telemetry, because do not need it really.

demvlad, 2024-06-13:
> It might be useful to indicate it in the configurator, disabled switch or something

Will think about this.

haslinghuis, 2024-06-13:
Already working on it