betaflight/betaflight #12419: Battery Telemetry Not Updating after 4.4

russoa20, 2023-02-25:
### Describe the bug

Ugraded Firmware to 4.4.

First flight out I noticed the battery related telemetry on my osd was not updating: Voltage,  Amp Draw, ( Subsequently mah draw)
All other OSD data was updating consistently: RSSi dbm, LQ, armed/disarmed, altitude, throttle %, mode and G Force

When taking it for its first flight with the new firmware and reciever I noticed the battery voltage and ampmeter was "stuck" on its last recorded reading. The MAH counter would increase at a rate proportional to these last stuck readings. This was stuck in both DJI's native voltage readout as well as the Betaflight voltage readout on the OSD.

I have noticed that the voltage gets stuck about a second after arming.

The rest of my osd elements: RSSi dbm, LQ, armed/disarmed, altitude, throttle % and mode all update consistently and appear accurate when compared to the telemetry readout on my remote.

As I flew I had my spotter read the voltage readout on my remote, and it was consistantly stuck at the same voltage as what my goggles read. However, RSSI and LQ updated on the remote consistently.

After flying a few packs and powercycling my goggles and remote, nothing changed.

After plugging the drone into BetaFlight, removing the propellers, plugging a battery in, and spinning the motors through the motors tab, it appears as though the voltage is also frozen until the drone is powercycled.

### To Reproduce

Save CLI dump
Flash BetaFlight 4.4 on KakuteH7mini
Run CLI dump through update checker
Paste and Run CLI dump
Update relevent information that wasn't carried over

### Expected behavior

Accurate voltage and amperage readout on both OSD and remote screen.

Both betaflight and DJI standard osd to report accurate data.

### Support ID

```txt
417b10fa-da9a-4f53-a581-f7cceeed40da
```

### Flight controller

HolyBro Kakute H7 Mini 

### Other components

ESC: Tekko 32
VTX: Vista w/ WTFos Betaflight OSD Root
Goggles: DJI v2 w/ WTFos Betaflight OSD Root
Reciever: RadioMaster ELRS EP1 Fw V.3.20
Remote: RadioMaster Tx16s Mk2
TX Module: RadioMaster Ranger Fw V.3.20

### How are the different components wired up (including port information)

Vista is wired to UART 1
Reciever is wired to UART 6

### Add any other context about the problem that you think might be relevant here

Unfortunately, at the same time I updated betaflight, I also updated EdgeTx to 2.8 and both ELRS modules to 3.2 which makes it a little bit harder to narrow down the issue.

AlexOrlo, 2023-03-02:
Have the same issue with kakuteH7.
ESC: MambaF55_128K
VTX: O3 air unit
Goggles: DJI  FPV v2 
Reciever: TBS diversity nano
Remote: RadioMaster Tx16s Mk2
TX Module: TBS micro

russoa20, 2023-03-02:
@AlexOrlo I am glad I am not alone then. 

Since this post I have tried reflashing BF4.4 with every optional plug in besides Servos but had no luck.

I reverted back to BF4.3 for the time being and everything is back to working as expected.

AlexOrlo, 2023-03-02:
I think to go try to use esc telemetry by uart.

russoa20, 2023-03-02:
@klutvott123 Thank you, I will try in a bit and keep you updated!

AlexOrlo, 2023-03-02:
Hi klutvott123,
Thank you! It's solve the problem.

klutvott123, 2023-03-02:
@AlexOrlo Thanks for testing! It's based on master so compared to 4.4 there's some other stuff too, but the PR I linked to is the only changes  for H7 adc I think.

russoa20, 2023-03-02:
@klutvott123 when I flash, do you recommend enabling "full Chip Erase"?

klutvott123, 2023-03-02:
@russoa20 yes. Especially when flashing to another version like this.

russoa20, 2023-03-02:
@klutvott123 I seem to be getting a "Failed to Load Remote Firmware Error"
Do you notice anything wrong within the attached photo?

[image]

klutvott123, 2023-03-02:
In the dropdown that says "Release", you have to select "Development". Then you should get the "Select commit" dropdown shown in my screenshot above, and that's where you should paste #12439

russoa20, 2023-03-02:
Thank you, I just flashed the firmware. Is it safe to load my previous 4.4 preset?

klutvott123, 2023-03-02:
Should be. Try it 🙂 But remember that this is based on master and there could be issues. I recommend bench testing with props off to start with. If this turns out to be the fix, then we'll add it to the next maintenance release of 4.4 too.

AlexOrlo, 2023-03-02:
// Is it safe to load my previous 4.4 preset?

I simple deselect full erase (for my risk), and just flush this fw. 
flighte 7 pack's, look good.

russoa20, 2023-03-02:
On the bench it appears successful!

Before on a bench test it would sometimes work but then fail once I go to actually fly it. 
I will test again and during a quick flight once it stops raining.

klutvott123, 2023-03-02:
@russoa20 Thanks for testing! 👍

closed 2023-03-03 by blckmn

RupertFinnigan, 2023-03-06:
Hi - sorry if this should be a new bug report - happy to open one if so - however having a problem with this specific update.

I've got four Holybro Katute H7 V1's - all had this problem with Battery Telemetry not updating. Have attempted to apply this fix by selecting the commit, either based on 4.5-zulu or 4.4-zulu - doing so renders the F/C inoperable - doesn't show up in configurator at all. Rolling back to 4.4 release resolves the issue - but obviously then back without Battery Telemetry.

As F/C is inoperable, can't connect into CLI to debug.

klutvott123, 2023-03-06:
@RupertFinnigan Try 4.4.1-zulu. Fix is merged so there's no need to select a PR. Don't forget to do a full chip erase

RupertFinnigan, 2023-03-06:
> @RupertFinnigan Try 4.4.1-zulu. Fix is merged so there's no need to select a PR. Don't forget to do a full chip erase

Apologies - should have mentioned that I've tried that too (Latest 4.4-maintenance) - and 4.5-zulu (master) - same result - FC won't boot.

haslinghuis, 2023-03-06:
[image]

*** Support data submitted *** 
Id: "af1d0073-6dd0-46c2-8016-294228982942"

RupertFinnigan, 2023-03-06:
> [image]
> 
> *** Support data submitted *** Id: "af1d0073-6dd0-46c2-8016-294228982942"
> 
> No problem here. What do you have attached to the board etc.

Nothing too unexpected - Tracer Receiver, and Vista. That's it. 

I didn't include PinIO or GPS in the build, and I'm using configurator 10.9.0.

F/C is Holybro Kakute H7 - Target HBRO/KAKUTEH7(STM32H743). This F/C doesn't have onboard flash for Black Box - uses a SD Card instead. Not sure if that makes any difference.....

haslinghuis, 2023-03-06:
Still okay here ?

[image]

[image]

haslinghuis, 2023-03-06:
Show the ports tab pls

RupertFinnigan, 2023-03-07:
See below - really appreciate your help with this. Thanks.

[image]

[image]

haslinghuis, 2023-03-07:
[image]

Disable MSP from UART2 - not sure why it's there.

MSP on UART1 (needed for VTX (MSP + DP) is probably causing the issue. Don't have DJI hardware for testing.

AndrewMi66, 2023-04-02:
Im having the same issue on the mamba H743 fc aswell, battery voltage updates randomly every few minutes. Ive been trying to flash the FC to 4.4.1 and 4.5, but every time it says failed. i can flash any thing previous to 4.4 but anything after doesnt work. If i change the selected board type, it works fine. I believe this is the error code from the log, "No rule to make target 'MAMBAH743_2022B'.  Stop. make failed with 2" Im sure ive set it up correct. Do you see anything wrong with the attached photo.
[image]

haslinghuis, 2023-04-06:
FYI: 

MAMBAH743 is the right target to use.

EdoardoNunzi96, 2023-09-30:
Hi everyone, I also encountered this problem with a Matek H743, ESC ZeeZ 60A, O3 DJI and Betaflight 4.4, how do you recommend me to act?
Are there any solutions that definitely work today?