betaflight/betaflight #12427: USB don't work properly after update to 4.4.0

Andrey-Svetlichny, 2023-02-27:
### Describe the bug

After update from 4.2.0 to 4.4.0 board can be open in Betaflight configurator, but connection lost in a few seconds/fraction of seconds, depending on features selected in cloud build.
If I use "Flush Firmware" - board switches to mode when only one red LED is active, then betaflight failed to open serial port.
If I switch to DFU mode by button - same result - failed to open serial port. I tried it on 3 different PC (Windows 7-10).
To recover: 
- desolder receiver from T1 (in my case R9MM).
- disconnect GPS
- connect in DFU mode and flash back.
After restoring 4.2 - no problem with USB, so it's not a cable/driver issue.

### To Reproduce

Copter: iFlight TITAN Chimera7 with R9MM
Board: SucceX-D F7 TWING
Target: IFLIGHT_F722_TWING
Build configuration:
- default
- OR Radio protocol: FPORT

Flush and connect to Betaflight Configurator.

### Expected behavior

Board should be accessible from Betaflight Configurator for at least 1 minute.

### Support ID

```txt
Sorry, I flashed it back and don't have Support ID
```

### Flight controller

SucceX-D F7 TWING

### Other components

R9MM receiver on T1 - FPort

### How are the different components wired up (including port information)

Default configuration from iFlight + R9MM receiver connected to T1.
Same result when receiver disconnected from T1.

reflexsa, 2023-03-27:
I was having a similar issue when I upgrade my MAMBAF722 from 4.2 to 4.4. After flashing to 4.4 I was able to connect but the configurator would 'freeze' after a few seconds. Once frozen the configurator menu tree is unresponsive however not 'frozen' according to windows. If I unplug and replug the USB it would connect again for a few seconds. 
What appeared to fix the issue was to drop the PID rate to 4kHz and dshot300 (don't think dshot would have mattered), however I now have a different issue with OSD not working which I will make a separate issue for.

Nihilistic, 2023-03-27:
I'm also experiencing this with my Matek F722SE on BF 4.4.

snowphish, 2023-04-06:
Same, Iflight F7 TwinG BT ... once I apply "defaults" - board won't boot /connect anymore. 
Flashed back to 4.3 and it's working

simoneenomis, 2023-04-07:
Idem problem Iflight F7 TwinG BT ... once I apply "defaults" - board won't boot /connect anymore.
Flash to 4.4.1 
Flashed back to 4.4.0 and it's working

maxx, 2023-04-08:
I am having the same problem with the iflight F722 TWING (mini). 4.4.1 does not work after applying custom defaults.

4.4.0 is fine.

Siclair, 2023-04-10:
Board: SucceX-D F7 TWING
Target: IFLIGHT_F722_TWING
Betaflight FW: 4.4.1

After applying custom defaults it won't connect to configurator 10.9.0. Downgrading to 4.4.0 everything works. The usual problem is that you must disconnect uart1 for DFU to work.

haslinghuis, 2023-05-13:
Please try:

[image]

kikoR2, 2023-05-15:
> I am having the same problem with the iflight F722 TWING (mini). 4.4.1 does not work after applying custom defaults.
> 
> 4.4.0 is fine.
> 
> Configurator is 10.9.0.
> 
> I don't think it's just a problem with USB either. I don't get any beeps after ESC beeps when powering via ESC.

I am also having the same problem described above with the iflight mini-twing.

kikoR2, 2023-05-15:
I tried both development targets and I was unable to access the ESC's unfortuntately.

kikoR2, 2023-05-16:
The problem is likely in my esc board...just connected the FC in question to a different ESC in an existing quad and the esc's initialized with no problem.

haslinghuis, 2023-05-16:
Thanks for testing.

Closing as issue is fixed in 4.4.2 or 4.5.0

closed 2023-05-16 by haslinghuis

psychonaut97, 2023-05-29:
HELP! Same problem same FC, but i cant even flash an old BF Firmware, beacuse the device doesnt gets detected. Tried to use the impulseRC driver fixer, tried different usb cables and also tried to press boot button on fc when plugging in, all methods dont work, now i have a briked brand new TWIN G flightcontroller i am very very sad :(

haslinghuis, 2023-05-29:
Restart computer or configurator and connect flight controller to PC while holding the boot button and flash 4.4.2 pre-release version.