betaflight/betaflight #12428: Onboard Storage missing after flashing 4.4

skaman82, 2023-02-27:
### Describe the bug

After flashing BF 4.4 the onboard storage chip is not recognized anymore. Target OMNIBUSF4V6
 

### To Reproduce

Build options 
[image]

Result: 
[image]

`# resource
resource BEEPER 1 B04
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 A03
resource MOTOR 4 B05
resource MOTOR 5 C08
resource MOTOR 6 C09
resource LED_STRIP 1 B06
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 B10
resource SERIAL_TX 6 C06
resource SERIAL_RX 1 A10
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 A01
resource SERIAL_RX 6 C07
resource I2C_SCL 1 B08
resource I2C_SDA 1 B09
resource LED 1 A08
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 3 C10
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MISO 3 C11
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 B15
resource SPI_MOSI 3 C12
resource ADC_BATT 1 C02
resource ADC_RSSI 1 A00
resource ADC_CURR 1 C01
resource BARO_CS 1 B03
resource OSD_CS 1 A15
resource GYRO_EXTI 1 C04
resource GYRO_CS 1 A04
resource GYRO_CS 2 C14
`

### Expected behavior

Onboard flash usable for logging

### Support ID

```txt
ID: "f90aa97d-2c83-4853-b0e6-bacc378028de"
```

### Flight controller

OMNIBUSF4V6

### Other components

FC: https://www.rotorama.de/akce/airbot-omnibus-f4-v6

skaman82, 2023-02-27:
The onboard storage only showing up flashing the "CORE" build

haslinghuis, 2023-02-27:
FLASH: JEDEC ID=0x00ef4018 16M tells you need to add `FLASH_W25Q128FV` in custom defines. (Enable expert mode on firmware flasher tab). We'll add this to configuration

skaman82, 2023-02-27:
Thanks, FLASH_W25Q128FV in custom defines solved that problem

closed 2023-03-04 by KarateBrot