betaflight/betaflight #12436: Missing Target

Strangedave, 2023-02-28:
### Describe the bug

Iflight_h743_aio target is missing from the betaflight 4.4 Can anyone tell me what to use or add the target?

### To Reproduce

Target is missing in the configurator

### Expected behavior

Would expect to find trhe target

### Support ID

```txt
# Betaflight / IFLIGHT_H743_AIO (IF7A) 4.3.0 Apr 15 2022 / 09:59:09 (9360ab1) MSP API: 1.44

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name IFLIGHT_H743_AIO
mcu_id 004800353130510e32303437
signature 

# name: Flaca 3b

# feature
feature -RSSI_ADC
feature -LED_STRIP

# serial
serial 3 64 115200 57600 0 115200
serial 6 8192 115200 57600 0 115200

# beeper
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TAER1234

# aux
aux 0 0 0 1700 2100 0 0
aux 1 0 2 1725 2100 0 0
aux 2 1 1 900 1125 0 0
aux 3 2 1 1300 1700 0 0
aux 4 13 2 1025 1400 0 0
aux 5 35 2 1400 2100 0 0
aux 6 36 5 1700 2100 0 0
aux 7 36 2 1600 2100 0 0
aux 8 49 3 1900 2100 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BAND_A   A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BAND_B   B CUSTOM  5733 5752 5771 5790 5999 5828 5847 5866
vtxtable band 3 BAND_E   E CUSTOM  5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 AIRWAVE  F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 5
vtxtable powervalues 25 100 200 400 600
vtxtable powerlabels 25 50 100 200 MAX

# master
set gyro_lpf1_static_hz = 375
set gyro_lpf2_static_hz = 750
set gyro_to_use = FIRST
set gyro_lpf1_dyn_min_hz = 375
set gyro_lpf1_dyn_max_hz = 750
set acc_calibration = -6,24,-67,1
set mag_hardware = NONE
set rc_smoothing_auto_factor = 20
set rc_smoothing_auto_factor_throttle = 25
set fpv_mix_degrees = 40
set serialrx_provider = CRSF
set motor_pwm_protocol = DSHOT600
set motor_poles = 12
set yaw_motors_reversed = ON
set small_angle = 180
set simplified_gyro_filter_multiplier = 150
set osd_units = IMPERIAL
set osd_cap_alarm = 500
set osd_alt_alarm = 200
set osd_vbat_pos = 33
set osd_rssi_pos = 45
set osd_link_quality_pos = 2092
set osd_tim_2_pos = 2103
set osd_flymode_pos = 2414
set osd_current_pos = 38
set osd_mah_drawn_pos = 49
set osd_craft_name_pos = 2380
set osd_warnings_pos = 2346
set osd_avg_cell_voltage_pos = 2081
set osd_stat_max_spd = OFF
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_max_esc_rpm = ON
set osd_stat_max_fft = ON
set debug_mode = GYRO_SCALED
set vtx_band = 5
set vtx_channel = 5
set vtx_power = 1
set vtx_freq = 5806
set vcd_video_system = NTSC
set name = Flaca 3b

profile 0

# profile 0
set dterm_lpf1_dyn_min_hz = 82
set dterm_lpf1_dyn_max_hz = 165
set dterm_lpf1_static_hz = 82
set dterm_lpf2_static_hz = 165
set p_pitch = 70
set i_pitch = 113
set d_pitch = 87
set f_pitch = 252
set p_roll = 58
set i_roll = 93
set d_roll = 72
set f_roll = 210
set p_yaw = 68
set i_yaw = 92
set f_yaw = 80
set d_min_roll = 54
set d_min_pitch = 64
set feedforward_averaging = 2_POINT
set feedforward_smooth_factor = 65
set feedforward_jitter_factor = 3
set feedforward_boost = 18
set simplified_pids_mode = RP
set simplified_master_multiplier = 130
set simplified_i_gain = 90
set simplified_d_gain = 140
set simplified_feedforward_gain = 135
set simplified_pitch_d_gain = 105
set simplified_pitch_pi_gain = 115
set simplified_dterm_filter_multiplier = 110

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set thr_mid = 0
set thr_expo = 40
set rates_type = BETAFLIGHT
set roll_rc_rate = 87
set pitch_rc_rate = 87
set yaw_rc_rate = 99
set roll_expo = 25
set pitch_expo = 25
set roll_srate = 71
set pitch_srate = 71
set tpa_rate = 70
set throttle_limit_type = SCALE
set throttle_limit_percent = 95

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
```

### Flight controller

Iflight Beast H743 V1

haslinghuis, 2023-02-28:
There is a V2 ?

Strangedave, 2023-02-28:
Seems to work, i will test asap and confirm

Strangedave, 2023-02-28:
I lied only spinng 2 motors, and msp osd is not working.

Strangedave, 2023-03-02:
from iflight " Beast H743 AIo currently does not have BF4.4 parameters, please continue to use the original version " So seems they aren’t going to do their job. A different person with the same issue said. Now i Have Flashed JHEH743_AIO target  it will fly, but motors will need to be remapped. I’m planning on trying that later, I am closing this issue though.

closed 2023-03-02 by Strangedave