betaflight/betaflight #12442: Telemetry doesn't work on GHF420AIO-BGA

anthropo, 2023-03-03:
### Describe the bug

I have R-XSR with ACCESS, and a JHEMCU GHF420AIOPro Board.
I tried Sbus and smartPort on different Uart TX, and also Fport on different UARts, and finally sBus and SoftSerial, smartPort and non inverted smartport, none of them worked, but receiver works (I can see orders in receiver tab).
I tried the same on aother board (mini MK2 Mamba) and my receivers work (R-XSR).

So my conclusion is that there is a bug with telemetry on GHF420AIO-BGA

### To Reproduce

Connect sbus to sbus pad, smartPort to TX3 pad, configure UART 6 to serial rx , UART 3 telemetry output to smartPort, and put your receiver options on smartport.
Check new devices on telemetry on the transmitter, nothing happens. Only BtRx, RSSI and VFR 

### Expected behavior

Telemetry devices should appear on the transmitter

### Support ID

```txt
7a382ee5-2efd-4949-8f96-15dcd0b52bf0
```

### Flight controller

board: manufacturer_id: JHEF, board_name: JHEF405PRO

### How are the different components wired up (including port information)

sbus to sbus pad, non inverted (hacked) smartPort to TX3 pad

haslinghuis, 2023-03-03:
Disable softserial as it took the resource from TX3

Might have to reassign:
```
resource SERIAL_TX 3 B10
save
```
https://build.betaflight.com/api/support/7a382ee5-2efd-4949-8f96-15dcd0b52bf0

anthropo, 2023-03-03:
Thank you.
But the extract i made was after many tries.
What i'd really like to do is :
Have sbus on sbus pad and smartport on TX3, with the normal pin on the R-XSR (inverted smartPort).

What i did : 
unmap UART 3
remap softserial on TX3 pad 
tried tlm_inverted and tlm_halfduplex ...

But no more joy 

here is the new extract : 
[fac151a9-970e-4aa0-adbc-74379681f97d](https://build.betaflight.com/api/support/fac151a9-970e-4aa0-adbc-74379681f97d)

anthropo, 2023-03-03:
resource SERIAL_TX 11 B10
tlm_inverted = OFF
tlm_halfduplex = ON

haslinghuis, 2023-03-03:
> Disable softserial as it took the resource from TX3

anthropo, 2023-03-04:
This is what i did :

resource SERIAL_TX 3 none
resource SERIAL_RX 3 none
resource SERIAL_TX 11 B10
save

feature SOFTSERIAL
set serialrx_provider = SBUS
feature TELEMETRY

# serial
serial 20 1 115200 57600 0 115200
serial 0 0 115200 57600 0 115200
serial 1 0 115200 57600 0 115200
serial 3 0 115200 57600 0 115200
serial 5 64 115200 57600 0 115200
serial 30 32 115200 57600 0 115200

haslinghuis, 2023-03-04:
Please consider joining Discord. Github is for development.

closed 2023-03-04 by haslinghuis