betaflight/betaflight #12455: Vtx output very limited after flashing BF4.4 on Happymodel ELRSF4 2G4 V1.0

DerBierBaron, 2023-03-05:
### Describe the bug

Hi there,

after flashing the most recent stable BF4.4.0 firmware to Happymodel ELRSF4 2G4 V1.0 the vtx output is too low (putting my body between fc/vtx and vrx breaks up signal completely). I set up the vtx tables after having contacted HM customer service according to their recommendation as follows:

`# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740    0    0
vtxtable powerlevels 3
vtxtable powervalues 0 1 2
vtxtable powerlabels 25 100 200`

This is the only vtx table I tried so far that got me a picture at the desired frequency.
When using the vtx_info command I get this:

`# vtx_info
level 1 dBm, power 1 mW
level 1 dBm, power 1 mW
level 1 dBm, power 1 mW
level 1 dBm, power 1 mW
level 1 dBm, power 1 mW
level 1 dBm, power 1 mW
level 1 dBm, power 1 mW
level 1 dBm, power 1 mW`

I'll attach the initial backup I did before the first flash and the most recent where I tried flashing BF4.4 with most recent nightly configurator build of 10.10 (v20230302.400). Note: I did not set up the fc completely for the latest test, I figured it would be enough to have the features important for the vtx updated and just some others that got the warnings out of the way.

I also somewhere in between tries flashed core only, sent the support data via the "send support data" button in the cli tab and received a support-id, which I noted here in the according form.

Regards,
Klaus

[BTFL_cli_backup_ELRSF4_BF4_4_CRAZYBEEF4SX1280_NightlyConfigurator_20230305.txt](https://github.com/betaflight/betaflight/files/10892572/BTFL_cli_backup_ELRSF4_BF4_4_CRAZYBEEF4SX1280_NightlyConfigurator_20230305.txt)
[BTFL_cli_backup_ELRSF4_BF4_3_20230220_174333.txt](https://github.com/betaflight/betaflight/files/10892573/BTFL_cli_backup_ELRSF4_BF4_3_20230220_174333.txt)

### To Reproduce

Just flashed a HM ELRSF4 2G4 V1.0 with factory settings and BF 4.3 installed with BF4.4 stable and BF configurator 10.9 stable and using the auto-detect function to get the target (GUI stated that CRAZYBEEF4SX1280 was detected).

### Expected behavior

To have signal strength comparable to my HM X12 ELRS (at 25mW I get signal through 3 brick walls at about 15 meters distance).

### Support ID

```txt
bfc60f9c-abff-4857-b114-54aa1baded93
```

### Flight controller

Happymodel ELRSF4 2G4 V1.0

DerBierBaron, 2023-04-07:
Bump to try and ask for help looking into this again.

haslinghuis, 2023-05-08:
Have no access to facebook but please take a look and report back if & what solves the [issue](https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=&cad=rja&uact=8&ved=2ahUKEwi56sHFm-b-AhW1SvEDHSPiDmYQFnoECFEQAQ&url=https%3A%2F%2Fwww.facebook.com%2Fgroups%2F636441730280366%2Fposts%2F790125814911956%2F&usg=AOvVaw2MD69llOvUYwFA9evQTDSg)

closed 2023-06-16 by github-actions[bot]

IkerBlanco, 2024-04-21:
Your vtx power values are wrong, try with 14 , 20 and 26