betaflight/betaflight #12471: no gyro

macsleep, 2023-03-07:
### Describe the bug

Upgrade from BF 4.2.11 to 4.4.0 (full chip erase) on a NameLessRC Super AIO FC+ESC+DVR F4 15A 720p 1S (MATEKF405MINI) leaves FC with "no gyro". With "core only" gyro is there.   

### To Reproduce

Install BF 4.4.0 MATEKF405MINI on a NameLessRC Super AIO FC+ESC+DVR F4 15A 720p 1S.

### Expected behavior

gyro available after upgrade.

### Support ID

```txt
597af532-098f-45c4-8d59-d2e21ae98162
```

### Flight controller

NameLessRC Super AIO FC+ESC+DVR F4 15A 720p 1S

haslinghuis, 2023-03-07:
1. Enable show release candidates
2. Enable expert mode
3. Click Auto-detect
4. Add `GYRO_SPI_MPU6000 USE_ACC_SPI_MPU6000` to custom defines
5. Load  Firmware Online
6. Flash Firmware

macsleep, 2023-03-08:
Thanks Mark, worked like a charm with one small adaption: I removed the USE_ from the USE_ACC_SPI_MPU6000 define.

razzeee, 2023-03-08:
I think Iam setting this with my speedybee f405 v3 too, would the same fix be applicable?

macsleep, 2023-03-08:
Speedybee F405 V3 uses BMI270 gyro if I'm not mistaken. If you also get "no gyro" message with BF 4.4.0 then the MPU6000 defines will probably not help. There is a USE_ACCGYRO_BMI270 define in BF but I have no clue if this will help (Mark is probably more qualified to answer this :-) ).

haslinghuis, 2023-03-08:
Yes ACCGYRO_BMI270 will work

http://betaflight.com/docs/manufacturer/manufacturer-design-guidelines#421-defines-for-gyro-and-acc

haslinghuis, 2023-03-09:
Show your screen when flashing (hint enable expert mode in firmware flasher tab)

razzeee, 2023-03-09:
Expert Mode actually hides custom define for me

macsleep, 2023-03-09:
Just curious if you have tried the "core only" installation for testing (slider in the title bar of the build configuration)? If gyro does not show with "core only" that would point to a bigger issue...

razzeee, 2023-03-11:
Yeah, just got a new FC, the exact same one - I could flash it and it still shows gyro etc. So the old FC might be lost :(

macsleep, 2023-03-12:
Speedybee FCs are known to be good value for money but I don't know how durable they are. Good luck with the new FC :-) .

chimera15, 2023-03-29:
Same problem, matekf411, on jhemcu ghf13 aio.  Everything working in 4.2, gyro dead on 4.4.  

I have 2 other jhemcu ghf13's with no issues on 4.4. but this one is apparently a newer one, maybe it uses a different gyro?

LordGG, 2023-03-30:
Same issue with SpeedyBee F7 Mini (SPEEDYBEEF7MINIV2).

Solved with custom defines ACCGYRO_BMI270, thanks.

chimera15, 2023-03-30:
> Same issue with SpeedyBee F7 Mini (SPEEDYBEEF7MINIV2).
> 
> Solved with custom defines ACCGYRO_BMI270, thanks.

Weird I have that fc and didn't have any issues upgrading to 4.4...I wonder if it also has different gyros.

LordGG, 2023-03-30:
> Weird I have that fc and didn't have any issues upgrading to 4.4...I wonder if it also has different gyros.

Mine is a V2 and has a BMI270. I have tried flashing it 3 times to make sure nothing went bad and default settings were correctly applied, but it always gave "no gyro". Forcing the BMI270 did the trick :)

glennvenghaus, 2023-04-04:
Same issue on SPEEDYBEEF7MINI  target, which has a BMI270
Adding ACCGYRO_BMI270 to custom defines fixes issue.
Please also include there while you are at it with speedybee.
Tnx

haslinghuis, 2023-05-05:
Completed

closed 2023-05-05 by haslinghuis