betaflight/betaflight #12472: Holybro Kakute H7 Fails to Boot with 4.4.1-zulu, and requires SD card present to boot on 4.5-zulu

RupertFinnigan, 2023-03-07:
### Describe the bug

Attempted to flash 4.4.1-zulu and 4.5-zulu to gain benefits of issue resolved in #12439 H7 Fix frozen ADC values.

Configurator suggests flash is successful, but Flight Controller doesn't boot at all under 4.4.1-zulu, and will only boot with SD card inserted for 4.5-zulu. Even then, no custom defaults are suggested - leaving F/C unusable.

4.5-zulu will only boot if SD card is present. Remove SD card - no boot. Insert SD card - boots.

### To Reproduce

Ensure no SD card is inserted into Flight controller. 
Attempt to flash 4.4.1-zulu, or 4.5-zulu. 
Configurator suggests flashing process completes successfully, however FC does not boot.

Insert SD card into slot on FC.
Attempt to flash 4.5-zulu.
Configurator suggests flashing process completes successfully, FC does boot, however no custom default are suggested leaving FC unusable.

Remove SD card and power cycle FC.
FC will not boot.

Reinsert SD card and power cycle FC.
FC boots.

### Expected behavior

Would expect flashing of either 4.4.1-zulu or 4.5-zulu to be successful, regardless of presence of SD card in FC, and custom defaults to be suggested on first boot following flashing.

This was the case up to and including 4.4.0 release.

### Support ID

```txt
FC will not boot - so can't generate information.
```

### Flight controller

Holybro Kakute H7

### Other components

Tracer Receiver.
DJI Vista.

### How are the different components wired up (including port information)

Tracer Receiver on UART 6.
DJI Vista MSP on UART 1.

RupertFinnigan, 2023-03-08:
I've spent a bit more time trying to understand exactly where the bug was introduced - stepping through each commit in 4.4-maintainence branch until the F/C stops booting - building and flashing the hex's locally.

The bug preventing booting was introduced with commit 8703139 - "4.4.1: Latest changes from unified targets, and sdcard build fix." Builds based on commits up to this booted fine.

Wish I had a bit more knowledge to be able to suggest a valid fix for this - but hope this is somewhat helpful.

haslinghuis, 2023-03-08:
Thanks. Already tried to revert that part but it does not solve the issue

PR #12488 (2023-03-09): CONFIG: Fix SDCARD CS
Fixes: #12472
closed 2023-03-09 by blckmn