betaflight/betaflight #12501: Missing Blackbox After Flashing 4.4

CoolwippFPV, 2023-03-10:
### Describe the bug

After flashing to 4.4 the Blackbox is no longer visible. After reviewing similar issues here the fix of USE_FLASH_W25Q128FV fails when trying to load the firmware online. 

Target: FOXEERF722V4

### To Reproduce

Flashing any firmware from 4.4.0 to higher

### Expected behavior

Reflashing in order to get ICM gyro fix, expected Blackbox tab to show normal options.

### Support ID

```txt
"9e321c75-d3a0-4b93-ab7a-fa52cf509ed1"
```

### Flight controller

FOXEER F722 V4 DUAL BEC 5V/10V HV 4-8S MINI FLIGHT CONTROLLER - 20X20MM

haslinghuis, 2023-03-11:
Fixed in: https://github.com/betaflight/betaflight/pull/12498

closed 2023-03-11 by haslinghuis

CoolwippFPV, 2023-03-11:
Hello Haslinghuis, I just tried to reflash again and I am getting an error when trying to load remote firmware. I am not sure where I am going wrong. 
I have here of the error I am getting if that will help

https://build.betaflight.com/api/builds/290893fedd99324994bc9816d9620708/log

haslinghuis, 2023-03-11:
It has timed out :disappointed: Just try again :smiley: