betaflight/betaflight #12503: Cloud build not working

TheIsotopes, 2023-03-11:
### Describe the bug

Cloud build on controllers with flash are failing,

MATEKF722MINI - https://build.betaflight.com/api/builds/e0ad5890a8b645c12a774baac1d8b5ee/log
MATEKF405TEMINI - https://build.betaflight.com/api/builds/778a85a1199c15736fe559ea8bafbeff/log

but with sdcard all is working.

MATEKH743 - https://build.betaflight.com/api/builds/833dfb1640297473a36c22875205bc29/log

@haslinghuis @blckmn please can someone check this.

### To Reproduce

use cloud build on controllers with flash like MATEKF722MINI, MATEKF405TEMINI

### Expected behavior

working cloud build

### Support ID

```txt
xxx
```

### Flight controller

MATEKF722MINI, MATEKF405TEMINI

TheIsotopes, 2023-03-11:
fixed in #12504

TheIsotopes, 2023-03-11:
reopen again ... `BLACKBOX` as custom defines is still needed. otherwise no flash will be displayed in the blackbox configurator.

@blckmn any ideas?

blckmn, 2023-03-11:
This should now be resolved.

TheIsotopes, 2023-03-11:
@blckmn nope ... not working.

TheIsotopes, 2023-03-11:
@blckmn nope ... not working.

Another question. How can i rebuild a cached build?

blckmn, 2023-03-11:
> @blckmn nope ... not working.
> 
> 
> 
> Another question. How can i rebuild a cached build?

Sorry the original build error.

Will add Blackbox as an option.

blckmn, 2023-03-11:
Rebuilding a cached build... easiest option is to put the current master commit hash into the commit option.

Unless something has changed the result will be the same :)

haslinghuis, 2023-03-12:
Tested master 4.5 cloud build:

- MATEKF405TEMINI and MATEKH743 working
- SPEEDYBEEF7V3, KAKUTEH7 and KAKUTEH7MINI needs BLACKBOX as custom define
- MAMBA740MK4 does not work with or without BLACKBOX as custom define

TheIsotopes, 2023-03-12:
@haslinghuis that's weird,
I don't know how you tested it, but the MATEKF045TEMINI doesn't work. Just tried it, blackbox as an option is not in the list and without custome defines, the settings do not appear in the configurator.
MATEKH743 does not have this problem as it works with sd card as i said above.
imho this only affects fc's with flash memory.

haslinghuis, 2023-03-12:
Okay today have to add BLACKBOX to custom defines to get it working.
MATEKH743 is working because it already has the USE_BLACKBOX define.

[image]

https://build.betaflight.com/api/builds/4e05623427e5c0a4dff1eb95efb56b59/log

[image]

TheIsotopes, 2023-03-12:
> MATEKH743 is working because it already has the USE_BLACKBOX define.

and he is the only one with this define.

PR #12511 (2023-03-13): CONFIG: Fix blackbox
Fixes: #12503 

Don't know if it's better to add `USE_BLACKBOX` to each board having either `USE_FLASH` or `USE_SDCARD` defined.
closed 2023-03-13 by blckmn