betaflight/betaflight #12521: whimpering motors with ICM42605

TheIsotopes, 2023-03-15:
### Describe the bug

Since #12444 is merged the motors on my MATEKF405TEMINI with ICM42605 only whimpering.
It seems like the gyro has become much more sensitive. If i revert the PR on actual master everything is fine again.
In the worst case, a "fly me to the moon" begins.

see also https://github.com/betaflight/betaflight/pull/12444#issuecomment-1466867145

### To Reproduce

using MATEKF405TEMINi with ICM42605

### Expected behavior

no whimpering motors

### Support ID

```txt
Id: "32a2a436-f11d-4f42-97d2-d07e088107bf"
```

### Flight controller

MATEKF405TEMINI

sugaarK, 2023-03-15:
https://github.com/betaflight/betaflight/pull/12524  Can you test this please @TheIsotopes

closed 2023-03-17 by KarateBrot

brandonrc, 2023-04-07:
@TheIsotopes did this fix your issue? I still have the problem with this FC.. might just stop using them... I have like 5 of them :(

brandonrc, 2023-04-08:
I still have this issue.. flew 4.5 and it goes to the moon on defaults.. but flash to 4.3 and it flys smooth as butter on defaults.

brandonrc, 2023-04-08:
Board

http://www.mateksys.com/?portfolio=f405-minite#tab-id-2

IMU ICM42605  works on 4.4.x and 4.5
IMU ICM42688-p does not work on 4.4.x and 4.5 (goes to the moon
)

commit 8659032c (2023-03-17): Fixes #12521 Add separate AAF values for ICM-42605 (#12524)

Add separate AAF values for ICM-42605