betaflight/betaflight #12522: GPS Module BN-220 not working in BetaFlight Firmware 4.4.0

anilvermab, 2023-03-15:
### Describe the bug

GPS module BN-220 does not work in BTFL 4.4.0.

I have set up the port and enabled the GPS in configuration but does not seem to activate after saving & rebooting. Goes back to a disabled state.

### To Reproduce

upgraded from 4.3.0 to 4.4.0
The same GPS module works fine when I revert to the old firmware.

### Expected behavior

GPS modules should work in the latest firmware.

### Support ID

```txt
32df3a27-f071-42d1-8899-3b7adaa71832
```

### Flight controller

SpeedyBee F7 V3 BL32 50A 30x30 Stack

### Other components

Camera - DJI 03.
1350KV motor

### How are the different components wired up (including port information)

GPS is on Port 6 baud rate 57600 (works fine in 4.3.3 firmware)
DJI O3 is on UART 1 & 2

x4code, 2023-03-15:
it works on all my quads :) have you check GPS option when flashing?

anilvermab, 2023-03-15:
I used the speedy bee app to flash firmware through wifi, haven't noticed any such options. Although reverting to the old firmware with the same mobile app does bring the GPS back. So not sure if it's a firmware bug or issue with the app. I'll try it on my laptop with the configurator app and update. Thanks for your response

closed 2023-03-16 by anilvermab