betaflight/betaflight #12527: Failed to open USB device

przepka1985, 2023-03-16:
### Describe the bug

Cannot flasf firmware on tmotorf7 board.
Drivers seams ok until you not get to flashing and switch to DFU mode.
Used zadig-2.8 and ImpulseRC_Driver_Fixer with no positive results

### To Reproduce

Windows 11
Beta flight configurator 10.9.0
TMOTORF7 firmware 4.3.0

### Expected behavior

Able to flash firmware

### Support ID

```txt
# version
# Betaflight / STM32F7X2 (S7X2) 4.3.0 Apr 25 2022 / 01:06:24 (9d71184) MSP API: 1.44
# config: manufacturer_id: TMTR, board_name: TMOTORF7, version: c2b8c0d3, date: 2021-06-18T03:42:42Z
# board: manufacturer_id: TMTR, board_name: TMOTORF7

# status
MCU F722 Clock=216MHz, Vref=3.28V, Core temp=44degC
Stack size: 2048, Stack address: 0x20010000
Configuration: CONFIGURED, size: 4034, max available: 16384
Devices detected: SPI:1, I2C:1
Gyros detected: gyro 1 locked dma
GYRO=MPU6000, ACC=MPU6000, BARO=BMP280
OSD: MAX7456
System Uptime: 734 seconds, Current Time: 2023-03-16T12:41:47.522+00:00
CPU:57%, cycle time: 125, GYRO rate: 8000, RX rate: 15, System rate: 9
Voltage: 1 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 7
SD card: Not configured
Arming disable flags: RXLOSS CLI MSP RPMFILTER

# flash_info
Flash sectors=256, sectorSize=65536, pagesPerSector=256, pageSize=256, totalSize=16777216
Paritions:
  0: FLASHFS   0 255
FlashFS size=16777216, usedSize=501760

# dump hardware

# version
# Betaflight / STM32F7X2 (S7X2) 4.3.0 Apr 25 2022 / 01:06:24 (9d71184) MSP API: 1.44
# config: manufacturer_id: TMTR, board_name: TMOTORF7, version: c2b8c0d3, date: 2021-06-18T03:42:42Z

# start the command batch
batch start

board_name TMOTORF7
manufacturer_id TMTR

# resources
resource BEEPER 1 C13
resource MOTOR 1 C06
resource MOTOR 2 C07
resource MOTOR 3 B00
resource MOTOR 4 B01
resource MOTOR 5 B06
resource MOTOR 6 B08
resource MOTOR 7 B07
resource MOTOR 8 C09
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 A15
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP 1 A08
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 C10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 C12
resource SERIAL_TX 6 NONE
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 C11
resource SERIAL_RX 4 A01
resource SERIAL_RX 5 D02
resource SERIAL_RX 6 NONE
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 NONE
resource SERIAL_RX 12 NONE
resource I2C_SCL 1 NONE
resource I2C_SCL 2 B10
resource I2C_SCL 3 NONE
resource I2C_SCL 4 NONE
resource I2C_SDA 1 NONE
resource I2C_SDA 2 B11
resource I2C_SDA 3 NONE
resource I2C_SDA 4 NONE
resource LED 1 B02
resource LED 2 NONE
resource LED 3 NONE
resource RX_BIND 1 NONE
resource RX_BIND_PLUG 1 NONE
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 3 B03
resource SPI_SCK 4 NONE
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MISO 3 B04
resource SPI_MISO 4 NONE
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 B15
resource SPI_MOSI 3 B05
resource SPI_MOSI 4 NONE
resource CAMERA_CONTROL 1 B09
resource ADC_BATT 1 C01
resource ADC_RSSI 1 C03
resource ADC_CURR 1 C02
resource ADC_EXT 1 NONE
resource BARO_CS 1 C15
resource BARO_EOC 1 NONE
resource BARO_XCLR 1 NONE
resource COMPASS_CS 1 NONE
resource COMPASS_EXTI 1 NONE
resource SDCARD_CS 1 NONE
resource SDCARD_DETECT 1 NONE
resource PINIO 1 C14
resource PINIO 2 NONE
resource PINIO 3 NONE
resource PINIO 4 NONE
resource USB_MSC_PIN 1 NONE
resource FLASH_CS 1 C08
resource OSD_CS 1 B12
resource RX_SPI_CS 1 NONE
resource RX_SPI_EXTI 1 NONE
resource RX_SPI_BIND 1 NONE
resource RX_SPI_LED 1 NONE
resource RX_SPI_CC2500_TX_EN 1 NONE
resource RX_SPI_CC2500_LNA_EN 1 NONE
resource RX_SPI_CC2500_ANT_SEL 1 NONE
resource GYRO_EXTI 1 C04
resource GYRO_EXTI 2 NONE
resource GYRO_CS 1 A04
resource GYRO_CS 2 NONE
resource USB_DETECT 1 NONE
resource PULLUP 1 NONE
resource PULLUP 2 NONE
resource PULLUP 3 NONE
resource PULLUP 4 NONE
resource PULLDOWN 1 NONE
resource PULLDOWN 2 NONE
resource PULLDOWN 3 NONE
resource PULLDOWN 4 NONE

# timer
timer C06 AF2
# pin C06: TIM3 CH1 (AF2)
timer C07 AF2
# pin C07: TIM3 CH2 (AF2)
timer B00 AF3
# pin B00: TIM8 CH2N (AF3)
timer B01 AF2
# pin B01: TIM3 CH4 (AF2)
timer B06 AF2
# pin B06: TIM4 CH1 (AF2)
timer B08 AF2
# pin B08: TIM4 CH3 (AF2)
timer A15 AF1
# pin A15: TIM2 CH1 (AF1)
timer B09 AF3
# pin B09: TIM11 CH1 (AF3)
timer A08 AF1
# pin A08: TIM1 CH1 (AF1)
timer B07 AF2
# pin B07: TIM4 CH2 (AF2)
timer C09 AF3
# pin C09: TIM8 CH4 (AF3)

# dma
dma SPI_MOSI 1 NONE
dma SPI_MOSI 2 NONE
dma SPI_MOSI 3 NONE
dma SPI_MOSI 4 NONE
dma SPI_MISO 1 NONE
dma SPI_MISO 2 NONE
dma SPI_MISO 3 NONE
dma SPI_MISO 4 NONE
dma SPI_TX 1 NONE
dma SPI_TX 2 NONE
dma SPI_TX 3 NONE
dma SPI_TX 4 NONE
dma SPI_RX 1 NONE
dma SPI_RX 2 NONE
dma SPI_RX 3 NONE
dma SPI_RX 4 NONE
dma ADC 1 1
# ADC 1: DMA2 Stream 4 Channel 0
dma ADC 2 NONE
dma ADC 3 NONE
dma UART_TX 1 NONE
dma UART_TX 2 NONE
dma UART_TX 3 NONE
dma UART_TX 4 NONE
dma UART_TX 5 NONE
dma UART_TX 6 NONE
dma UART_TX 7 NONE
dma UART_TX 8 NONE
dma UART_RX 1 NONE
dma UART_RX 2 NONE
dma UART_RX 3 NONE
dma UART_RX 4 NONE
dma UART_RX 5 NONE
dma UART_RX 6 NONE
dma UART_RX 7 NONE
dma UART_RX 8 NONE
dma pin C06 0
# pin C06: DMA1 Stream 4 Channel 5
dma pin C07 0
# pin C07: DMA1 Stream 5 Channel 5
dma pin B00 0
# pin B00: DMA2 Stream 2 Channel 0
dma pin B01 0
# pin B01: DMA1 Stream 2 Channel 5
dma pin B06 0
# pin B06: DMA1 Stream 0 Channel 2
dma pin B08 0
# pin B08: DMA1 Stream 7 Channel 2
dma pin A15 0
# pin A15: DMA1 Stream 5 Channel 3
dma pin B09 NONE
dma pin A08 0
# pin A08: DMA2 Stream 6 Channel 0
dma pin B07 0
# pin B07: DMA1 Stream 3 Channel 2
dma pin C09 0
# pin C09: DMA2 Stream 7 Channel 7

# feature
feature -RX_PPM
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -RANGEFINDER
feature -TELEMETRY
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -OSD
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -RX_SPI
feature -ESC_SENSOR
feature -ANTI_GRAVITY
feature RX_SERIAL
feature GPS
feature TELEMETRY
feature LED_STRIP
feature OSD
feature AIRMODE
feature ESC_SENSOR
feature ANTI_GRAVITY

# serial
serial 20 1 115200 57600 0 115200
serial 0 64 115200 57600 0 115200
serial 1 2 115200 57600 0 115200
serial 2 2048 115200 57600 0 115200
serial 3 1024 115200 57600 0 115200
serial 4 0 115200 57600 0 115200

# master
set gyro_to_use = FIRST
set align_mag = CW180
set mag_align_roll = 0
set mag_align_pitch = 0
set mag_align_yaw = 1800
set mag_bustype = I2C
set mag_i2c_device = 2
set mag_i2c_address = 0
set mag_spi_device = 0
set baro_bustype = I2C
set baro_spi_device = 3
set baro_i2c_device = 2
set baro_i2c_address = 0
set rx_spi_protocol = V202_250K
set rx_spi_bus = 0
set rx_spi_led_inversion = OFF
set adc_device = 1
set blackbox_device = NONE
set dshot_burst = OFF
set dshot_bitbang = AUTO
set dshot_bitbang_timer = AUTO
set current_meter = ADC
set battery_meter = ADC
set vbat_scale = 110
set ibata_scale = 400
set beeper_inversion = ON
set beeper_od = OFF
set beeper_frequency = 0
set sdcard_detect_inverted = OFF
set sdcard_mode = OFF
set sdcard_spi_bus = 0
set sdio_clk_bypass = OFF
set sdio_use_cache = OFF
set sdio_use_4bit_width = OFF
set max7456_clock = NOMINAL
set max7456_spi_bus = 2
set max7456_preinit_opu = OFF
set cc2500_spi_chip_detect = ON
set led_inversion = 0
set pinio_config = 1,1,1,1
set pinio_box = 39,0,0,0
set usb_msc_pin_pullup = ON
set flash_spi_bus = 3
set gyro_1_bustype = SPI
set gyro_1_spibus = 1
set gyro_1_i2cBus = 0
set gyro_1_i2c_address = 0
set gyro_1_sensor_align = CW0
set gyro_1_align_roll = 0
set gyro_1_align_pitch = 0
set gyro_1_align_yaw = 0
set gyro_2_bustype = SPI
set gyro_2_spibus = 0
set gyro_2_i2cBus = 0
set gyro_2_i2c_address = 0
set gyro_2_sensor_align = CW0
set gyro_2_align_roll = 0
set gyro_2_align_pitch = 0
set gyro_2_align_yaw = 0
set i2c1_pullup = OFF
set i2c1_clockspeed_khz = 800
set i2c2_pullup = OFF
set i2c2_clockspeed_khz = 800
set i2c3_pullup = OFF
set i2c3_clockspeed_khz = 800
set mco2_on_pc9 = OFF
set scheduler_relax_rx = 25
set scheduler_relax_osd = 25
set box_user_1_name = -
set box_user_2_name = -
set box_user_3_name = -
set box_user_4_name = -

# end the command batch
batch end

*** Support data submitted *** 
Id: "34978d43-7f00-4bb5-8771-d6f765811593"

# copy ID and provide to the betaflight team.
```

### Flight controller

TMOTORF7

### Other components

Frsky TDMX recievier, Matek VTX, Tmotor Velox 45amp 4in1 ESC 32bit

x4code, 2023-03-16:
problem is switching to DFU mode? have you tried using another computer? 
i had that issue recently and was something bad related to my laptop. now fixed (not sure how to be honest) haha

przepka1985, 2023-03-16:
Yes sorted out, impulse was not able to fix dfu drivers. I have done it
manually.
Anyway I got another issue 😔
When already updated from 4.3.0 to 4.4.0 don't have GPS, mag, Rx, LEDs,
basically nothing 😃 except new firmware. When I applied cli backup from
4.3.0 Rx working. But nothing more.

SteveCEvans, 2023-03-19:
> problem is switching to DFU mode? have you tried using another computer?
> i had that issue recently and was something bad related to my laptop. now fixed (not sure how to be honest) haha

Unintended DFU can be caused by devices such as ELRS RX or GPS on a serial port. You may need to switch which ports are in use. Try disconnecting them.

closed 2023-03-26 by blckmn

dengdizzy, 2023-04-04:
https://betaflight.com/docs/wiki/archive/DFU-Hijacking#what-is-dfu-hijacking

PR #12528 (2023-03-26): CONFIG: Add BMP280 to TMOTORF7
Fixes: #12527