betaflight/betaflight #12552: impossible to save 2 different pid preset with different filters

samuelaugry, 2023-03-21:
### Describe the bug

i have 2 different preset of PID but it's impossible to have different filters between the 2 preset . 
when i change filter settings in the first preset, it's automatically changing on the second preset ! 

### To Reproduce

just create 2 profil of pid and test to have different filter settings between this 2 profil 

### Expected behavior

juste want to have 2 different preset with different filter settings .

### Support ID

```txt
8e268521-324e-4386-9774-79f7b43cc2c4
```

### Flight controller

aikon F7V2

### Other components

Aikon AK55A V3 esc /fusion C220 frame
crossfire / axisflying AF227 1960KV / vista with dji camera 

### Add any other context about the problem that you think might be relevant here

i run Bf configurator 10.9 on windows 11 and quad is on BF4.4

x4code, 2023-03-21:
On 4.4 I think only the right side are associate with that specific PID.
Left side are the same for every PID.
(If im not wrong ofc)

haslinghuis, 2023-03-22:
@limonspb can we load a preset into a different profile?