betaflight/betaflight #12580: Analog OSD not working MAMBAF722 with 4.4

reflexsa, 2023-03-27:
### Describe the bug

After flashing to 4.4 I can't seem to get the OSD to function. Flashing back to 4.2.5 (my previous version) it works perfectly again.
Things I have tried:
- Cloud build with OSD (SD). Cloud build with OSD (SD) and OSD (HD)
- OSD is activated in the configuration tab.
- Switched between PAL, NTSC and auto.
- Clean flash with only 'apply custom defaults'
- OSD elements are in view

### To Reproduce

Flash BF4.4 including OSD (SD).

### Expected behavior

There should be a working OSD.

### Support ID

```txt
Sorry flashed back to 4.2.11 now so did not generate and ID.
```

### Flight controller

Diatone MAMBAF722

ASDosjani, 2023-03-27:
Which target exactly?
Try 4.5 and send your current diff

reflexsa, 2023-03-27:
Target: MAMBAF722

4.2.11 diff (was on 4.2.5 but went to this version as it was in the configurator, seems to work with my diff)
`# 

# diff all

# version
# Betaflight / STM32F7X2 (S7X2) 4.2.11 Nov  9 2021 / 20:29:32 (948ba6339) MSP API: 1.43

# config: YES

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name MAMBAF722
manufacturer_id DIAT
mcu_id 002b00333338510d39393832
signature 

# name: XL7

# resources
resource SERIAL_TX 3 NONE
resource SERIAL_RX 3 NONE
resource I2C_SCL 2 B10
resource I2C_SDA 2 B11

# feature
feature -RX_PARALLEL_PWM
feature GPS

# beacon
beacon RX_LOST
beacon RX_SET

# serial
serial 0 0 115200 57600 0 115200
serial 1 64 115200 57600 0 115200
serial 4 2 115200 57600 0 115200

# led
led 0 0,0::C:2
led 1 15,0::C:3
led 2 15,2::C:3
led 3 15,1::C:3
led 4 15,3::C:3

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 2 1800 2100 0 0
aux 2 2 2 1350 1625 0 0
aux 3 46 4 1700 2100 0 0
aux 4 13 3 1700 2100 0 0
aux 5 35 1 1700 2100 0 0
aux 6 47 6 1700 2100 0 0

# rxrange
rxrange 0 988 2011
rxrange 1 988 2011
rxrange 2 988 2011
rxrange 3 988 2011

# rxfail
rxfail 3 s 1250
rxfail 6 s 2000

[... 282 lines cut ...]
profile 0

rateprofile 0

# rateprofile 0
set rates_type = BETAFLIGHT
set roll_rc_rate = 77
set pitch_rc_rate = 77
set yaw_rc_rate = 77
set roll_srate = 74
set pitch_srate = 74
set yaw_srate = 74

rateprofile 1

rateprofile 2

rateprofile 3

# restore original rateprofile selection
rateprofile 0

# save configuration
save
# `

ASDosjani, 2023-03-27:
Is 4.3 working? Are you using the latest configurator?

reflexsa, 2023-03-28:
I previously tried 4.3 when it was still RC (RC5 I think), but had that issue with the crossfire negotiated baud so reverted back. OSD did work on that RC of 4.3. 
Yes, I am using the latest configurator. I will try flash 4.5 if I have time in the next day or two, will be away for 2 weeks or so but will try do it before then

closed 2023-05-06 by github-actions[bot]