betaflight/betaflight #12581: No Transmitter Input in Betaflight

Ostamoa, 2023-03-27:
### Describe the bug

Hello,
i recently built my first own fpv drone. And now i have the first problem.
My FC isn´t getting any input from my transmitter.

### To Reproduce

i dont know how you can reproduce the behavior, for a lot of people, the solutions on the internet helped.

### Expected behavior

The FC should show the Inputs when moving the sticks on my transmitter.

### Support ID

```txt
60bd9a6b-cf8b-42b2-bdd5-22ce5e9d6b82
```

### Flight controller

SpeedyBee F405 V3 BLS 50A 30x30 FC&ESC Stack

### Other components

I have the FrSky Taranis X9D Plus 2019 EU + Frksy R-XSR EU + SpeedyBee F405 V3 BLS 50A 30x30 FC&ESC Stack

### How are the different components wired up (including port information)

I have already connected my trasnmitter to my receiver (green light is on, as well as the blue one for SBUS) and soldered my receiver to my FC.
Red to 5V / Black to G / Yellow to SBUS (i did also try R2)
in Betaflight i put UART 2 at "Serieller Empfänger" (I´m German)
under the category "Receiver" i set it to "seriell" and "SBUS"

### Add any other context about the problem that you think might be relevant here

I did also try to invert the signal, as this should may help, but it didnt.
[image]
[image]
[image]
[image]
[image]

druckgott, 2023-06-09:
How did you solve this I have the same problem?

Ostamoa, 2023-06-09:
@druckgott 
Ich denk du bist deutscher😅
Hab transmitter mit receiver verwechselt, aber ist egal, du hast verstanden was ich mein
Der stecker mit den kabel dran ist falsch eingepint, der ist normalerweise zum flaschen des receivers.
Du musst das gelbe kabel auf einen anderen pin legen, schau ins wiring diagram

I meant receiver, not transmitter...
But u have to move the yellow wire one or 2 pins, the curennt wiring is for flashing the receiver, look at the wiring diagram, and move the yellow wire to sbus

druckgott, 2023-06-09:
das hab ich schon geprüft. Ich habe TX an RX und RX an TX, habs aber per Stecker dran. Du hast es aber schon an UART2? Da hattest du kein Probleme. Ich habe allerdings ein ExpressLRS Empfänger, dieser geht aber, weil ich habe an der Drone gerade "nur" das Board umgebaut. Der Empänger war vorher schon im Einsatz. Deswegen bin ich etwas verwundert.
Hab auch über die Pads gerade extra nochmal gegengemessen, damit RX an TX und TX an RX ist.
[image]
[image]

druckgott, 2023-06-09:
hier die Lösung falls jemand sucht
https://github.com/betaflight/betaflight/issues/12879