betaflight/betaflight #12585: DarwinFPV tiny ape no gryo when upgrading to betaflight 4.4

Arby5577, 2023-03-27:
### Describe the bug

When flashing the DarwinFPV Tiny Ape to betaflight 4.4.0 with auto detect and apply custom defaults, no gyro is detected.

### To Reproduce

Flash the DarwinFPV tiny ape to betaflight 4.4.0 using auto detect and apply custom defaults.

### Expected behavior

Gyro should be detected.

### Support ID

```txt
c7877ae7-8b3b-49df-85f1-20c75c7dcdb0
```

### Flight controller

Darwin ELRS2.0 F411 15A Bluejay AIO

ot0tot, 2023-03-27:
Use the DARWINF4SX1280HD target, not the BETAFPV one.

haslinghuis, 2023-03-27:
Try DARWINF4SX1280HD

haslinghuis, 2023-03-27:
Lols @ot0tot :)

closed 2023-03-28 by Arby5577