betaflight/betaflight #12589: Missing "numeric climb rate" in OSD ( BF 4.4)

NODFips, 2023-03-28:
### Describe the bug

I can activate the "climb rate" icon in OSD tab. But I have no OSD element in the FPV feed..... and in the diff  there is no entry like "set osd_nvario_pos = 1234". 

### To Reproduce

 I flashed back to 4.3.2 it works! Flashed 4.4 again.... not working :-(

### Expected behavior

..having "numeric climb rate"

### Support ID

```txt
fe3f97a3-b4f7-4b12-8b61-c649c8085789
```

### Flight controller

Airbot Omnibus F4V6

haslinghuis, 2023-03-28:
Please enable show release candidates and enable expert mode. Add `VARIO` in custom defines field before reflashing.

NODFips, 2023-03-28:
Thanks ... works! Now, I'm happy ;-)

closed 2023-03-28 by NODFips

dereksynkopa, 2023-09-22:
Continuing the topic. Let me ask you, how to calibrate numeric climb rate? In my Mark5 F722 the values end at 14.8 m/s which is definitely not true.