betaflight/betaflight #12606: Third led on the fc starts to flicker when using gps

TheIsotopes, 2023-04-02:
### Describe the bug

For a few days i have a GPS module for testing. Everything seems to be working well, except that when the GPS is switched on, the third LED on the FC starts to flicker after a while and doesn't stop.
If i deactivate GPS it stops flickering and does not start again, but after turning GPS on again, does the flickering starts again.

tested on MATEKH722MINI and MATEKH743

### To Reproduce

using latest master

### Expected behavior

I don't know if this is a bug, but i think so.

### Support ID

```txt
Id: "db743958-7c80-4c35-b483-2f15b5ebc760"
```

### Flight controller

MATEKH722MINI, MATEKH743

### Other components

MATEK GNSS SAM-M10Q

TheIsotopes, 2023-06-07:
With the current master https://github.com/betaflight/betaflight/commit/7d7b6596b0839c1047704d29ae2a69aec015bd3e the problem seems to be solved.
The third LED stops flashing ... no idea why, but it's ok.

TheIsotopes, 2023-06-13:
@haslinghuis pls can you reopen this issue ... since #12886 its back

blckmn, 2023-06-13:
@TheIsotopes https://build.betaflight.com/api/support/db743958-7c80-4c35-b483-2f15b5ebc760 shows there is no pin defined for LED 3, i.e. `resource led 3 none`.

Do you know what pin is supposed to be defined for the third LED? Do you have a photo of the board?

TheIsotopes, 2023-06-13:
@blckmn updated first post with new support id

blckmn, 2023-06-13:
LED 2 is tied to PA13 (which is also the SWDIO debug pin). Will keep looking.

TheIsotopes, 2023-07-14:
nothing new at the moment

closed 2026-09-26 by blckmn