betaflight/betaflight #12607: No barometer and no blackbox for target airF7

Tonynux, 2023-04-02:
### Describe the bug

After installing BF 4.4.0 I can't see barometer on Configurator. Also I can't set a SPIFLASH Blackbox. 

### To Reproduce

Just flashing 4.4.0 and configuring everything ready to fly,  the issue is present.

### Expected behavior

Barometer working and SPIFLASH Blackbox working. 

### Support ID

```txt
ee1dbbca-7e41-4af4-ac98-58653bc7491c
```

### Flight controller

Racerstar/Airbot AirF7

haslinghuis, 2023-04-02:
@Tonynux please check http://betaflight.com/docs/wiki/configurator/firmware-flasher-tab#board-defines

We need the support ID after flashing core only build.

Tonynux, 2023-04-03:
> @Tonynux please check http://betaflight.com/docs/wiki/configurator/firmware-flasher-tab#board-defines
> We need the support ID after flashing core only build.

OK. Sorry! 
I need some time. I will re-flash as soon as possible.
Thank you.

haslinghuis, 2023-04-03:
No need to be sorry as core build contains all drivers and information needed for support.

Tonynux, 2023-04-06:
> @Tonynux please check http://betaflight.com/docs/wiki/configurator/firmware-flasher-tab#board-defines
> We need the support ID after flashing core only build.

New Support ID (BF 4.4.1 Core only build) is:  17915166-31e4-4ed5-ac6a-3acfb67c66c8

closed 2023-04-10 by blckmn