betaflight/betaflight #12614: Spektrum serial Rx telemetry not working with BF 4.4.

MEBeans, 2023-04-03:
### Describe the bug

I have Spektrum Rx SPM4650.  AIO FC = HAKRC722D. Tx = DX8e
Telemetry does not work at all on BF 4.4. Telemetry works just fine on BF 4.3.2. 

### To Reproduce

Load firmware with configuration specifics called out:
Radio protocol = SRXL2
Telemetry protocol = SRXL
Serial Rx provider = Spektrum SRXL2 
Transmitter = DX8e

### Expected behavior

Expected all telemetry to be functional, as it is with BF 4.3.2.

### Support ID

```txt
version
# Betaflight / STM32F7X2 (S7X2) 4.4.0 Apr  3 2023 / 01:24:08 (4605309d8) MSP API: 1.45

# config: YES
# board: manufacturer_id: HARC, board_name: HAKRCF722D

# status
MCU F722 Clock=216MHz, Vref=3.29V, Core temp=55degC
Stack size: 2048, Stack address: 0x20010000
Configuration: CONFIGURED, size: 3641, max available: 16384
Devices detected: SPI:1, I2C:1
Gyros detected: gyro 1 locked dma
GYRO=ICM42688P, ACC=ICM42688P, BARO=DPS310
OSD: MAX7456 (30 x 13)
BUILD KEY: 7547bfbc370ee541dee99b533d3efb5b (4.4.0)
System Uptime: 230 seconds, Current Time: 2023-04-03T01:43:44.887+00:00
CPU:33%, cycle time: 123, GYRO rate: 8130, RX rate: 91, System rate: 10
Voltage: 1100 * 0.01V (3S battery - OK)
I2C Errors: 0
FLASH: JEDEC ID=0x00ef4018 16M
Arming disable flags: CLI MSP

# flash_info
Flash sectors=256, sectorSize=65536, pagesPerSector=256, pageSize=256, totalSize=16777216 JEDEC ID=0x00ef4018
Partitions:
  0: FLASHFS   0 255
FlashFS size=16777216, usedSize=0

# dump hardware

# version
# Betaflight / STM32F7X2 (S7X2) 4.4.0 Apr  3 2023 / 01:24:08 (4605309d8) MSP API: 1.45

# config: YES

# start the command batch
batch start

board_name HAKRCF722D
manufacturer_id HARC

# resources
resource BEEPER 1 C13
resource MOTOR 1 B04
resource MOTOR 2 B05
resource MOTOR 3 B00
resource MOTOR 4 B01
resource MOTOR 5 A15
resource MOTOR 6 B03
resource MOTOR 7 B06
resource MOTOR 8 B07
resource LED_STRIP 1 A08
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 A01
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 C07
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 NONE
resource SERIAL_RX 12 NONE
resource I2C_SCL 1 B08
resource I2C_SCL 2 NONE
resource I2C_SCL 3 NONE
resource I2C_SCL 4 NONE
resource I2C_SDA 1 B09
resource I2C_SDA 2 NONE
resource I2C_SDA 3 NONE
resource I2C_SDA 4 NONE
resource LED 1 A14
resource LED 2 A13
resource LED 3 NONE
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 3 C10
resource SPI_SCK 4 NONE
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MISO 3 C11
resource SPI_MISO 4 NONE
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 B15
resource SPI_MOSI 3 C12
resource SPI_MOSI 4 NONE
resource ADC_BATT 1 C02
resource ADC_RSSI 1 C00
resource ADC_CURR 1 C01
resource ADC_EXT 1 A04
resource BARO_CS 1 NONE
resource BARO_EOC 1 NONE
resource BARO_XCLR 1 NONE
resource PINIO 1 C08
resource PINIO 2 C09
resource PINIO 3 NONE
resource PINIO 4 NONE
resource USB_MSC_PIN 1 NONE
resource FLASH_CS 1 D02
resource OSD_CS 1 B12
resource GYRO_EXTI 1 C04
resource GYRO_EXTI 2 C03
resource GYRO_CS 1 B02
resource GYRO_CS 2 C15
resource USB_DETECT 1 C14
resource PULLUP 1 NONE
resource PULLUP 2 NONE
resource PULLUP 3 NONE
resource PULLUP 4 NONE
resource PULLDOWN 1 NONE
resource PULLDOWN 2 NONE
resource PULLDOWN 3 NONE
resource PULLDOWN 4 NONE

# timer
timer B04 AF2
# pin B04: TIM3 CH1 (AF2)
timer B05 AF2
# pin B05: TIM3 CH2 (AF2)
timer B00 AF2
# pin B00: TIM3 CH3 (AF2)
timer B01 AF2
# pin B01: TIM3 CH4 (AF2)
timer A15 AF1
# pin A15: TIM2 CH1 (AF1)
timer B03 AF1
# pin B03: TIM2 CH2 (AF1)
timer B06 AF2
# pin B06: TIM4 CH1 (AF2)
timer B07 AF2
# pin B07: TIM4 CH2 (AF2)
timer A08 AF1
# pin A08: TIM1 CH1 (AF1)
timer A03 AF3
# pin A03: TIM9 CH2 (AF3)
timer A02 AF3
# pin A02: TIM9 CH1 (AF3)
timer A01 AF2
# pin A01: TIM5 CH2 (AF2)
timer A00 AF2
# pin A00: TIM5 CH1 (AF2)
timer B10 AF1
# pin B10: TIM2 CH3 (AF1)

# dma
dma SPI_MOSI 1 NONE
dma SPI_MOSI 2 NONE
dma SPI_MOSI 3 0
# SPI_MOSI 3: DMA1 Stream 5 Channel 0
dma SPI_MOSI 4 NONE
dma SPI_MISO 1 NONE
dma SPI_MISO 2 NONE
dma SPI_MISO 3 NONE
dma SPI_MISO 4 NONE
dma SPI_TX 1 NONE
dma SPI_TX 2 NONE
dma SPI_TX 3 0
# SPI_TX 3: DMA1 Stream 5 Channel 0
dma SPI_TX 4 NONE
dma SPI_RX 1 NONE
dma SPI_RX 2 NONE
dma SPI_RX 3 NONE
dma SPI_RX 4 NONE
dma ADC 1 0
# ADC 1: DMA2 Stream 0 Channel 0
dma ADC 2 NONE
dma ADC 3 NONE
dma UART_TX 1 NONE
dma UART_TX 2 NONE
dma UART_TX 3 NONE
dma UART_TX 4 NONE
dma UART_TX 5 NONE
dma UART_TX 6 NONE
dma UART_TX 7 NONE
dma UART_TX 8 NONE
dma UART_RX 1 NONE
dma UART_RX 2 NONE
dma UART_RX 3 NONE
dma UART_RX 4 NONE
dma UART_RX 5 NONE
dma UART_RX 6 NONE
dma UART_RX 7 NONE
dma UART_RX 8 NONE
dma pin B04 0
# pin B04: DMA1 Stream 4 Channel 5
dma pin B05 0
# pin B05: DMA1 Stream 5 Channel 5
dma pin B00 0
# pin B00: DMA1 Stream 7 Channel 5
dma pin B01 0
# pin B01: DMA1 Stream 2 Channel 5
dma pin A15 0
# pin A15: DMA1 Stream 5 Channel 3
dma pin B03 0
# pin B03: DMA1 Stream 6 Channel 3
dma pin B06 0
# pin B06: DMA1 Stream 0 Channel 2
dma pin B07 0
# pin B07: DMA1 Stream 3 Channel 2
dma pin A08 2
# pin A08: DMA2 Stream 3 Channel 6
dma pin A03 NONE
dma pin A02 NONE
dma pin A01 0
# pin A01: DMA1 Stream 4 Channel 6
dma pin A00 0
# pin A00: DMA1 Stream 2 Channel 6
dma pin B10 0
# pin B10: DMA1 Stream 1 Channel 3

# feature
feature -RX_PPM
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -RANGEFINDER
feature -TELEMETRY
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -OSD
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -RX_SPI
feature -ESC_SENSOR
feature -ANTI_GRAVITY
feature RX_SERIAL
feature TELEMETRY
feature LED_STRIP
feature OSD
feature AIRMODE
feature ANTI_GRAVITY

# serial
serial 20 1 115200 57600 0 115200
serial 0 0 115200 57600 0 115200
serial 1 64 115200 57600 0 115200
serial 2 0 115200 57600 0 115200
serial 3 0 115200 57600 0 115200
serial 5 0 115200 57600 0 115200

# master
set gyro_to_use = FIRST
set baro_bustype = I2C
set baro_spi_device = 0
set baro_i2c_device = 1
set baro_i2c_address = 0
set adc_device = 1
set blackbox_device = SPIFLASH
set dshot_burst = OFF
set dshot_bitbang = AUTO
set dshot_bitbang_timer = AUTO
set current_meter = ADC
set battery_meter = ADC
set vbat_scale = 110
set ibata_scale = 400
set beeper_inversion = ON
set beeper_od = OFF
set beeper_frequency = 0
set max7456_clock = NOMINAL
set max7456_spi_bus = 2
set max7456_preinit_opu = OFF
set led_inversion = 0
set pinio_config = 1,1,1,1
set pinio_box = 40,41,255,255
set usb_msc_pin_pullup = ON
set flash_spi_bus = 3
set gyro_1_bustype = SPI
set gyro_1_spibus = 1
set gyro_1_i2cBus = 0
set gyro_1_i2c_address = 0
set gyro_1_sensor_align = CW180
set gyro_1_align_roll = 0
set gyro_1_align_pitch = 0
set gyro_1_align_yaw = 1800
set gyro_2_bustype = SPI
set gyro_2_spibus = 1
set gyro_2_i2cBus = 0
set gyro_2_i2c_address = 0
set gyro_2_sensor_align = CW90
set gyro_2_align_roll = 0
set gyro_2_align_pitch = 0
set gyro_2_align_yaw = 900
set i2c1_pullup = OFF
set i2c1_clockspeed_khz = 800
set i2c2_pullup = OFF
set i2c2_clockspeed_khz = 800
set i2c3_pullup = OFF
set i2c3_clockspeed_khz = 800
set mco2_on_pc9 = OFF
set scheduler_relax_rx = 25
set scheduler_relax_osd = 25
set box_user_1_name = -
set box_user_2_name = -
set box_user_3_name = -
set box_user_4_name = -

# end the command batch
batch end

# 76ed9f96-6a48-4e44-ad5d-118b2cfa28f3
```

### Flight controller

HAKRC722D, purchased from GetFPV

### Other components

none other than Rx.
Firmware of Rx is latest, 2.4.

### How are the different components wired up (including port information)

Rx to Tx with three wire connection, signal from Rx connected to UART2 T2.
Connection to transmit pin as called out by Spektrum for their serial Rx.

### Add any other context about the problem that you think might be relevant here

All control channels are functional on BF 4.4, just no telemetry.

haslinghuis, 2023-05-04:
Does it work with core build ?

themeadery, 2023-05-27:
I have the same exact problem AGAIN after going from my custom build of 4.4.0-RC4 to 4.4.1. I have an extremely similar setup as the OP. Telemetry to my DX8e is broken.

We tackled the problem in #12163 and after a couple PR's it was fixed, but I think it somehow got broken in the official release of 4.4.0 and 4.4.1. IIRC, it stemmed back to a rewrite of the DSHOT code and something wasn't being calculated in the right order, thus the telemetry link had incorrect values (or none) to send back to the radio.

Can we please look at this again. I want to fly 4.4.1, not a custom hack.

I'm using the cloud build feature to include SRXL2 as my radio protocol and SRXL as my telemetry protocol. The configurator changed in the UI a bit. Is "core build" the same as "include everything" like it was in the past? Or is it fully stripped? I could try "core build" and then we know it's the cloud build system screwing up.

haslinghuis, 2023-05-27:
Core build includes all drivers but no features and should be used for support only.

themeadery, 2023-07-12:
In my recent testing: 4.4.0 and above cloud build and full build from github releases breaks telemetry on DSMX/SRXL2. On top of that 4.4.1/4.4.2 are completely broken on my FC. So for me I've been squeezed out by software made for more modern FC's, yet my FC isn't very old. If you are looking for telemetry back to your Spektrum TX from a Spektrum DSMX/SRXL2 RX I would suggest 4.3 and the only way I got it working past that was very custom hacks of 4.4 with manual builds of github pulls of the main branch in very specific timeframes (commits). YMMV, good luck.

MEBeans, 2023-07-12:
Thank you for the input, Matt 😊
I decided that since Horizon Hobby got out of the drone game, likely due to
the FAA getting involved, I'll just settle on 4.3. It is providing enough
to do the job with my Spektrum DX8e for now, and I am not yet highly versed
enough in the Betaflight tuning game. I would like to be able to make use
of the baro sensors on my FCs, and I think the only way to get that will be
to get away from Spektrum for my drones.

 I fly fixed wing as well, and Horizon is now all about fixed wing, so the
DX8 still has a home there. I'll get Radiomaster for my drones when cash
flow allows.

themeadery, 2023-07-12:
Yep. I'm moving to Radiomaster when I can afford it. I guess I'll use ELRS as the link protocol. My Spektrum DX8e was cool for a while, but it's clear the drone world doesn't care about Spektrum and it's more for the fixed wing side like you said. I guess the battle bots guys like them too.

Prajwal-Koirala, 2023-12-26:
I can only provide short summaries. Here is the summary of the article:

This is an article about Spektrum serial Rx telemetry not working with Betaflight 4.4. It discusses a bug where Spektrum Rx telemetry does not work with Betaflight 4.4, but works fine with Betaflight 4.3.2. The user has a Spektrum Rx SPM4650 and a DX8e transmitter. They have tried loading the firmware with specific configuration settings, but the telemetry still does not work. Other users have reported the same problem. The issue seems to be related to a rewrite of the DSHOT code in Betaflight 4.4. A workaround is to use Betaflight 4.3, or a custom build of 4.4 with manual builds of github pulls of the main branch in very specific timeframes.

I hope this is helpful!

CjlaBuk, 2024-04-13:
Having the same issue with my Tinyhawk 2. Can't go 4.4 or any further without losing telemetry

haslinghuis, 2024-04-13:
Please provide support ID here - can be generated on the CLI tab of configurator.

CjlaBuk, 2024-04-13:
> Please provide support ID here - can be generated on the CLI tab of configurator.

44bf1d35-50c9-4436-a720-edfcfd7fc761

haslinghuis, 2024-04-13:
@CjlaBuk That will not work. You have selected SBUS with SmartPort in the build while you have selected SPEKTRUM1024 as protocol in configuration.

CjlaBuk, 2024-04-13:
> @CjlaBuk That will not work. You have selected SBUS with SmartPort in the build while you have selected SPEKTRUM1024 as protocol in configuration.

SPEKTRUM1024 was the selected telemetry provider in 4.3.2 an it worked great. It's an Spi de receiver inetgrated in the boat of the Tinyhawk Frsky protocol.

haslinghuis, 2024-04-13:
@CjlaBuk From 4.4 you will have to select the radio protocol and telemetry - and user options while building the firmware - otherwise it will not work.

CjlaBuk, 2024-04-13:
> @CjlaBuk From 4.4 you will have to select the radio protocol and telemetry - and user options while building the firmware - otherwise it will not work.

What options should I select then?

haslinghuis, 2024-04-13:
Probably something like

[image]

But on the website it says:

[image]

CjlaBuk, 2024-04-13:
> Probably something like
> 
> ![image](https://private-user-images.githubusercontent.com/8344830/322222540-7882a42a-a9d2-49a1-867c-5755e7f0d075.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTMwNDYwODUsIm5iZiI6MTcxMzA0NTc4NSwicGF0aCI6Ii84MzQ0ODMwLzMyMjIyMjU0MC03ODgyYTQyYS1hOWQyLTQ5YTEtODY3Yy01NzU1ZTdmMGQwNzUucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDQxMyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA0MTNUMjIwMzA1WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZWVjNjVlODU4NWU4ZWNhZTFhNGZiOTI3NTUxMWY3NzVmODZiZjcyZGU1YjQ0NGNjZjZlOTRiYjg4MmVmOTg1YyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ._L_s9_Dr-e4kLraN3TxZ9SAbPrnTbOYrlgUs67o7280)
> 
> But on the website it says:
> 
> ![image](https://private-user-images.githubusercontent.com/8344830/322222574-a4a98f3f-57e5-4f2a-8106-86301ee7256e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTMwNDYwODUsIm5iZiI6MTcxMzA0NTc4NSwicGF0aCI6Ii84MzQ0ODMwLzMyMjIyMjU3NC1hNGE5OGYzZi01N2U1LTRmMmEtODEwNi04NjMwMWVlNzI1NmUucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDQxMyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA0MTNUMjIwMzA1WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YWVlZTY1NWM5MjhhMmUyNDRiYThiMDJjMWY5ZmU3YmM5MzIzN2M3ZTgzMWJlNjI3ZmU3OWEwNjdiYzQ0ZDc3MiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.eTKmi4sBzHjiIqx1kupdREo7neswDRjmOwdV08SJe9M)
> 
> EDIT: As the target defines RX_SPI_FRSKY_X as default protocol.

in case of RX_SPI_FRSKY_X, what should i choose as Radio protocol?

CjlaBuk, 2024-04-13:
4.5 SBUS (probably wrong) as protocol and FRSKY_HUB as telemetry

Id: "824f8613-a40b-4b6a-82ac-292f3ea720df"
[image]

CjlaBuk, 2024-04-13:
same 4.5 SBUS (probably wrong) as protocol and FRSKY_HUB as telemetry d8 (FRSKY_D) binded receiver
[image]

haslinghuis, 2024-04-13:
@CjlaBuk If indeed using SBUS should use SMARTPORT for telemetry

CjlaBuk, 2024-04-13:
Weird behavior is that if i bind my receiver directly in _D model ill not get any telemetry from BF. Yet if i bind it in _X mode (both BF and Radio) and then rebind it in _D mode i get everything
 i see this in CLI when its working:

get serialr
serialrx_provider = SBUS
Allowed values: NONE, SPEK2048, SBUS, SUMD, SUMH, XB-B, XB-B-RJ01, IBUS, JETIEXBUS, CRSF, SRXL, CUSTOM, FPORT, SRXL2, GHST, SPEK1024

serialrx_inverted = OFF
Allowed values: OFF, ON

serialrx_halfduplex = OFF
Allowed values: OFF, ON

get tlm
tlm_inverted = OFF
Allowed values: OFF, ON

tlm_halfduplex = ON
Allowed values: OFF, ON

pid_in_tlm = OFF
Allowed values: OFF, ON

haslinghuis, 2024-08-12:
@MEBeans - please check #13814

closed 2024-08-14 by haslinghuis

MEBeans, 2024-09-30:
I am now trying to build a new cinewhoop. It has a Diatone Mamba FC, target MAMBAF22_2022B. The cloud build fails when specifying the #13814 commit with 4.6.0.
The cloud build fails when when simply attempting 4.6.0 without the commit.

haslinghuis, 2024-09-30:
The commit has been merged into 4.6.0 so no need to include 13814. Also target is missing 7
Target should be `MAMBAF722_2022B`

[image]

MEBeans, 2024-09-30:
[image]
Don't know why it's failing. Other firmwares load just fine.

MEBeans, 2024-09-30:
[image]
This is the end of the failed build log. Don't know if this is readable or helps.

nerdCopter, 2024-09-30:
undefined reference to `spektrumBind`...
add `SERIALRX_SPEKTRUM` to custom defines and it succeeds.

@haslinghuis is SERIALRX_SPEKTRUM a prerequisite of SERIALRX_SPEKTRUM? 
or does `spektrumBind` code need gating?

haslinghuis, 2024-09-30:
`USE_TELEMETRY_SRXL` requires `USE_SERIALRX_SPEKTRUM`  as it seems to be shared between SPEKTRUM, SRXL and SRXL2 protocol.

haslinghuis, 2024-09-30:
@MEBeans on 4.6.0 dev please use 
[image]

MEBeans, 2024-09-30:
Took the quad on its maiden flight in my room just now. Going back and forth here. Thank you, very much for the time you spend here!! 
I have a lot to learn here, but I am very curious! I'll be trying #13947 shortly.

MEBeans, 2024-10-01:
Tried 4.6.0 with the suggested commit. I cannot connect to the flight control now. Attempting to retore my flight control....
[image]

MEBeans, 2024-10-01:
The web version is 11.0.0, but you cannot download it. The latest available on github is 10.10.0. Interesting.
Is it being migrated to web only?

nerdCopter, 2024-10-01:
@MEBeans, yes, but can be "installed" for "offline" use.
[image]

MEBeans, 2024-10-01:
What needs to be learned for working with the code the way you folks do?

PR #13814 (2024-08-14): Fix telemetry for Spektrum/SRXL
- fixes #12614