betaflight/betaflight #12631: BF 4.4.1 - Holybro Kakute H7 won't boot up after flashing

x8n1620, 2023-04-06:
### Describe the bug

After flashing BF 4.4.1 on my Kakute H7 (old Version / v1) the FC won't boot up.
I can't connect with the Configurator.
[image]

After pressing the Boot-Button and downgrade 4.4.0, connect works fine.
I have try this multiple times.

I try the stable Configurator and the Nightly Build betaflight-configurator_10.10.0-debug-397aba0_amd64 on my Ubuntu Workstation.
I als try another PC (Win11) with stable Configurator
I try also to flash the Core Only Firmware 4.4.1, same.
[image]

### To Reproduce

FC: Kakute H7 
Configurator: betaflight-configurator_10.10.0-debug-397aba0_amd64
BF Upgrade: 4.4.0 -> 4.4.1
Flash with full chip Erease

### Expected behavior

FC boots up also with 4.4.1, connect with the configurator possible

### Support ID

```txt
###

# # Problem description

# # Can't flash to 4.4.1, FC won't boot up. I have to reflash 4.4.0

# ###

# version
# Betaflight / STM32H743 (SH74) 4.4.0 Jan 31 2023 / 16:17:06 (4605309d8) MSP API: 1.45

# config: YES
# board: manufacturer_id: HBRO, board_name: KAKUTEH7

# status
MCU H743 (Rev.V) Clock=480MHz, Vref=3.30V, Core temp=49degC
Stack size: 2048, Stack address: 0x2001fff8
Configuration: CONFIGURED, size: 4248, max available: 131072
Devices detected: SPI:1, I2C:2
Gyros detected: gyro 1 locked dma
GYRO=MPU6000, ACC=MPU6000, BARO=BMP280, MAG=QMC5883
OSD: MAX7456 (30 x 13)
BUILD KEY: 765a942c9f3dddf76eba1675a96d4274 (4.4.0)
System Uptime: 50 seconds, Current Time: 2023-04-06T20:51:42.621+00:00
CPU:9%, cycle time: 125, GYRO rate: 8000, RX rate: 15, System rate: 9
Voltage: 0 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 7
SD card: Manufacturer 0x3, 31166976kB, 11/2021, v8.5, 'SK32G'
Filesystem: Ready
Arming disable flags: RXLOSS CLI MSP MOTOR_PROTO

# flash_info
Flash sectors=0, sectorSize=0, pagesPerSector=0, pageSize=0, totalSize=0 JEDEC ID=0x00000000
Partitions:
FlashFS size=0, usedSize=0

# dump hardware

# version
# Betaflight / STM32H743 (SH74) 4.4.0 Jan 31 2023 / 16:17:06 (4605309d8) MSP API: 1.45

# config: YES
[... 360 lines cut ...]
set gyro_2_align_yaw = 0
set i2c1_pullup = OFF
set i2c1_clockspeed_khz = 800
set i2c2_pullup = OFF
set i2c2_clockspeed_khz = 800
set i2c3_pullup = OFF
set i2c3_clockspeed_khz = 800
set i2c4_pullup = OFF
set i2c4_clockspeed_khz = 800
set scheduler_relax_rx = 25
set scheduler_relax_osd = 25
set box_user_1_name = -
set box_user_2_name = -
set box_user_3_name = -
set box_user_4_name = -

# end the command batch
batch end

#
```

### Flight controller

Holybro Kakute H7

HThuren, 2023-04-06:
PS. just make this 4.5, and it boot well https://build.betaflight.com/api/builds/26b7aa20aba9976521316b28714134bb/log

x8n1620, 2023-04-07:
> PS. just make this 4.5, and it boot well https://build.betaflight.com/api/builds/26b7aa20aba9976521316b28714134bb/log

Is there any tutorial, how i can build and flash BF 4.5?

x8n1620, 2023-04-07:
I have flashed to 4.5, the FC boots up.
Would be nice to have a stable 4.4 release on my quad. So i go temporary down to 4.4.0.

Martivip, 2023-04-07:
I confirm my KakuteH7 don't boot as well after BF 4.4.1 flash

RupesF, 2023-04-08:
Same here, for my KakuteH7's - four of them.

Sounds like this is a result of the same bug reported in #12472. This was fixed in the master branch, but I don't believe the fix went into 4.4-maintenance too.

culhane, 2023-04-08:
Same here, for Lumenier Lux H7 HD. 
https://github.com/betaflight/betaflight/issues/12638

flysixty, 2023-04-10:
Same here, for iFlight SucceX-D Mini F7 TwinG.

RupesF, 2023-04-14:
Hi,

Wanted to provide a bit of feedback after some testing - hopefully this is at least partially helpful.

Building a custom hex from from head of the 4.4-maintenance branch after @haslinghuis committed last night (aa228ea) now allows me to flash successfully - FC comes back up after the flash no problems. However, I'm not prompted to apply custom defaults - so nothing is recognized really, and after a restart of the FC or load of a prior backup renders the FC un-connectable.

If I download the "custom defaults" from the unified-targets repo and manually apply that, the FC becomes fully functional - albeit with a blank config. And, if I edit the backup file to remove any of the "defaults nosave" lines at the top of the backup, and then apply that, I get a fully working FC with restored backup. Obviously subsequent backup files / diffs contain warnings about no custom defaults found and full resource allocation parameters - but not the end of the world by any means.

Thanks!

RupesF, 2023-04-14:
Apologies - I fear I was being an idiot re: custom defaults.

Tried flashing again with configurator set to releases only, and not development, presented the custom defaults as expected. Therefore issues with this FC appear to be completely resolved in aa228ea.

haslinghuis, 2023-04-14:
Fixed in #12671

closed 2023-04-14 by haslinghuis