betaflight/betaflight #12634: HGLRC F722 barometer not working

aiye400, 2023-04-07:
### Describe the bug

Updated HGLRC F722 Flight controller and barometer no longer shows up

### To Reproduce

update to Betaflight 4.4.1 and barometer is no longer working

### Expected behavior

working barometer

### Support ID

```txt
# version
# Betaflight / STM32F7X2 (S7X2) 4.4.1 Apr  6 2023 / 07:38:36 (e43d591b2) MSP API: 1.45

# config: YES
# board: manufacturer_id: HGLR, board_name: HGLRCF722

# status
MCU F722 Clock=216MHz, Vref=3.26V, Core temp=51degC
Stack size: 2048, Stack address: 0x20010000
Configuration: CONFIGURED, size: 3644, max available: 16384
Devices detected: SPI:1, I2C:0
Gyros detected: gyro 1 locked dma shared
GYRO=MPU6000, ACC=MPU6000
OSD: MSP (60 x 22)
BUILD KEY: 8e871fa9cf8093428ed9ce9d3cf8f7c3 (4.4.1)
System Uptime: 18 seconds, Current Time: 2023-04-07T09:16:37.459+00:00
CPU:51%, cycle time: 125, GYRO rate: 8000, RX rate: 15, System rate: 9
Voltage: 21 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
FLASH: JEDEC ID=0x00ef4018 16M
Arming disable flags: RXLOSS CLI MSP RPMFILTER

# flash_info
Flash sectors=256, sectorSize=65536, pagesPerSector=256, pageSize=256, totalSize=16777216 JEDEC ID=0x00ef4018
Partitions:
  0: FLASHFS   0 255
FlashFS size=16777216, usedSize=16777216

# dump hardware

# version
# Betaflight / STM32F7X2 (S7X2) 4.4.1 Apr  6 2023 / 07:38:36 (e43d591b2) MSP API: 1.45

# config: YES

# start the command batch
batch start

board_name HGLRCF722
manufacturer_id HGLR

# resources
resource BEEPER 1 C13
resource MOTOR 1 B04
resource MOTOR 2 B05
resource MOTOR 3 B00
resource MOTOR 4 B01
resource MOTOR 5 A15
resource MOTOR 6 B03
resource MOTOR 7 B06
resource MOTOR 8 B07
resource LED_STRIP 1 A08
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 A01
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 C07
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 NONE
resource SERIAL_RX 12 NONE
resource I2C_SCL 1 B08
resource I2C_SCL 2 NONE
resource I2C_SCL 3 NONE
resource I2C_SCL 4 NONE
resource I2C_SDA 1 B09
resource I2C_SDA 2 NONE
resource I2C_SDA 3 NONE
resource I2C_SDA 4 NONE
resource LED 1 A14
resource LED 2 A13
resource LED 3 NONE
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 3 C10
resource SPI_SCK 4 NONE
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MISO 3 C11
resource SPI_MISO 4 NONE
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 B15
resource SPI_MOSI 3 C12
resource SPI_MOSI 4 NONE
resource ADC_BATT 1 C02
resource ADC_RSSI 1 C00
resource ADC_CURR 1 C01
resource ADC_EXT 1 NONE
resource BARO_CS 1 A04
resource BARO_EOC 1 NONE
resource BARO_XCLR 1 NONE
resource PINIO 1 C08
resource PINIO 2 C09
resource PINIO 3 NONE
resource PINIO 4 NONE
resource USB_MSC_PIN 1 NONE
resource FLASH_CS 1 D02
resource OSD_CS 1 B12
resource GYRO_EXTI 1 C04
resource GYRO_EXTI 2 NONE
resource GYRO_CS 1 B02
resource GYRO_CS 2 NONE
resource USB_DETECT 1 C14
resource PULLUP 1 NONE
resource PULLUP 2 NONE
resource PULLUP 3 NONE
resource PULLUP 4 NONE
resource PULLDOWN 1 NONE
resource PULLDOWN 2 NONE
resource PULLDOWN 3 NONE
resource PULLDOWN 4 NONE

# timer
timer B04 AF2
# pin B04: TIM3 CH1 (AF2)
timer B05 AF2
# pin B05: TIM3 CH2 (AF2)
timer B00 AF2
# pin B00: TIM3 CH3 (AF2)
timer B01 AF2
# pin B01: TIM3 CH4 (AF2)
timer A15 AF1
# pin A15: TIM2 CH1 (AF1)
timer B03 AF1
# pin B03: TIM2 CH2 (AF1)
timer B06 AF2
# pin B06: TIM4 CH1 (AF2)
timer B07 AF2
# pin B07: TIM4 CH2 (AF2)
timer A08 AF1
# pin A08: TIM1 CH1 (AF1)
timer A03 AF3
# pin A03: TIM9 CH2 (AF3)
timer A02 AF3
# pin A02: TIM9 CH1 (AF3)
timer A01 AF2
# pin A01: TIM5 CH2 (AF2)
timer A00 AF2
# pin A00: TIM5 CH1 (AF2)

# dma
dma SPI_MOSI 1 NONE
dma SPI_MOSI 2 NONE
dma SPI_MOSI 3 NONE
dma SPI_MOSI 4 NONE
dma SPI_MISO 1 NONE
dma SPI_MISO 2 NONE
dma SPI_MISO 3 NONE
dma SPI_MISO 4 NONE
dma SPI_TX 1 NONE
dma SPI_TX 2 NONE
dma SPI_TX 3 NONE
dma SPI_TX 4 NONE
dma SPI_RX 1 NONE
dma SPI_RX 2 NONE
dma SPI_RX 3 NONE
dma SPI_RX 4 NONE
dma ADC 1 1
# ADC 1: DMA2 Stream 4 Channel 0
dma ADC 2 NONE
dma ADC 3 NONE
dma UART_TX 1 NONE
dma UART_TX 2 NONE
dma UART_TX 3 NONE
dma UART_TX 4 NONE
dma UART_TX 5 NONE
dma UART_TX 6 NONE
dma UART_TX 7 NONE
dma UART_TX 8 NONE
dma UART_RX 1 NONE
dma UART_RX 2 NONE
dma UART_RX 3 NONE
dma UART_RX 4 NONE
dma UART_RX 5 NONE
dma UART_RX 6 NONE
dma UART_RX 7 NONE
dma UART_RX 8 NONE
dma pin B04 0
# pin B04: DMA1 Stream 4 Channel 5
dma pin B05 0
# pin B05: DMA1 Stream 5 Channel 5
dma pin B00 0
# pin B00: DMA1 Stream 7 Channel 5
dma pin B01 0
# pin B01: DMA1 Stream 2 Channel 5
dma pin A15 0
# pin A15: DMA1 Stream 5 Channel 3
dma pin B03 0
# pin B03: DMA1 Stream 6 Channel 3
dma pin B06 0
# pin B06: DMA1 Stream 0 Channel 2
dma pin B07 0
# pin B07: DMA1 Stream 3 Channel 2
dma pin A08 2
# pin A08: DMA2 Stream 3 Channel 6
dma pin A03 NONE
dma pin A02 NONE
dma pin A01 0
# pin A01: DMA1 Stream 4 Channel 6
dma pin A00 0
# pin A00: DMA1 Stream 2 Channel 6

# feature
feature -RX_PPM
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -RANGEFINDER
feature -TELEMETRY
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -OSD
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -RX_SPI
feature -ESC_SENSOR
feature -ANTI_GRAVITY
feature RX_SERIAL
feature GPS
feature TELEMETRY
feature OSD
feature AIRMODE
feature ESC_SENSOR
feature ANTI_GRAVITY

# serial
serial 20 1 115200 57600 0 115200
serial 0 2 115200 57600 0 115200
serial 1 64 115200 57600 0 115200
serial 2 0 115200 57600 0 115200
serial 3 131073 115200 57600 0 115200
serial 5 1024 115200 57600 0 115200

# master
set gyro_to_use = FIRST
set baro_bustype = SPI
set baro_spi_device = 1
set baro_i2c_device = 0
set baro_i2c_address = 0
set adc_device = 1
set blackbox_device = SPIFLASH
set dshot_burst = ON
set dshot_bitbang = AUTO
set dshot_bitbang_timer = AUTO
set current_meter = ADC
set battery_meter = ADC
set vbat_scale = 110
set ibata_scale = 179
set beeper_inversion = ON
set beeper_od = OFF
set beeper_frequency = 0
set max7456_clock = NOMINAL
set max7456_spi_bus = 2
set max7456_preinit_opu = OFF
set led_inversion = 0
set pinio_config = 1,1,1,1
set pinio_box = 40,41,255,255
set usb_msc_pin_pullup = ON
set flash_spi_bus = 3
set gyro_1_bustype = SPI
set gyro_1_spibus = 1
set gyro_1_i2cBus = 0
set gyro_1_i2c_address = 0
set gyro_1_sensor_align = CW180FLIP
set gyro_1_align_roll = 0
set gyro_1_align_pitch = 1800
set gyro_1_align_yaw = 1800
set gyro_2_bustype = SPI
set gyro_2_spibus = 0
set gyro_2_i2cBus = 0
set gyro_2_i2c_address = 0
set gyro_2_sensor_align = CW0
set gyro_2_align_roll = 0
set gyro_2_align_pitch = 0
set gyro_2_align_yaw = 0
set i2c1_pullup = OFF
set i2c1_clockspeed_khz = 800
set i2c2_pullup = OFF
set i2c2_clockspeed_khz = 800
set i2c3_pullup = OFF
set i2c3_clockspeed_khz = 800
set mco2_on_pc9 = OFF
set scheduler_relax_rx = 25
set scheduler_relax_osd = 25
set box_user_1_name = -
set box_user_2_name = -
set box_user_3_name = -
set box_user_4_name = -

# end the command batch
batch end

*** Support data submitted *** 
Id: "e00ad69b-1d4f-4dc0-9c65-95f0c7cfdbf3"

# copy ID and provide to the betaflight team
```

### Flight controller

HGLRC F722

### Other components

Original Air Unit, BN-180 GPS unit, crossfire nano

### How are the different components wired up (including port information)

GPS and RX are soldered on, Air Unit is connected via plug.

GPS is UART 1

RX is UART 2

Air Unit is UART 4

### Add any other context about the problem that you think might be relevant here

I found the same issue with Betaflight 4.3 and the HGLRC F722, fixed by #11215 , however putting in the command lines 

“
set baro_bustype = SPI
set baro_spi_device = 1
set baro_i2c_device = 0
save
”

still does not give the barometer back.

https://github.com/betaflight/betaflight/pull/11215

haslinghuis, 2023-04-07:
You can use HGLRCF722E target

aiye400, 2023-04-07:
thanks for the quick reply, I just reflashed f722e and after applying custom defaults, I got the warnings regarding motor output protocol. The Baro was yellow on the top status bar, but unfortunately F722E firmware causes my betaflight to freeze within 3-5 seconds of connecting. The board also goes from flashing red/blue lights to solid red/blue lights and I cannot change any settings in that time frame. BF even freezes while in DFU mode

haslinghuis, 2023-04-07:
Please check: https://betaflight.com/docs/wiki/configurator/firmware-flasher-tab#troubeshooting

haslinghuis, 2023-04-07:
If it did work in 4.4.0 with HGLRCF722 then there could be another problem.

aiye400, 2023-04-07:
Hi sorry for not being clearer earlier, I updated from 4.2.8, and the baro was working correctly on 4.2.8. As for HGLRCF722E target, it still continues to freeze/lock up after 3-5 seconds. I have restarted my computer, swapped USB cables, swapped USB ports, making sure nothing else is plugged into the computer, and I even disconnected all of my peripherals, down to the naked board, and it still freezes.  When it freezes the red and blue lights go solid, and it does that even if I don't connect to betaflight configurator. Luckily the ImpulseRC driver fixer will allow it to load into DFU mode and I can reflash the firmware HGLRCF722 and get the board working again. Unfortunately if I use the ImpulseRC driver fixer, once I disconnect the USB and try to plug it back in to reconnect to betaflight configurator, it freezes within 3-5 seconds again

aiye400, 2023-04-07:
OK so I just downgraded to HGLRC F722 4.4.0, and still no barometer, even after running the 
" set baro_bustype = SPI
set baro_spi_device = 1
set baro_i2c_device = 0
save
”

However, I am able to run firmware HGLRC F722E 4.4.0, and it **does** have the barometer and does not seem to freeze the board. I still have no peripherals soldered on so I will put it back to flying condition and see if it still allows it to connect without freezing.

So for now it seems that for HGLRC F722 version 4.4.0 and 4.4.1 my FC will not have the barometer regardless of changing the baro to SPI, and also something about HGLRCF722E only on 4.4.1 causes the FC to lock up after 3-5 seconds.

SteveCEvans, 2023-04-07:
Looking at https://github.com/betaflight/unified-targets/blob/master/configs/default/HGLR-HGLRCF722.config we have:

```
set baro_bustype = I2C
set baro_i2c_device = 1
```

aiye400, 2023-04-07:
> Looking at https://github.com/betaflight/unified-targets/blob/master/configs/default/HGLR-HGLRCF722.config we have:
> 
> ```
> set baro_bustype = I2C
> set baro_i2c_device = 1
> ```

yeah thats what the board originally ships with, but for some reason in later versions of betaflight, at least sometime after 4.2.8 it seems we have to change it to 
``` set baro_bustype = SPI
set baro_spi_device = 1
set baro_i2c_device = 0
save
```

as per https://github.com/betaflight/betaflight/pull/11215 this thread

aiye400, 2023-04-07:
Also, I hooked everything back up and it seems like target HGLRC F722E works on 4.4.0, all peripherals being read and sending data, something about HGLRC F722E on 4.4.1 causes the board to lock up. For now I'll keep it on F722E 4.4.0

haslinghuis, 2023-04-07:
HGLRCF722 uses `I2C` baro driver + settings
HGLRCF722E uses `SPI` baro driver + settings

aiye400, 2023-04-07:
> HGLRCF722 uses `I2C` baro driver + settings HGLRCF722E uses `SPI` baro driver + settings

ah okay that makes sense. Looks like maybe just an issue with HGLRCF722E on 4.4.1

ctausendfreund, 2023-04-09:
I have the same problem as @aiye400.

I have the HGLRC Zeus F722 Mini DJI HD (MPU6000 and SPI BMP280). On 4.3.x and 4.4.0 the barometer is working, but on 4.4.1 the FC freezes after a couple of seconds and no communication is possible. If I disable the baro on 4.4.1 it's not freezing.

TommyV6, 2023-04-12:
I got HGLRC F722 mini v1 my fc freezing after trying to open any tab in bf-configurator on HGLRCF722E firmware.

benlumley, 2023-04-13:
I get the same as above.

I've previously used the HGLRCF722 target (4.3). 

Using that target on 4.4.1 it had no baro.
Using HGLRCF722E on 4.4.1 it has baro, but freezes. Disable baro and no freezing.

(A dump from 4.3 shows that the baro was on SPI before; so based on above I switched to HGLRCF722E)

<details>
<summary>4.3 diff</summary>

```
batch start

# reset configuration to default settings
defaults nosave

board_name HGLRCF722
manufacturer_id HGLR
mcu_id 0014002a4e30500d20343135
signature 

# resources
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 B04
resource MOTOR 4 B05

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature TELEMETRY
feature OSD

# serial
serial 3 1 115200 57600 0 115200

# beacon
beacon RX_LOST
beacon RX_SET

# aux
aux 0 0 0 1500 2100 0 0
aux 1 1 1 1250 2100 0 0
aux 2 13 2 900 1325 0 0
aux 3 35 3 900 1225 0 0

# master
set gyro_lpf1_static_hz = 375
set gyro_lpf2_static_hz = 750
set dyn_notch_count = 2
set dyn_notch_min_hz = 125
set dyn_notch_max_hz = 850
set gyro_lpf1_dyn_min_hz = 375
set gyro_lpf1_dyn_max_hz = 750
set acc_trim_roll = -1
set acc_calibration = -46,28,3,1
set baro_bustype = SPI
set baro_spi_device = 1
set baro_i2c_device = 0
set rc_smoothing_auto_factor = 140
set rc_smoothing_setpoint_cutoff = 25
set rc_smoothing_feedforward_cutoff = 25
set serialrx_provider = CRSF
set dshot_bidir = ON
set vbat_scale = 112
set small_angle = 180
set simplified_gyro_filter_multiplier = 150
set report_cell_voltage = ON
[... 223 lines cut ...]
set rates_type = BETAFLIGHT
set roll_rc_rate = 127
set pitch_rc_rate = 127
set yaw_rc_rate = 100
set roll_expo = 40
set pitch_expo = 40
set roll_srate = 72
set pitch_srate = 72
set yaw_srate = 75

rateprofile 1

rateprofile 2

rateprofile 3

# restore original rateprofile selection
rateprofile 0

# save configuration
save
```
</details>

can get any other info or try stuff if needed (@ me) - thanks!

TommyV6, 2023-04-13:
I flashed with hglrc722e 4.4.1 and fc is freezing. But on 4.4.0 everything works. I got 3 hglrc f7 fc all have the same problem. Two are 30x30 one is 20x20 all are V1 with MPU6000 and BMP280 on 4.3 and 4.2 my diff is 
set baro_bustype = SPI
set baro_spi_device = 1
set baro_i2c_device = 0
and it works.

haslinghuis, 2023-04-13:
Also check 4..5 development version please

MomoBrut, 2023-04-14:
I have the exact same problem, HGLRCF722E always freezes on version 4.4.1. If the baro is deactivated, the FC works normally, the moment it activates, the FC freezes.

I just tried version 4.5.zulu and FC doesn't freeze, it doesn't freeze in all other older versions I've tried, it only freezes in release 4.4.1.

# status
MCU F722 Clock=216MHz, Vref=3.28V, Core temp=53degC
Stack size: 2048, Stack address: 0x20010000
Configuration: CONFIGURED, size: 3290, max available: 16384
Devices detected: SPI:2, I2C:0
Gyros detected: gyro 1 locked dma shared
GYRO=MPU6000, ACC=MPU6000, BARO=BMP280
OSD: MSP (60 x 22)
BUILD KEY: 17d116af74e65f1a473af06e3b6da85f (4.5.0-zulu)
System Uptime: 444 seconds, Current Time: 2023-04-14T04:21:43.319+00:00
CPU:56%, cycle time: 125, GYRO rate: 8000, RX rate: 15, System rate: 9
Voltage: 3 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
FLASH: JEDEC ID=0x00ef4017 8M
Arming disable flags: RXLOSS THROTTLE NOPREARM CLI MSP GPS RPMFILTER

porlock, 2023-04-15:
I have exact same problem

On BF 4.4.1
When i flash target HGLRCF722E it causing FC freeze after powering up.
When i flash target HGLRCF722 Barometer dont work.

I have 3 PCS of this FC, every of them act same way.

To Reproduce
Flash BF 4.4.1 target HGLRCF722E and notice that FC freeze after powering/connecting to BF configurator
Flash BF 4.4.1 target HGLRCF722 and notice that barometer don't work

Expected behavior
Stable and working BF 4.4.1 for HGLRC Zeus F722 Mini MPU6000 version, with barometer support.

Support ID
For HGLRCF722

*** Support data submitted *** 
Id: "ea67be18-1218-4f0b-ae63-67c3e9429843"

I was not able to dump config for target HGLRCF722E, because of freeze issue
Flight controller
HGLRC Zeus F722 Mini MPU6000 version

haslinghuis, 2023-04-15:
```
make HGLRCF722E EXTRA_FLAGS="-D'BUILD_KEY=9b5cc52e82fd68943a646c0be0431474' -D'RELEASE_NAME=4.4.1' -DCLOUD_BUILD -DUSE_DSHOT -DUSE_GPS -DUSE_GPS_PLUS_CODES -DUSE_LED_STRIP -DUSE_OSD -DUSE_OSD_HD -DUSE_OSD_SD -DUSE_PINIO -DUSE_SERIALRX -DUSE_SERIALRX_CRSF -DUSE_TELEMETRY -DUSE_TELEMETRY_CRSF -DUSE_VTX"
```

Please test:
[betaflight_4.4.2_STM32F7X2_HGLRCF722E.zip](https://github.com/betaflight/betaflight/files/11240016/betaflight_4.4.2_STM32F7X2_HGLRCF722E.zip)

MomoBrut, 2023-04-15:
Well, it's freezing, but in a different way. At one point it lasts for a few minutes and then clicks on another configurator tab and freezes. Couldn't it be the 10.10.0 configurator?

And it seems that it has something to do with temperature, when the core is warmer, it freezes in moments ...

Status:
MCU F722 Clock=216MHz, Vref=3.28V, Core temp=53degC
Stack size: 2048, Stack address: 0x20010000
Configuration: CONFIGURED, size: 3644, max available: 16384
Devices detected: SPI:2, I2C:0
Gyros detected: gyro 1 locked dma shared
GYRO=MPU6000, ACC=MPU6000, BARO=BMP280
OSD: MSP (60 x 22)
BUILD KEY: 9b5cc52e82fd68943a646c0be0431474 (4.4.1)
System Uptime: 17 seconds, Current Time: 2023-04-15T17:04:40.405+00:00
CPU:55%, cycle time: 124, GYRO rate: 8064, RX rate: 15, System rate: 9
Voltage: 3 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
FLASH: JEDEC ID=0x00ef4017 8M
Arming disable flags: RXLOSS THROTTLE NOPREARM CLI MSP GPS RPMFILTER

MomoBrut, 2023-04-15:
After few minutes of testing, when it is warmer, it freezes instantly, same as with standard release 4.4.1. ...

haslinghuis, 2023-04-15:
Is this 10.9 or 10.10 nightly configurator? (There are portable builds so they don't have to be installed)

[betaflight_4.4.2_STM32F7X2_HGLRCF722E.zip](https://github.com/betaflight/betaflight/files/11240181/betaflight_4.4.2_STM32F7X2_HGLRCF722E.zip)

Please provide also support ID from CLI.

MomoBrut, 2023-04-15:
I do not think it is caused by configurator. It seems there is some kind of collisions on SPI. When I disable the Gyro on SPI and Baro stays on SPI, it is not freezing.

MCU F722 Clock=216MHz, Vref=3.28V, Core temp=54degC
Stack size: 2048, Stack address: 0x20010000
Configuration: CONFIGURED, size: 3644, max available: 16384
Devices detected: SPI:1, I2C:0
Gyros detected: gyro 1 locked
, BARO=BMP280
OSD: MSP (60 x 22)
BUILD KEY: 9b5cc52e82fd68943a646c0be0431474 (4.4.1)
System Uptime: 36 seconds, Current Time: 2023-04-15T17:31:19.383+00:00
CPU:0%, cycle time: 0, GYRO rate: 0, RX rate: 0, System rate: 9
Voltage: 4 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
FLASH: JEDEC ID=0x00ef4017 8M
Arming disable flags: NOGYRO BOOTGRACE CLI MSP

haslinghuis, 2023-04-15:
Third try:

[betaflight_4.4.2_STM32F7X2_HGLRCF722E.zip](https://github.com/betaflight/betaflight/files/11240246/betaflight_4.4.2_STM32F7X2_HGLRCF722E.zip)

MomoBrut, 2023-04-15:
The same ...

ctausendfreund, 2023-04-15:
I also don't think that it's caused by the configurator. The FC freezes also after a few secs when I power up my drone without the configurator as long as the baro is activated.

MomoBrut, 2023-04-16:
Has anyone else tried the test firmwares? For me, no change on any of them, with the activated barometer it still freezes. Does anyone have a different experience?

martin-korinek-1985, 2023-04-16:
I had the same problem, you need o selecet target HGLRCF722E, without the E baro does not work.
And be aware BF 4.4.1 causes me freezing of FC, so I recommend to use 4.4.0

LoakAudio, 2023-04-16:
> Has anyone else tried the test firmwares? For me, no change on any of them, with the activated barometer it still freezes. Does anyone have a different experience?

Hi, I own an HGLRC F722 mini wired to nothing. I can confirm 4.4.0 works, 4.4.1 freezes, last 4.4.2 given by  @haslinghuis freezes and 4.5 works

porlock, 2023-04-16:
I noticed something more disturbing, after i flashed 4.4.1 and freezing issue occurred I get back to 4.3.2 with full chip erase. Baro start works again, it was fine for some time, i started to PID tuning my quad, i took a few hours and suddenly happened same like on 4.4.1, FC frez once, and since that it frez every time few sec after reboot. When i disabling baro magically works again (without baro of course), no idea what to think about really, does BF 4.4.1 did something "permanently" that cripple HGLRCF722 or maybe newest BF configurator does that.

haslinghuis, 2023-04-16:
https://github.com/betaflight/betaflight/issues/12634#issuecomment-1509889375 provides build with ARM SDK 9.

MomoBrut, 2023-04-17:
Today I tested 4.5.0 zulu in flight, I have got some problems with losing and appearing GPS module, l mean, the GPS module completely disappears from OSD with all parameters related to GPS for about a minute, than suddenly appears again, very strange and scary behavior on LR quad. Fix, please, the current release version for HGLRCF722E target. The 4.5.0 is not the way, at least for now, it is not polished.

haslinghuis, 2023-04-17:
[betaflight_4.4.2_STM32F7X2_HGLRCF722E.zip](https://github.com/betaflight/betaflight/files/11256448/betaflight_4.4.2_STM32F7X2_HGLRCF722E.zip)

MomoBrut, 2023-04-18:
Could you please comment those two versions more? None of them detect either gyro or baro.

# resource SPI_MISO 1 A06
###ERROR IN resource: INVALID RESOURCE NAME: 'SPI_MISO'###

# resource SPI_MISO 2 B14
###ERROR IN resource: INVALID RESOURCE NAME: 'SPI_MISO'###

# resource SPI_MISO 3 C11
###ERROR IN resource: INVALID RESOURCE NAME: 'SPI_MISO'###

# resource SPI_MISO 4 NONE
###ERROR IN resource: INVALID RESOURCE NAME: 'SPI_MISO'###

# resource SPI_MOSI 1 A07
###ERROR IN resource: INVALID RESOURCE NAME: 'SPI_MOSI'###

# resource SPI_MOSI 2 B15
###ERROR IN resource: INVALID RESOURCE NAME: 'SPI_MOSI'###

# resource SPI_MOSI 3 C12
###ERROR IN resource: INVALID RESOURCE NAME: 'SPI_MOSI'###

# resource SPI_MOSI 4 NONE
###ERROR IN resource: INVALID RESOURCE NAME: 'SPI_MOSI'###

# dma SPI_MOSI 1 NONE
###ERROR IN dma: BAD DEVICE: SPI_MOSI###

# dma SPI_MOSI 2 NONE
###ERROR IN dma: BAD DEVICE: SPI_MOSI###

# dma SPI_MOSI 3 NONE
###ERROR IN dma: BAD DEVICE: SPI_MOSI###

# dma SPI_MOSI 4 NONE
###ERROR IN dma: BAD DEVICE: SPI_MOSI###

# dma SPI_MISO 1 NONE
###ERROR IN dma: BAD DEVICE: SPI_MISO###

# dma SPI_MISO 2 NONE
###ERROR IN dma: BAD DEVICE: SPI_MISO###

# dma SPI_MISO 3 NONE
###ERROR IN dma: BAD DEVICE: SPI_MISO###

# dma SPI_MISO 4 NONE
###ERROR IN dma: BAD DEVICE: SPI_MISO###

porlock, 2023-04-18:
> 🤞
> 
> Local ref: Change Resources: [betaflight_4.4.2_STM32F7X2_HGLRCF722E.zip](https://github.com/betaflight/betaflight/files/11256448/betaflight_4.4.2_STM32F7X2_HGLRCF722E.zip)
> 
> Local ref: Fix Baro SPI + change resources: [betaflight_4.4.2_STM32F7X2_HGLRCF722E.zip](https://github.com/betaflight/betaflight/files/11256481/betaflight_4.4.2_STM32F7X2_HGLRCF722E.zip)

On both firmware same effect, no gyro, no accl, no baro, no flash
i also tries to manually load unified target settings but that change nothing, but i get following errors.

```
# resource PPM 1 A03
###ERROR IN resource: INVALID RESOURCE NAME: 'PPM'###

# resource PWM 1 A02
###ERROR IN resource: INVALID RESOURCE NAME: 'PWM'###

# resource PWM 2 A01
###ERROR IN resource: INVALID RESOURCE NAME: 'PWM'###

# resource PWM 3 A00
###ERROR IN resource: INVALID RESOURCE NAME: 'PWM'###

# resource SPI_MISO 1 A06
###ERROR IN resource: INVALID RESOURCE NAME: 'SPI_MISO'###

# resource SPI_MISO 2 B14
###ERROR IN resource: INVALID RESOURCE NAME: 'SPI_MISO'###

# resource SPI_MISO 3 C11
###ERROR IN resource: INVALID RESOURCE NAME: 'SPI_MISO'###

# resource SPI_MOSI 1 A07
###ERROR IN resource: INVALID RESOURCE NAME: 'SPI_MOSI'###

# resource SPI_MOSI 2 B15
###ERROR IN resource: INVALID RESOURCE NAME: 'SPI_MOSI'###

# resource SPI_MOSI 3 C12
###ERROR IN resource: INVALID RESOURCE NAME: 'SPI_MOSI'###

# set mag_bustype = I2C
###ERROR IN set: INVALID NAME: mag_bustype = I2C###

# set mag_i2c_device = 1
###ERROR IN set: INVALID NAME: mag_i2c_device = 1###
```

MomoBrut, 2023-04-19:
> Previous two builds were just guess work. Reverted to 4.4.0 and added 4.4.1 related PR's on top without config updates:
> 
> [betaflight_4.4.0_STM32F7X2.zip](https://github.com/betaflight/betaflight/files/11266475/betaflight_4.4.0_STM32F7X2.zip)
> 
> make STM32F7X2 EXTRA_FLAGS="-D'BUILD_KEY=04d63849d791c0fac5854fccba1765d1' -D'RELEASE_NAME=4.4.0' -D'BOARD_NAME=HGLRCF722E' -D'MANUFACTURER_ID=HGLR' -DCLOUD_BUILD -DUSE_ACC -DUSE_ACC_SPI_MPU6000 -DUSE_BARO -DUSE_BARO_SPI_BMP280 -DUSE_DSHOT -DUSE_FLASH -DUSE_FLASH_W25Q128FV -DUSE_GPS -DUSE_GPS_PLUS_CODES -DUSE_GYRO -DUSE_GYRO_SPI_MPU6000 -DUSE_LED_STRIP -DUSE_MAX7456 -DUSE_OSD -DUSE_OSD_HD -DUSE_OSD_SD -DUSE_PINIO -DUSE_SERIALRX -DUSE_SERIALRX_CRSF -DUSE_TELEMETRY -DUSE_TELEMETRY_CRSF -DUSE_VTX"

This version does not freeze. Will the firmware be available via the standard way through the configurator? I have one more request, can you add VARIO to the build? Thank you.

MomoBrut, 2023-04-19:
> > Previous two builds were just guess work. Reverted to 4.4.0 and added 4.4.1 related PR's on top without config updates:
> > [betaflight_4.4.0_STM32F7X2.zip](https://github.com/betaflight/betaflight/files/11266475/betaflight_4.4.0_STM32F7X2.zip)
> > make STM32F7X2 EXTRA_FLAGS="-D'BUILD_KEY=04d63849d791c0fac5854fccba1765d1' -D'RELEASE_NAME=4.4.0' -D'BOARD_NAME=HGLRCF722E' -D'MANUFACTURER_ID=HGLR' -DCLOUD_BUILD -DUSE_ACC -DUSE_ACC_SPI_MPU6000 -DUSE_BARO -DUSE_BARO_SPI_BMP280 -DUSE_DSHOT -DUSE_FLASH -DUSE_FLASH_W25Q128FV -DUSE_GPS -DUSE_GPS_PLUS_CODES -DUSE_GYRO -DUSE_GYRO_SPI_MPU6000 -DUSE_LED_STRIP -DUSE_MAX7456 -DUSE_OSD -DUSE_OSD_HD -DUSE_OSD_SD -DUSE_PINIO -DUSE_SERIALRX -DUSE_SERIALRX_CRSF -DUSE_TELEMETRY -DUSE_TELEMETRY_CRSF -DUSE_VTX"
> 
> After flashing this firmware, loading unified defaults for HGLRCF722E shows flowing errors
> 
> ```
> # resource PPM 1 A03
> ###ERROR IN resource: INVALID RESOURCE NAME: 'PPM'###
> 
> # resource PWM 1 A02
> ###ERROR IN resource: INVALID RESOURCE NAME: 'PWM'###
> 
> # resource PWM 2 A01
> ###ERROR IN resource: INVALID RESOURCE NAME: 'PWM'###
> 
> # resource PWM 3 A00
> ###ERROR IN resource: INVALID RESOURCE NAME: 'PWM'###
> 
> # set mag_bustype = I2C
> ###ERROR IN set: INVALID NAME: mag_bustype = I2C###
> 
> # set mag_i2c_device = 1
> ###ERROR IN set: INVALID NAME: mag_i2c_device = 1###
> ```
> 
> It's doesn't hang, baro, gyro and flash are available.. but baro readings are useless
> 
> Local flashed firmware (same for 4.4,0 downloaded by BF config) [image]
> 
> 4.3.2 [image]
> 
> So i guess right barometer implementation for this board is on 4.3.2, on 4.4.0 it was already something crooked, 4.4.1 make it just worse.

Baro works perfectly normally, it's not a bug, it's a feature. As of version 4.4.0, the baro has been completely rewritten. As long as the quad is not armed, only the unfiltered barometer signal is always available, in sensors or in debug. The barometer really starts to work only after arming the quad, when its low pass filters are activated, one is for ALTITUDE and the other for VARIO, in the CLI it is possible to set their limit frequencies altitude_lpf for ALTITUDE and altitude_d_lpf for VARIO. Remove propellers during testing.

haslinghuis, 2023-04-19:
Added VARIO
```
make STM32F7X2 EXTRA_FLAGS="-D'BUILD_KEY=04d63849d791c0fac5854fccba1765d1' -D'RELEASE_NAME=4.4.0' -D'BOARD_NAME=HGLRCF722E' -D'MANUFACTURER_ID=HGLR' -DCLOUD_BUILD -DUSE_ACC -DUSE_ACC_SPI_MPU6000 -DUSE_BARO -DUSE_BARO_SPI_BMP280 -DUSE_DSHOT -DUSE_FLASH -DUSE_FLASH_W25Q128FV -DUSE_GPS -DUSE_GPS_PLUS_CODES -DUSE_GYRO -DUSE_GYRO_SPI_MPU6000 -DUSE_LED_STRIP -DUSE_MAX7456 -DUSE_OSD -DUSE_OSD_HD -DUSE_OSD_SD -DUSE_PINIO -DUSE_SERIALRX -DUSE_SERIALRX_CRSF -DUSE_TELEMETRY -DUSE_TELEMETRY_CRSF -DUSE_VTX -DUSE_VARIO"
```
[betaflight_4.4.0_STM32F7X2.zip](https://github.com/betaflight/betaflight/files/11272260/betaflight_4.4.0_STM32F7X2.zip)

haslinghuis, 2023-04-19:
All PR's in the 4.4.1 release notes are there except the configuration PR's

porlock, 2023-04-19:
> This version does not freeze. Will the firmware be available via the standard way through the configurator? I have one more request, can you add VARIO to the build? Thank you.

You have a proper baro readings ? because fact is available doesn't means it works in my case.

haslinghuis, 2023-04-19:
On the bench you get raw baro values afaik. When flying readings should be solid.

haslinghuis, 2023-04-19:
@MomoBrut Please also test 4.5 version - so we don't have to repeat the process :smirk:

MomoBrut, 2023-04-20:
Ok, I will, but not until a few days at the earliest, I'm out now. When I tested version 4.5 a few days back, it was solid, no freezing with the barometer.

porlock, 2023-04-20:
i testes it on 4.5.0 don't hangs, gyro, barometer and flash are available.

porlock, 2023-04-21:
@haslinghuis so can we expect that it will work on 4.4.1 :-D ?

MomoBrut, 2023-05-01:
I finally got around to it today, uploading the last 4.5 master. I can confirm that the barometer works perfectly reliably in this firmware version.

moriwaky, 2023-05-03:
> I finally got around to it today, uploading the last 4.5 master. I can confirm that the barometer works perfectly reliably in this firmware version.

did your GPS works fine on bf 4.5 master? mine still disappearing from the OSD and I can´t not get the GPS sensor light "on" on the configurator. And I get continuously gps lost gps recovery messages from telemetry on the radio

MomoBrut, 2023-05-03:
No, it is not working, the GPS configuration issues are not solved yet. And it seems, they are solving the newer M10 modules only, so the problem with older M8s probably persist ...

moriwaky, 2023-05-03:
> No, it is not working, the GPS configuration issues are not solved yet. And it seems, they are solving the newer M10 modules only, so the problem with older M8s probably persist ...

thanks for the info, Do you know if the developers are aware of the M8s gps problems on the HGLRCF722E target? Do we have to make a separate PR to solve? because this threat is relative to a baro problem (sorry if my questions are "noob" but I am pretty new in github and still learning)

haslinghuis, 2023-05-03:
Agree GNSS is another topic (and unrelated to the board configuration) as we are working on some issues.

b4sh, 2023-05-11:
> @porlock Please test firmware 4.4.2 build for HGLRCF722E [betaflight_4.4.2_STM32F7X2.zip](https://github.com/betaflight/betaflight/files/11435089/betaflight_4.4.2_STM32F7X2.zip)
> 

This firmware hangs my FC after few seconds. betaflight_4.4.0_STM32F7X2.zip works ok (small problem, fc hangs after tcp/ip connection via elrs rx).

haslinghuis, 2023-05-11:
@b4sh You can use nightly configurator to fix the tcp issue.

New test firmware:
[betaflight_4.4.2_STM32F7X2.zip](https://github.com/betaflight/betaflight/files/11454015/betaflight_4.4.2_STM32F7X2.zip)
[betaflight_4.4.2_STM32F7X2_VARIO.zip](https://github.com/betaflight/betaflight/files/11454016/betaflight_4.4.2_STM32F7X2_VARIO.zip)

haslinghuis, 2023-05-11:
Like to know if VARIO is causing the issue.

b4sh, 2023-05-11:
betaflight_4.4.2_STM32F7X2_PR12784.zip and betaflight_4.4.2_STM32F7X2_NVIC_SPI_PRIORITY.zip - upgrade, reboot, apply custom defaults, hang, power cycle, apply CLI dump with my working config, reboot and hang. Only DFU via boot button working.

b4sh, 2023-05-11:
@haslinghuis 

betaflight_4.4.2_STM32F7X2_PR12784.zip - hang after few seconds
betaflight_4.4.2_STM32F7X2_PR12784_VARIO.zip - hang after few seconds

betaflight_4.4.2_STM32F7X2_NVIC_SPI_PRIORITY.zip - works on bench.
betaflight_4.4.2_STM32F7X2_NVIC_SPI_PRIORITY_VARIO.zip - works on bench.

haslinghuis, 2023-05-11:
Cool - thanks for conformation !!!

NVIC SPI priority was changed in [ba71dfe](https://github.com/betaflight/betaflight/commit/ba71dfe6d355d5ba056ef2273d56734c7e41c388)

catphish, 2023-05-15:
I just wanted to comment on my experience with this. I have an HGLRCF722, it seems there are several revisions of this board with slight hardware variations. Mine has a barometer on SPI.

First of all I can confirm that running 4.4.1, the board crashes if I enable the baro, and with 4.4.2 the problem appears to be fully resolved by #12795 :+1: 

However, I had a lot of trouble making a build that would work with the baro on my board. Your build works, but nether a HGLRCF722 nor a HGLRCF722E cloud build will detect the baro for me. I wasn't able to get to the bottom of why. In the end I created my own cloud build using the following flags, which worked.

```
make STM32F7X2 EXTRA_FLAGS="-D'RELEASE_NAME=4.4.2-zulu' -D'BOARD_NAME=HGLRCF722' -D'MANUFACTURER_ID=HGLR' -DCLOUD_BUILD -DUSE_ACC_SPI_MPU6000 -DUSE_GYRO_SPI_MPU6000 -DUSE_FLASH_W25Q128FV -DUSE_MAX7456 -DUSE_DSHOT -DUSE_GPS -DUSE_GPS_PLUS_CODES -DUSE_LED_STRIP -DUSE_OSD -DUSE_OSD_HD -DUSE_OSD_SD -DUSE_PINIO -DUSE_SERIALRX -DUSE_SERIALRX_CRSF -DUSE_SERIALRX_DEFAULT -DUSE_TELEMETRY -DUSE_TELEMETRY_CRSF -DUSE_VTX -DUSE_MAG -DUSE_BARO -DUSE_BARO_BMP280 -DUSE_BARO_SPI_BMP280 -DUSE_BARO_BMP388 -DUSE_BARO_SPI_BMP388 -DUSE_FLASH -DUSE_FLASH_W25Q128FV -DUSE_FLASH_M25P16 -DUSE_FLASH_W25M"
```

I'm sure some of those baro and flash options are redundant, but I was getting a little frustrated and casting a wide net to try to get it working.

It's also worth noting that the 9V power switch PINIO doesn't seem to work on my board. I suspect it may be an older version that doesn't even have this switch. Happy to provide any further information if it's helpful.

catphish, 2023-05-15:
Status output:
```
MCU F722 Clock=216MHz, Vref=3.31V, Core temp=35degC
Stack size: 2048, Stack address: 0x20010000
Configuration: CONFIGURED, size: 3672, max available: 16384
Devices detected: SPI:2, I2C:0
Gyros detected: gyro 1 locked dma shared
GYRO=MPU6000, ACC=MPU6000, BARO=BMP280
OSD: MSP (53 x 20)
System Uptime: 38 seconds, Current Time: 2023-05-15T19:20:08.318+00:00
CPU:51%, cycle time: 125, GYRO rate: 8000, RX rate: 15, System rate: 9
Voltage: 58 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
FLASH: JEDEC ID=0x00ef4018 16M
Arming disable flags: RXLOSS CLI MSP RPMFILTER
```

catphish, 2023-05-15:
I'm happy to try the cloud build for HGLRCF722E again. It's unclear why it wouldn't work.

haslinghuis, 2023-05-15:
Yes definite the BMP280 SPI.

4.4.2 has just be updated a minute ago. Might have to add or remove an option to trigger the change otherwise you get a cached version of the build.

[19 more replies]

PR #12795 (2023-05-12): [4.4.2] Revert NVIC SPI priority
Fixes: #12634
closed 2023-05-20 by haslinghuis