betaflight/betaflight #12635: I can no longer connect to the FC IFLIGHT_F722_TWING

simoneenomis, 2023-04-07:
### Describe the bug

once I flashed the betaflight 4.4.1 version I can no longer connect to the FC IFLIGHT_F722_TWING if I try to meat the development version 4.5.0-zulu it works for now I will go back to 4.4.0 which worked correctly waiting for a solution thanks

### To Reproduce

 I can no longer connect to the FC IFLIGHT_F722_TWING

### Expected behavior

 I can no longer connect to the FC IFLIGHT_F722_TWING

### Support ID

```txt
# dump

# version
# Betaflight / STM32F7X2 (S7X2) 4.4.0 Feb  1 2023 / 23:18:44 (4605309d8) MSP API: 1.45

# config: YES

# start the command batch
batch start

board_name IFLIGHT_F722_TWING
manufacturer_id IFRC

# name: iFlight DC5 HD

# resources
resource BEEPER 1 C15
resource MOTOR 1 C08
resource MOTOR 2 C06
resource MOTOR 3 C09
resource MOTOR 4 C07
resource MOTOR 5 B06
resource MOTOR 6 B07
resource MOTOR 7 B01
resource MOTOR 8 B00
resource LED_STRIP 1 A01
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 C10
resource SERIAL_TX 5 C12
resource SERIAL_TX 6 NONE
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 C11
resource SERIAL_RX 5 D02
resource SERIAL_RX 6 NONE
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 NONE
resource SERIAL_RX 12 NONE
resource I2C_SCL 1 NONE
resource I2C_SCL 2 B10
resource I2C_SCL 3 NONE
resource I2C_SCL 4 NONE
resource I2C_SDA 1 NONE
[... 960 lines cut ...]
set quickrates_rc_expo = OFF
set roll_rc_rate = 7
set pitch_rc_rate = 7
set yaw_rc_rate = 7
set roll_expo = 0
set pitch_expo = 0
set yaw_expo = 0
set roll_srate = 67
set pitch_srate = 67
set yaw_srate = 67
set throttle_limit_type = OFF
set throttle_limit_percent = 100
set roll_rate_limit = 1998
set pitch_rate_limit = 1998
set yaw_rate_limit = 1998
set roll_level_expo = 0
set pitch_level_expo = 0

# end the command batch
batch end
```

### Flight controller

IFLIGHT_F722_TWING

baboon9999, 2023-04-07:
Got same issue.
How did you revert to 4.4 as using the boot button to get into dfu is not working.

4stah, 2023-04-07:
I can confirm.
Boot button and DFU work normally.
I had to go back to 4.4.0.

baboon9999, 2023-04-07:
Thanks for the quick reply.
My boot button works but not detected by windows now. Can tell by the leds
Worked before flash.

4stah, 2023-04-07:
well, try to disconnect everything from the board RX, VTX, GPS and so on... and keep pressed boot button for a while when powering on. restart BF configurator... it's working but sometime it requires some gentle fight ;)

baboon9999, 2023-04-09:
@4stah.
Reverted back to 4.40 had to disconnect everything for boot to work.

baboon9999, 2023-04-10:
Definatley happens after applying custom defaults.

haslinghuis, 2023-04-10:
Dual gyro and SDCard users have bad luck with 4.4.1, please keep it on 4.4.0 or (better) try 4.5.0 with nightly build of configurator

YI-BOYANG, 2023-04-12:
> 
Yes, I tried the dual ICM20689 flight control burn 4.4.1 firmware, it did not work properly after importing the configuration, and it returned to normal after downgrade back to 4.4.0.

woxulv, 2023-04-15:
Disconnect all from FC and hold boot button while plugging USB cable and DFU mode should come back to life. 
Tried multiple times with 4.4.1 with and without custom defaults. Without custom defaults you can at least reach Betaflight, but since nothing is configured you are still lost.. and with custom defaults you are in a soft bricked situation..

haslinghuis, 2023-04-18:
make STM32F7X2 EXTRA_FLAGS="-D'BUILD_KEY=6a60517a39fa4343b99b9aed31808d4a' -D'RELEASE_NAME=4.4.0' -D'BOARD_NAME=IFLIGHT_F722_TWING' -D'MANUFACTURER_ID=IFRC' -DCLOUD_BUILD -DUSE_ACC -DUSE_ACC_SPI_ICM20689 -DUSE_BARO -DUSE_BARO_BMP280 -DUSE_BARO_DPS310 -DUSE_DSHOT -DUSE_FLASH -DUSE_FLASH_M25P16 -DUSE_FLASH_W25M512 -DUSE_FLASH_W25N01G -DUSE_GPS -DUSE_GPS_PLUS_CODES -DUSE_GYRO -DUSE_GYRO_SPI_ICM20689 -DUSE_LED_STRIP -DUSE_MAX7456 -DUSE_OSD -DUSE_OSD_HD -DUSE_OSD_SD -DUSE_PINIO -DUSE_SERIALRX -DUSE_SERIALRX_CRSF -DUSE_SERIALRX_DEFAULT -DUSE_SERIALRX_GHST -DUSE_SERIALRX_SBUS -DUSE_TELEMETRY -DUSE_TELEMETRY_CRSF -DUSE_TELEMETRY_GHST -DUSE_VTX"

Please test this firmware:

[betaflight_4.4.x.IFLIGHT_F722_TWING.zip](https://github.com/betaflight/betaflight/files/11266769/betaflight_4.4.x.IFLIGHT_F722_TWING.zip)

haslinghuis, 2023-05-19:
Closing now as fixed in 4.4.2.

Continue in `#12780`

closed 2023-05-19 by haslinghuis