betaflight/betaflight #12636: GEPRCF411 - no flash

nerdCopter, 2023-04-07:
### Describe the bug

* 4.4 support-id aeea6a25-7549-436a-96b1-e84fedd48085 - `JEDEC ID=0x00000000`
* 4.5 support-id 5d63704b-1cc4-4709-8bec-f694603c3add - `flash_info` `###ERROR IN input: UNKNOWN COMMAND, TRY 'HELP'###`
* conversation https://discord.com/channels/868013470023548938/1054918937344745522/1093912247253872730

### To Reproduce

flash 4.4.1 or 4.5-zulu

### Expected behavior

flash defined in some way

### Support ID

```txt
4.4.1 https://build.betaflight.com/api/support/aeea6a25-7549-436a-96b1-e84fedd48085
4.5 https://build.betaflight.com/api/support/5d63704b-1cc4-4709-8bec-f694603c3add
```

### Flight controller

GEPRCF411

haslinghuis, 2023-04-07:
The board does not seem to come with blackbox hardware.

https://geprc.com/product/stable-f411/

nerdCopter, 2023-04-07:
User reports this is for open-lager, not on-board 🤕🤪

closed 2023-04-07 by nerdCopter