betaflight/betaflight #12638: Lumenier H7 Lux HD - USB dead on 4.4.1

culhane, 2023-04-08:
### Describe the bug

I have flashed 4.4.1 via the betaflight firmware updater and the device was no longer detected on usb or device manager after the flash. I have flashed back 4.4.0 via STM32 Cube Programmer and the device was again detected on usb. I again retried flashing 4.4.1 via STM32 Cube Programmer and again the usb was no longer detected in device manager.

### To Reproduce

Flash 4.4.1 via betaflight updater or stm32 cube programmer. The usb will no longer be detected in device manager, along with no "windows device detected" sounds played when connecting.

### Expected behavior

Flashing to 4.4.1 should not kill the usb device.

### Support ID

```txt
N/A (device has been corrupted).
```

### Flight controller

Lumenier H7 Lux HD

haslinghuis, 2023-05-08:
@culhane please confirm it's fixed in 4.5.0 development version.

haslinghuis, 2023-06-01:
Closing as fixed in 4.4.2 release.

closed 2023-06-01 by haslinghuis