betaflight/betaflight #12640: Magnetometer not working with new Walksnail WS-M181 GPS

fabriciopolonio, 2023-04-08:
### Describe the bug

I have replaced on my quad the GPS module from a Matek M8Q-5883 to Walksnail WS-M181, on Matek I can get to enable Magnetometer with no issues, while with Walksnail WS-M181 it simply doesn't work.

I've also tried to use the code from this PR to check if it would solve the issue, but it didn't work:
https://github.com/betaflight/betaflight/pull/12545

Important things to note:

- Wiring is correct, I already checked.
- I tested with two different modules of Walksnail WS-M181 (I bought two units), both has the same issue.
- GPS feature works fine.

### To Reproduce

Tried with version 4.4.0 and 4.5.0, since 4.4.1 is not working with Kakute H7 FC (Bug also reported by other users).

Hardware:
- Kakute H7 FC (V1)
- WS-M181 GPS

- Open Betaflight configurator.
- Try to enable Magnetometer
- Reboot

### Expected behavior

Magnetometer should be enabled correctly

### Support ID

```txt
version
# Betaflight / STM32H743 (SH74) 4.5.0 Apr  8 2023 / 15:11:57 (20755bd75) MSP API: 1.46
# board: manufacturer_id: HBRO, board_name: KAKUTEH7

# status
MCU H743 (Rev.V) Clock=480MHz, Vref=3.28V, Core temp=48degC
Stack size: 2048, Stack address: 0x2001fff8
Configuration: CONFIGURED, size: 4113, max available: 131072
Devices detected: SPI:1, I2C:1
Gyros detected: gyro 1 locked dma
GYRO=MPU6000, ACC=MPU6000, BARO=BMP280
OSD: MSP (30 x 16)
System Uptime: 511 seconds, Current Time: 2023-04-08T13:28:33.396+00:00
CPU:23%, cycle time: 125, GYRO rate: 8000, RX rate: 15, System rate: 9
Voltage: 0 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
SD card: Manufacturer 0x3, 15558144kB, 02/2016, v8.0, 'SL16G'
Filesystem: Ready
Arming disable flags: RXLOSS THROTTLE CLI MSP RESCUE_SW RPMFILTER

# flash_info
###ERROR IN input: UNKNOWN COMMAND, TRY 'HELP'###
# dump hardware

# version
# Betaflight / STM32H743 (SH74) 4.5.0 Apr  8 2023 / 15:11:57 (20755bd75) MSP API: 1.46

# start the command batch
batch start

board_name KAKUTEH7
manufacturer_id HBRO

# resources
resource BEEPER 1 C13
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 B03
[... 327 lines cut ...]
set scheduler_relax_rx = 25
set scheduler_relax_osd = 25
set box_user_1_name = -
set box_user_2_name = -
set box_user_3_name = -
set box_user_4_name = -

# end the command batch
batch end

#
```

### Flight controller

Kakute H7 FC (V1)

### Other components

GPS:
- WS-M181 GPS

### How are the different components wired up (including port information)

GPS module (WS-M181 GPS) is wired on UART 4.

haslinghuis, 2023-04-08:
Did you add Magnetometers in Other Options before flashing 4.5.0 ?
Could you provide support ID?

fabriciopolonio, 2023-04-08:
@haslinghuis Support ID is already on the issue.

The quad had 4.4.0, with working GPS + MAG (Matek M8Q-5883), no issues, when I replaced GPS module by Walksnail one (Walksnail WS-M181), it stopped working MAG, then I updated to 4.5.0, issue persisted, then I applied your PR on top of 4.5.0, same issue persists.

haslinghuis, 2023-04-08:
Add `MAG MAG_QMC5883` to custom defines.
Magnetometers option does not seem to work.

fabriciopolonio, 2023-04-08:
Do you mean adding #define USE_MAG_QMC5883 to my target file (config/KAKUTEH7/config.h)?

haslinghuis, 2023-04-08:
Also needs  #define USE_MAG

devova, 2023-04-09:
Having the same issue on iNAV6, GPS works on serial port, magnetometer - not(
FC `SPEEDYBEEF405V3` has USE_MAG_QMC5883

devova, 2023-04-10:
I wrote the I2C scanner and didn't find any new device after connecting. 100% a hardware issue.

fabriciopolonio, 2023-04-10:
@devova @haslinghuis 

I have contacted Walksnail support, the issue is that they incorrectly printed the sticker with SDA and SCL pinout labels reversed.

Swap these pins and you will get it working, I'll close this ticket since this is not an issue on Betaflight.

I have just confirmed the solution and now everything is working for me.

closed 2023-04-10 by fabriciopolonio

RipperDrone, 2023-10-11:
argh, fell for the same shyte, only found this after hrs of trying :-( Walksnail, watch your pinout / inking!!

haslinghuis, 2023-10-11:
Yes I2C pad's need to be reversed :)

sixtyfive, 2025-12-15:
Worth noting perhaps, for anyone else coming here via Google, that newer speciments have the labelling correct. SDA is supposed to be on the outside.