betaflight/betaflight #12641: GDB causes memory leak, uses high CPU resource and then crashes without error when debugging EXST ELF files.

hydra, 2023-04-08:
### Describe the bug

When debugging the H7RF, or other targets a debugger session with GDB is started, this debugger session loads the symbols from the .ELF file generated by BF as part of the target build, however the .ELF file seems to cause GDB to leak memory, use high CPU resources and then crashes after about 20 seconds or so without any error.

### To Reproduce

* Build with GCC 10.3.1.
* Start JLink debugger GDB server (tested all from 7.54 to latest 7.86g) configured to use GDB 10.3.1 from the same arm tools as the compiler.
* Load the symbols at GDB startup (e.g. `add-symbol-file obj/main/betaflight_STM32H730_SPRACINGH7RF_EXST.elf`)
* Observer that the debugger starts, then crashes.

### Expected behavior

Working debugger.

### Support ID

```txt
N/A
```

### Flight controller

N/A

### Add any other context about the problem that you think might be relevant here

I was able to work around the problem by ONLY changing the version of GDB that the debugger uses to GDB 11.3.1 /without/ recomiling BF with GCC 11.3.1.

i.e.
Compiler/Linker/Etc -> 10.3.1
GDB -> 11.3.1

Example launch configuration from Eclipse:

[image]

I'm posting this here as a reminder that when upgrading the compiler debugging also needs to be tested, and also so that any other devs working with BF are aware of this issue and this work-around.

Often just after a release, i.e. not during a release candidate phase, the tools and libraries for the project are updated, so that devs spend as much time with the new tools during the development cycle of the next release so that issue can be found and resolved.

GCC 10.3.1 is pretty old at this stage and many issues have been fixed.

Is there a plan to upgrade GCC, the latest version is now 12.2?,

Arm GNU Toolchain
Version 12.2.MPACBTI-Rel1
Released: March 22, 2023

https://developer.arm.com/downloads/-/arm-gnu-toolchain-downloads

haslinghuis, 2023-04-08:
Check: https://github.com/betaflight/betaflight/pull/12286

haslinghuis, 2023-04-09:
@hydra did you check?

haslinghuis, 2023-05-10:
Wondering if using ARM SDK GCC 9.3.1 would solve the issue as JETI seems to crash the firmware if compiled with 10.3.1 and works with 9.3.1

haslinghuis, 2025-08-10:
- Fixed in #12971

closed 2025-08-10 by haslinghuis