betaflight/betaflight #12642: Barometer can't be activated on XSRF4O using FRSKYF4 4.4.1 firmware

sybu0, 2023-04-09:
### Describe the bug

Up to 4.3.2 firmware, it was possible to activate barometer with configuration FRSKYF4, the frsky XSRF4O is equiped with a baro BMP280. Should you please activate this option in new firmware? I didn't see any possibility to activate the baro another way.
Best regards.

### To Reproduce

Just try to activate baro after update firmware..., it is not possible..

### Expected behavior

Possibility to activate baro after update firmware (I succeeded using "core update only")

### Support ID

```txt
Id: "26c30f38-e9c4-46d7-a9c1-28d6a61d5963"
```

### Flight controller

XSRF4O from HorusRC

### Other components

GPS,  MATEK VTX

haslinghuis, 2023-04-09:
You can add `BARO BARO_SPI_BMP280` in custom defines before flashing

sybu0, 2023-04-09:
Hello, thanks a lot, I only tried    "BARO_SPI_BMP280"  and  it didn't work, with "BARO BARO_SPI_BMP280", it is OK.

closed 2023-04-10 by blckmn