betaflight/betaflight #12643: revoltosd and OmniNXTF4 flash does not show up on Betaflight 4.4

oyvinla, 2023-04-09:
### Describe the bug

When flashing 4.4.0 or 4.4.1 on the RevoltOSD and the OmniNXTF4 boards, the flash memory does not show up. when flashing back to 4.3.2, the flash memory is available on both boards.

When flashing the Core only to the revoltosd, the flash memory is available, but when I flash the normal 4.4.1 again, the flash memory is not available.

I have tried adding the different flash memory configs like described in this post for the revoldtosd board without any luck
https://github.com/betaflight/betaflight/issues/12082

Thanks

### To Reproduce

Flash BF 4.4

### Expected behavior

Flash Memory for Blackbox should be available

### Support ID

```txt
version
# Betaflight / STM32F405 (S405) 4.4.1 Apr  8 2023 / 14:16:55 (e43d591b2) MSP API: 1.45

# config: YES
# board: manufacturer_id: FLON, board_name: REVOLTOSD

# status
MCU F40X Clock=168MHz (PLLP-HSE), Vref=3.30V, Core temp=40degC
Stack size: 2048, Stack address: 0x1000fff0
Configuration: CONFIGURED, size: 4167, max available: 16384
Devices detected: SPI:1, I2C:0
Gyros detected: gyro 1
GYRO=ICM20602, ACC=ICM20602
OSD: MAX7456 (30 x 13)
BUILD KEY: d0a2b860851005fc19f5922a1d9c3c21 (4.4.1)
System Uptime: 12 seconds, Current Time: 2023-04-09T10:05:04.381+00:00
CPU:43%, cycle time: 125, GYRO rate: 8000, RX rate: 15, System rate: 9
Voltage: 0 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
SD card: Not configured
FLASH: JEDEC ID=0x00ef4018 16M
Arming disable flags: RXLOSS ANGLE CLI MSP MOTOR_PROTO

# flash_info
Flash sectors=256, sectorSize=65536, pagesPerSector=256, pageSize=256, totalSize=16777216 JEDEC ID=0x00ef4018
Partitions:
  0: FLASHFS   0 255
FlashFS size=16777216, usedSize=24576

# dump hardware

# version
# Betaflight / STM32F405 (S405) 4.4.1 Apr  8 2023 / 14:16:55 (e43d591b2) MSP API: 1.45

# config: YES

# start the command batch
batch start

board_name REVOLTOSD
manufacturer_id FLON

# resources
resource BEEPER 1 B04
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 A03
resource MOTOR 4 A02
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 NONE
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource SONAR_TRIGGER 1 NONE
resource SONAR_ECHO 1 NONE
resource LED_STRIP 1 B06
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 NONE
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 NONE
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 A01
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 C07
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 NONE
resource SERIAL_RX 12 NONE
resource INVERTER 1 C00
resource INVERTER 2 NONE
resource INVERTER 3 NONE
resource INVERTER 4 NONE
resource INVERTER 5 NONE
resource INVERTER 6 NONE
resource INVERTER 7 NONE
resource INVERTER 8 NONE
resource INVERTER 9 NONE
resource INVERTER 10 NONE
resource INVERTER 11 NONE
resource INVERTER 12 NONE
resource I2C_SCL 1 B08
resource I2C_SCL 2 NONE
resource I2C_SCL 3 NONE
resource I2C_SDA 1 B09
resource I2C_SDA 2 NONE
resource I2C_SDA 3 NONE
resource LED 1 B05
resource LED 2 NONE
resource LED 3 NONE
resource RX_BIND 1 NONE
resource RX_BIND_PLUG 1 NONE
resource TRANSPONDER 1 NONE
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 3 C10
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MISO 3 C11
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 B15
resource SPI_MOSI 3 C12
resource ESCSERIAL 1 C06
resource ADC_BATT 1 C02
resource ADC_RSSI 1 NONE
resource ADC_CURR 1 C01
resource ADC_EXT 1 NONE
resource BARO_CS 1 NONE
resource BARO_EOC 1 NONE
resource BARO_XCLR 1 NONE
resource COMPASS_CS 1 NONE
resource COMPASS_EXTI 1 NONE
resource SDCARD_CS 1 NONE
resource SDCARD_DETECT 1 NONE
resource PINIO 1 NONE
resource PINIO 2 NONE
resource PINIO 3 NONE
resource PINIO 4 NONE
resource USB_MSC_PIN 1 NONE
resource FLASH_CS 1 B03
resource OSD_CS 1 D02
resource RX_SPI_CS 1 NONE
resource RX_SPI_EXTI 1 NONE
resource RX_SPI_BIND 1 NONE
resource RX_SPI_LED 1 NONE
resource RX_SPI_CC2500_TX_EN 1 NONE
resource RX_SPI_CC2500_LNA_EN 1 NONE
resource RX_SPI_CC2500_ANT_SEL 1 NONE
resource RX_SPI_EXPRESSLRS_RESET 1 NONE
resource RX_SPI_EXPRESSLRS_BUSY 1 NONE
resource GYRO_EXTI 1 C04
resource GYRO_EXTI 2 NONE
resource GYRO_CS 1 A04
resource GYRO_CS 2 NONE
resource USB_DETECT 1 C05
resource VTX_POWER 1 NONE
resource VTX_CS 1 NONE
resource VTX_DATA 1 NONE
resource VTX_CLK 1 NONE
resource PULLUP 1 NONE
resource PULLUP 2 NONE
resource PULLUP 3 NONE
resource PULLUP 4 NONE
resource PULLDOWN 1 NONE
resource PULLDOWN 2 NONE
resource PULLDOWN 3 NONE
resource PULLDOWN 4 NONE

# timer
timer B09 AF3
# pin B09: TIM11 CH1 (AF3)
timer B00 AF3
# pin B00: TIM8 CH2N (AF3)
timer B01 AF3
# pin B01: TIM8 CH3N (AF3)
timer A03 AF1
# pin A03: TIM2 CH4 (AF1)
timer A02 AF1
# pin A02: TIM2 CH3 (AF1)
timer B06 AF2
# pin B06: TIM4 CH1 (AF2)

# dma
dma SPI_MOSI 1 NONE
dma SPI_MOSI 2 NONE
dma SPI_MOSI 3 NONE
dma SPI_MISO 1 NONE
dma SPI_MISO 2 NONE
dma SPI_MISO 3 NONE
dma SPI_TX 1 NONE
dma SPI_TX 2 NONE
dma SPI_TX 3 NONE
dma SPI_RX 1 NONE
dma SPI_RX 2 NONE
dma SPI_RX 3 NONE
dma ADC 1 1
# ADC 1: DMA2 Stream 4 Channel 0
dma ADC 2 NONE
dma ADC 3 NONE
dma UART_TX 1 NONE
dma UART_TX 2 NONE
dma UART_TX 3 NONE
dma UART_TX 4 NONE
dma UART_TX 5 NONE
dma UART_TX 6 NONE
dma UART_TX 7 NONE
dma UART_TX 8 NONE
dma UART_RX 1 NONE
dma UART_RX 2 NONE
dma UART_RX 3 NONE
dma UART_RX 4 NONE
dma UART_RX 5 NONE
dma UART_RX 6 NONE
dma UART_RX 7 NONE
dma UART_RX 8 NONE
dma pin B09 NONE
dma pin B00 0
# pin B00: DMA2 Stream 2 Channel 0
dma pin B01 0
# pin B01: DMA2 Stream 2 Channel 0
dma pin A03 0
# pin A03: DMA1 Stream 7 Channel 3
dma pin A02 0
# pin A02: DMA1 Stream 1 Channel 3
dma pin B06 0
# pin B06: DMA1 Stream 0 Channel 2

# feature
feature -RX_PPM
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -RANGEFINDER
feature -TELEMETRY
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -OSD
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -RX_SPI
feature -ESC_SENSOR
feature -ANTI_GRAVITY
feature RX_SERIAL
feature OSD
feature AIRMODE
feature ANTI_GRAVITY

# serial
serial 20 1 115200 57600 0 115200
serial 0 0 115200 57600 0 115200
serial 2 0 115200 57600 0 115200
serial 3 0 115200 57600 0 115200
serial 5 0 115200 57600 0 115200

# master
set gyro_to_use = FIRST
set align_mag = DEFAULT
set mag_align_roll = 0
set mag_align_pitch = 0
set mag_align_yaw = 0
set mag_bustype = SPI
set mag_i2c_device = 0
set mag_i2c_address = 0
set mag_spi_device = 0
set baro_bustype = SPI
set baro_spi_device = 0
set baro_i2c_device = 0
set baro_i2c_address = 0
set rx_spi_protocol = V202_250K
set rx_spi_bus = 0
set rx_spi_led_inversion = OFF
set adc_device = 1
set blackbox_device = SERIAL
set dshot_burst = ON
set dshot_bitbang = AUTO
set dshot_bitbang_timer = AUTO
set current_meter = VIRTUAL
set battery_meter = NONE
set vbat_scale = 110
set ibata_scale = 400
set beeper_inversion = OFF
set beeper_od = ON
set beeper_frequency = 0
set sdcard_detect_inverted = OFF
set sdcard_mode = OFF
set sdcard_spi_bus = 0
set sdio_clk_bypass = OFF
set sdio_use_cache = OFF
set sdio_use_4bit_width = OFF
set system_hse_mhz = 8
set vtx_spi_bus = 0
set max7456_clock = NOMINAL
set max7456_spi_bus = 2
set max7456_preinit_opu = OFF
set cc2500_spi_chip_detect = ON
set led_inversion = 0
set dashboard_i2c_bus = 1
set dashboard_i2c_addr = 60
set pinio_config = 1,1,1,1
set pinio_box = 255,255,255,255
set usb_msc_pin_pullup = ON
set flash_spi_bus = 2
set gyro_1_bustype = SPI
set gyro_1_spibus = 1
set gyro_1_i2cBus = 0
set gyro_1_i2c_address = 0
set gyro_1_sensor_align = CW0
set gyro_1_align_roll = 0
set gyro_1_align_pitch = 0
set gyro_1_align_yaw = 0
set gyro_2_bustype = SPI
set gyro_2_spibus = 0
set gyro_2_i2cBus = 0
set gyro_2_i2c_address = 0
set gyro_2_sensor_align = CW0
set gyro_2_align_roll = 0
set gyro_2_align_pitch = 0
set gyro_2_align_yaw = 0
set i2c1_pullup = OFF
set i2c1_clockspeed_khz = 800
set i2c2_pullup = OFF
set i2c2_clockspeed_khz = 800
set i2c3_pullup = OFF
set i2c3_clockspeed_khz = 800
set mco2_on_pc9 = OFF
set scheduler_relax_rx = 25
set scheduler_relax_osd = 25
set box_user_1_name = -
set box_user_2_name = -
set box_user_3_name = -
set box_user_4_name = -

# end the command batch
batch end

#
```

### Flight controller

RevoltOSD / OMNINXTF4

oyvinla, 2023-04-09:
version
# Betaflight / STM32F405 (S405) 4.4.1 Apr  9 2023 / 06:39:43 (e43d591b2) MSP API: 1.45

# config: YES
# board: manufacturer_id: AIRB, board_name: OMNINXT4

# status
MCU F40X Clock=168MHz (PLLP-HSE), Vref=3.34V, Core temp=42degC
Stack size: 2048, Stack address: 0x1000fff0
Configuration: CONFIGURED, size: 4167, max available: 16384
Devices detected: SPI:2, I2C:0
Gyros detected: gyro 1, gyro 2 shared
GYRO=MPU6000, ACC=MPU6000, BARO=LPS
OSD: MAX7456 (30 x 13)
BUILD KEY: 00f3b553ac2c0bc8c506e8aede35718b (4.4.1)
System Uptime: 12 seconds, Current Time: 2023-04-09T13:41:01.210+00:00
CPU:41%, cycle time: 125, GYRO rate: 8000, RX rate: 15, System rate: 9
Voltage: 3 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 1
SD card: Not configured
FLASH: JEDEC ID=0x00ef4018 16M
Arming disable flags: RXLOSS CLI MSP MOTOR_PROTO

# flash_info
Flash sectors=256, sectorSize=65536, pagesPerSector=256, pageSize=256, totalSize=16777216 JEDEC ID=0x00ef4018
Partitions:
  0: FLASHFS   0 255
FlashFS size=16777216, usedSize=4421632

# dump hardware

# version
# Betaflight / STM32F405 (S405) 4.4.1 Apr  9 2023 / 06:39:43 (e43d591b2) MSP API: 1.45

# config: YES

# start the command batch
batch start

board_name OMNINXT4
manufacturer_id AIRB

# resources
resource BEEPER 1 C13
resource MOTOR 1 B05
resource MOTOR 2 B04
resource MOTOR 3 B00
resource MOTOR 4 B01
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 B07
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource SONAR_TRIGGER 1 NONE
[... 315 lines cut ...]
set gyro_2_spibus = 1
set gyro_2_i2cBus = 0
set gyro_2_i2c_address = 0
set gyro_2_sensor_align = CW270
set gyro_2_align_roll = 0
set gyro_2_align_pitch = 0
set gyro_2_align_yaw = 2700
set i2c1_pullup = OFF
set i2c1_clockspeed_khz = 800
set i2c2_pullup = OFF
set i2c2_clockspeed_khz = 800
set i2c3_pullup = OFF
set i2c3_clockspeed_khz = 800
set mco2_on_pc9 = OFF
set scheduler_relax_rx = 25
set scheduler_relax_osd = 25
set box_user_1_name = -
set box_user_2_name = -
set box_user_3_name = -
set box_user_4_name = -

# end the command batch
batch end

#

oyvinla, 2023-04-09:
Here is the support data for the omninxt4

version
# Betaflight / STM32F405 (S405) 4.4.1 Apr  9 2023 / 06:39:43 (e43d591b2) MSP API: 1.45

# config: YES
# board: manufacturer_id: AIRB, board_name: OMNINXT4

# status
MCU F40X Clock=168MHz (PLLP-HSE), Vref=3.34V, Core temp=42degC
Stack size: 2048, Stack address: 0x1000fff0
Configuration: CONFIGURED, size: 4167, max available: 16384
Devices detected: SPI:2, I2C:0
Gyros detected: gyro 1, gyro 2 shared
GYRO=MPU6000, ACC=MPU6000, BARO=LPS
OSD: MAX7456 (30 x 13)
BUILD KEY: 00f3b553ac2c0bc8c506e8aede35718b (4.4.1)
System Uptime: 12 seconds, Current Time: 2023-04-09T13:41:01.210+00:00
CPU:41%, cycle time: 125, GYRO rate: 8000, RX rate: 15, System rate: 9
Voltage: 3 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 1
SD card: Not configured
FLASH: JEDEC ID=0x00ef4018 16M
Arming disable flags: RXLOSS CLI MSP MOTOR_PROTO

# flash_info
Flash sectors=256, sectorSize=65536, pagesPerSector=256, pageSize=256, totalSize=16777216 JEDEC ID=0x00ef4018
Partitions:
  0: FLASHFS   0 255
FlashFS size=16777216, usedSize=4421632

# dump hardware

# version
# Betaflight / STM32F405 (S405) 4.4.1 Apr  9 2023 / 06:39:43 (e43d591b2) MSP API: 1.45

# config: YES

# start the command batch
batch start

board_name OMNINXT4
manufacturer_id AIRB

# resources
resource BEEPER 1 C13
resource MOTOR 1 B05
resource MOTOR 2 B04
resource MOTOR 3 B00
resource MOTOR 4 B01
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 B07
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
[... 317 lines cut ...]
set gyro_2_spibus = 1
set gyro_2_i2cBus = 0
set gyro_2_i2c_address = 0
set gyro_2_sensor_align = CW270
set gyro_2_align_roll = 0
set gyro_2_align_pitch = 0
set gyro_2_align_yaw = 2700
set i2c1_pullup = OFF
set i2c1_clockspeed_khz = 800
set i2c2_pullup = OFF
set i2c2_clockspeed_khz = 800
set i2c3_pullup = OFF
set i2c3_clockspeed_khz = 800
set mco2_on_pc9 = OFF
set scheduler_relax_rx = 25
set scheduler_relax_osd = 25
set box_user_1_name = -
set box_user_2_name = -
set box_user_3_name = -
set box_user_4_name = -

# end the command batch
batch end

#

haslinghuis, 2023-04-09:
RevoltOSD: You can add `FLASH FLASH_W25Q128FV` in custom defines before flashing

haslinghuis, 2023-04-09:
Same for OMNINXT4 :)

oyvinla, 2023-04-09:
Thanks, but it doesnt seem to work on either of them. I have tried both `FLASH_W25Q128FV` and `FLASH FLASH_W25Q128FV` but it still doesnt show up. 
[image]

haslinghuis, 2023-04-09:
Enable show release candidates and enable expert mode and try again. That bug is also resolved in latest configurator
https://github.com/betaflight/betaflight-configurator-nightlies/releases

oyvinla, 2023-04-09:
Great, thanks! Got it working :)

oyvinla, 2023-04-10:
BTW, its the same issue for the SynergyF4 board

fireduckfpv, 2023-04-10:
Same error on OMNIBUSF4 board.

haslinghuis, 2023-04-10:
@fireduckfpv OMNIBUSF4 should already work on 4.5 without custom defines

closed 2023-04-10 by blckmn