betaflight/betaflight #12644: Cannot Activate Barometer in BF 4.4.1 on FLYWOOF411

nuhbiwan, 2023-04-09:
### Describe the bug

Barometer cannot be enabled in BF 4.4.1 on FLYWOOF411 board

### To Reproduce

After flashing BF 4.4.1: Go to GUI, enable the (present) Barometer option, save and reboot FC. 

### Expected behavior

Barometer is selected and working

### Support ID

```txt
40946de9-b5b6-4bbd-b85a-e473ffd66aa6
```

### Flight controller

Flywoo GOKU GN 413 13A AIO V2.1 (MPU6000), bought on Flywoo homepage

### Other components

Happymodel EP1 RX 2.4GHz ExpressLRS
Beitian Dual BN-220 GPS GLONASS
DJI O3 Airunit

### How are the different components wired up (including port information)

DJI O3 via Softserial 11 (Motor 5 TX, LED Strip RX)
ELRS RX on UART1
GPS on UART2

### Add any other context about the problem that you think might be relevant here

Previously BF 4.2.9 was flashed. Working without this issue.
Flywoo has several iterations of this board with different HW config, i.e. 20A ESC instead of 13A AND with and without Baro.
The 13A version I am using has the Baro.

haslinghuis, 2023-04-09:
You can add `BARO BARO_BMP280` to custom defines before flashing

nuhbiwan, 2023-04-09:
Solution works. Baro available after adding defines.

closed 2023-04-09 by nuhbiwan