betaflight/betaflight #12646: Talon F7 Fusion FC not recognized after flashing 4.4.1

granolaFPV, 2023-04-09:
### Describe the bug

After flashing to 4.4.1, I can no longer connect to the FC via USB. Reverting to 4.4.0 or earlier via boot button DFU restores normal operation.

[Note that the "Support ID" output below is from a fresh flash back to 4.4.0 since I cannot connect to the FC after flashing to 4.4.1]

### To Reproduce

1. Start with 4.4.0 FW or earlier
2. Open Betaflight Configurator 10.9 (Mac)
3. Connect USB to computer
4. Observe "/dev/tty.usbmodemXXXXXXXX - Betaflight STM32F7x2" in Betaflight connection dropdown
5. Click "Connect" and FC connects as expected
6. Enter Update Firmware mode
7. Flash correct 4.4.1 FW to FC (TALONF7FUSION with default build configuration options) — shows "Programming: SUCCESSFUL"
8. Disconnect and reconnect FC to USB
9. Observe "Configurator has successfully detected and verified the board: TALONF7FUSION" in log
10. Click "Connect"
11. Observe custom defaults dialog — select "Apply Custom Defaults"
12. Observe Betaflight Configurator now only shows "Manual Selection" in dropdown
13. Observe clicking "Connect" doesn't work
14. Observe "Chrome API Error: Failed to connect to the port.." / "Failed to open serial port" in log
15. Activate DFU mode via FC boot button, reflash to 4.4.0 or pervious version, observe normal behavior restored

### Expected behavior

Expect FC to connect via USB.

### Support ID

```txt
version
# Betaflight / STM32F7X2 (S7X2) 4.4.0 Feb  8 2023 / 07:24:52 (4605309d8) MSP API: 1.45

# config: YES
# board: manufacturer_id: HENA, board_name: TALONF7FUSION

# status
MCU F722 Clock=216MHz, Vref=3.29V, Core temp=59degC
Stack size: 2048, Stack address: 0x20010000
Configuration: CONFIGURED, size: 3627, max available: 16384
Devices detected: SPI:2, I2C:0
Gyros detected: gyro 1, gyro 2 locked dma shared
GYRO=ICM20602, ACC=ICM20602
OSD: MAX7456 (30 x 13)
BUILD KEY: 4f0a4c9b0b2142e073bc5683d4caf12c (4.4.0)
System Uptime: 32 seconds, Current Time: 2023-04-09T15:33:44.520+00:00
CPU:33%, cycle time: 125, GYRO rate: 8000, RX rate: 15, System rate: 9
Voltage: 1 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
FLASH: JEDEC ID=0x00ef4018 16M
Arming disable flags: RXLOSS ANGLE CLI MSP

# flash_info
Flash sectors=256, sectorSize=65536, pagesPerSector=256, pageSize=256, totalSize=16777216 JEDEC ID=0x00ef4018
Partitions:
  0: FLASHFS   0 255
FlashFS size=16777216, usedSize=0

# dump hardware

# version
# Betaflight / STM32F7X2 (S7X2) 4.4.0 Feb  8 2023 / 07:24:52 (4605309d8) MSP API: 1.45

# config: YES

# start the command batch
batch start

board_name TALONF7FUSION
manufacturer_id HENA

# resources
resource BEEPER 1 B04
resource MOTOR 1 B06
resource MOTOR 2 B07
resource MOTOR 3 B08
resource MOTOR 4 C08
resource MOTOR 5 A01
resource MOTOR 6 B09
resource MOTOR 7 C09
resource MOTOR 8 B01
resource LED_STRIP 1 B01
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 C12
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 A01
resource SERIAL_RX 5 D02
resource SERIAL_RX 6 C07
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 NONE
resource SERIAL_RX 12 NONE
resource I2C_SCL 1 NONE
resource I2C_SCL 2 NONE
resource I2C_SCL 3 NONE
resource I2C_SCL 4 NONE
resource I2C_SDA 1 NONE
resource I2C_SDA 2 NONE
resource I2C_SDA 3 NONE
resource I2C_SDA 4 NONE
resource LED 1 B00
resource LED 2 NONE
resource LED 3 NONE
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 3 C10
resource SPI_SCK 4 NONE
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MISO 3 C11
resource SPI_MISO 4 NONE
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 B15
resource SPI_MOSI 3 B05
resource SPI_MOSI 4 NONE
resource ADC_BATT 1 C02
resource ADC_RSSI 1 C03
resource ADC_CURR 1 C01
resource ADC_EXT 1 NONE
resource PINIO 1 A14
resource PINIO 2 NONE
resource PINIO 3 NONE
resource PINIO 4 NONE
resource USB_MSC_PIN 1 NONE
resource FLASH_CS 1 B12
resource OSD_CS 1 A15
resource GYRO_EXTI 1 C04
resource GYRO_EXTI 2 C14
resource GYRO_CS 1 A04
resource GYRO_CS 2 C13
resource USB_DETECT 1 NONE
resource PULLUP 1 NONE
resource PULLUP 2 NONE
resource PULLUP 3 NONE
resource PULLUP 4 NONE
resource PULLDOWN 1 NONE
resource PULLDOWN 2 NONE
resource PULLDOWN 3 NONE
resource PULLDOWN 4 NONE

# timer
timer B03 AF1
# pin B03: TIM2 CH2 (AF1)
timer B06 AF2
# pin B06: TIM4 CH1 (AF2)
timer B07 AF2
# pin B07: TIM4 CH2 (AF2)
timer B08 AF2
# pin B08: TIM4 CH3 (AF2)
timer B09 AF2
# pin B09: TIM4 CH4 (AF2)
timer A01 AF2
# pin A01: TIM5 CH2 (AF2)
timer C08 AF3
# pin C08: TIM8 CH3 (AF3)
timer C09 AF3
# pin C09: TIM8 CH4 (AF3)
timer B01 AF2
# pin B01: TIM3 CH4 (AF2)

# dma
dma SPI_MOSI 1 NONE
dma SPI_MOSI 2 NONE
dma SPI_MOSI 3 NONE
dma SPI_MOSI 4 NONE
dma SPI_MISO 1 NONE
dma SPI_MISO 2 NONE
dma SPI_MISO 3 NONE
dma SPI_MISO 4 NONE
dma SPI_TX 1 NONE
dma SPI_TX 2 NONE
dma SPI_TX 3 NONE
dma SPI_TX 4 NONE
dma SPI_RX 1 NONE
dma SPI_RX 2 NONE
dma SPI_RX 3 NONE
dma SPI_RX 4 NONE
dma ADC 1 1
# ADC 1: DMA2 Stream 4 Channel 0
dma ADC 2 NONE
dma ADC 3 NONE
dma UART_TX 1 NONE
dma UART_TX 2 NONE
dma UART_TX 3 NONE
dma UART_TX 4 NONE
dma UART_TX 5 NONE
dma UART_TX 6 NONE
dma UART_TX 7 NONE
dma UART_TX 8 NONE
dma UART_RX 1 NONE
dma UART_RX 2 NONE
dma UART_RX 3 NONE
dma UART_RX 4 NONE
dma UART_RX 5 NONE
dma UART_RX 6 NONE
dma UART_RX 7 NONE
dma UART_RX 8 NONE
dma pin B03 0
# pin B03: DMA1 Stream 6 Channel 3
dma pin B06 0
# pin B06: DMA1 Stream 0 Channel 2
dma pin B07 0
# pin B07: DMA1 Stream 3 Channel 2
dma pin B08 0
# pin B08: DMA1 Stream 7 Channel 2
dma pin B09 NONE
dma pin A01 0
# pin A01: DMA1 Stream 4 Channel 6
dma pin C08 0
# pin C08: DMA2 Stream 2 Channel 0
dma pin C09 0
# pin C09: DMA2 Stream 7 Channel 7
dma pin B01 0
# pin B01: DMA1 Stream 2 Channel 5

# feature
feature -RX_PPM
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -RANGEFINDER
feature -TELEMETRY
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -OSD
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -RX_SPI
feature -ESC_SENSOR
feature -ANTI_GRAVITY
feature RX_SERIAL
feature TELEMETRY
feature OSD
feature AIRMODE
feature ANTI_GRAVITY

# serial
serial 20 1 115200 57600 0 115200
serial 0 2048 115200 57600 0 115200
serial 1 0 115200 57600 0 115200
serial 2 64 115200 57600 0 115200
serial 3 0 115200 57600 0 115200
serial 4 0 115200 57600 0 115200
serial 5 0 115200 57600 0 115200

# master
set gyro_to_use = BOTH
set adc_device = 1
set blackbox_device = SPIFLASH
set dshot_burst = OFF
set dshot_bitbang = AUTO
set dshot_bitbang_timer = AUTO
set current_meter = ADC
set battery_meter = ADC
set vbat_scale = 160
set ibata_scale = 150
set beeper_inversion = ON
set beeper_od = OFF
set beeper_frequency = 0
set max7456_clock = NOMINAL
set max7456_spi_bus = 3
set max7456_preinit_opu = OFF
set led_inversion = 0
set pinio_config = 1,1,1,1
set pinio_box = 40,255,255,255
set usb_msc_pin_pullup = ON
set flash_spi_bus = 2
set gyro_1_bustype = SPI
set gyro_1_spibus = 1
set gyro_1_i2cBus = 0
set gyro_1_i2c_address = 0
set gyro_1_sensor_align = CW0
set gyro_1_align_roll = 0
set gyro_1_align_pitch = 0
set gyro_1_align_yaw = 0
set gyro_2_bustype = SPI
set gyro_2_spibus = 1
set gyro_2_i2cBus = 0
set gyro_2_i2c_address = 0
set gyro_2_sensor_align = CW90
set gyro_2_align_roll = 0
set gyro_2_align_pitch = 0
set gyro_2_align_yaw = 900
set i2c1_pullup = OFF
set i2c1_clockspeed_khz = 800
set i2c2_pullup = OFF
set i2c2_clockspeed_khz = 800
set i2c3_pullup = OFF
set i2c3_clockspeed_khz = 800
set mco2_on_pc9 = OFF
set scheduler_relax_rx = 25
set scheduler_relax_osd = 25
set box_user_1_name = -
set box_user_2_name = -
set box_user_3_name = -
set box_user_4_name = -

# end the command batch
batch end

#
```

### Flight controller

Talon F7 Fusion

### Other components

This FC is not connected to anything currently — but I have 3 identical FCs already installed in builds and the same behavior applies to all.

### How are the different components wired up (including port information)

N/A — the issue occurs even when the FC is not wired to anything.

### Add any other context about the problem that you think might be relevant here

Possibly similar to other recently reported issues with 4.4.1:

https://github.com/betaflight/betaflight/issues/12635
https://github.com/betaflight/betaflight/issues/12631
https://github.com/betaflight/betaflight/issues/12638
https://github.com/betaflight/betaflight/issues/12598

coolhands21, 2023-04-16:
Great write-up! Seeing identical behavior on the JBF7 (v1) and JBF7_DJI targets. Seems like it is closed via: https://github.com/betaflight/betaflight/issues/12647  #12647

haslinghuis, 2023-04-16:
Please test also with latest 4.5 on latest configurator

haslinghuis, 2023-04-20:
Please check 4.4.0 build with 4.4.1 PRs without config changes

[TALONF7FUSION.zip](https://github.com/betaflight/betaflight/files/11290601/TALONF7FUSION.zip)

```
make STM32F7X2 EXTRA_FLAGS="-D'BUILD_KEY=241bacfbd56241ab131622ff4012a027' -D'RELEASE_NAME=4.4.0' -D'BOARD_NAME=TALONF7FUSION' -D'MANUFACTURER_ID=HENA' -DCLOUD_BUILD -DUSE_ACC -DUSE_ACC_SPI_MPU6500 -DUSE_DSHOT -DUSE_FLASH -DUSE_FLASH_W25Q128FV -DUSE_GPS -DUSE_GPS_PLUS_CODES -DUSE_GYRO -DUSE_GYRO_SPI_MPU6500 -DUSE_LED_STRIP -DUSE_MAX7456 -DUSE_OSD -DUSE_OSD_HD -DUSE_OSD_SD -DUSE_PINIO -DUSE_SERIALRX -DUSE_SERIALRX_CRSF -DUSE_SERIALRX_DEFAULT -DUSE_SERIALRX_GHST -DUSE_SERIALRX_SBUS -DUSE_TELEMETRY -DUSE_TELEMETRY_CRSF -DUSE_TELEMETRY_GHST -DUSE_VTX"
```

Redeque, 2023-04-27:
I stumbled across this post today after I upgraded my Talfon F7 Fusion from 4.4.0 to 4.4.1 and now it does not connect anymore.
Right after the step, apply custom defaults. Windows won't even recognise it at this point so, even when using the BL button. Looks bricked...

Redeque, 2023-04-27:
> I stumbled across this post today after I upgraded my Talfon F7 Fusion from 4.4.0 to 4.4.1 and now it does not connect anymore. Right after the step, apply custom defaults. Windows won't even recognise it at this point so, even when using the BL button. Looks bricked...

I managed to get it going by removing all hardware and putting the bare board in DFU mode with it's BL button.

@haslinghuis  I flashed the hex file you posted and I can confirm that it works.

smartinick, 2023-05-06:
Same here with Target IFLIGHT_F722_TWING.
Flashed from 4.something to 4.4.1, after "apply defaults" there's only a unrecognized USB Device (reboots/repowering FC,... doesn't change anything.
Flashed back to v4.4.0 and it works...

haslinghuis, 2023-05-07:
@smartinick confirm this target fails on 4.4.1. Please use 4.4.0 or 4.5.0 for the moment.

gmanrc, 2023-05-10:
I am experiencing the same thing with the Matek H743 slim v3. 4.3 to and 4.4.0 are fine. But with 4.4.1, the computer no longer detects the FC

haslinghuis, 2023-05-12:
The issue is fixed in 4.4.2 (not released yet) or 4.5.0 (development)