betaflight/betaflight #12647: BF 4.4.1 Breaking SD Card Readers

spatzengr, 2023-04-09:
### Describe the bug

BF 4.4.1 seems to have a glitch again for SD card readers.  See https://youtu.be/9cuVo-T6ddM

Seems like similar behavior to https://github.com/betaflight/betaflight/pull/11839

### To Reproduce

Flash BF 4.4.1 on a FC with an SD Card writer for blackbox.

### Expected behavior

SD Card writer will not work.  I tried an SpeedyBeeF405 v3 and an ObmibsF4SD FC and get the same result.

### Support ID

```txt
*** Support data submitted *** 
Id: "3233792c-a487-4137-a6ce-13949b99cf7f"
```

### Flight controller

Any

### Other components

nope

### How are the different components wired up (including port information)

none

### Add any other context about the problem that you think might be relevant here

none

aowi7280, 2023-04-09:
Can confirm ObmibsF4SD card does not work.

RupesF, 2023-04-14:
During testing I've found that commit aa228ea done last night on the 4.4-maintenance branch has fixed this issue for me on my HolyBro Kakute H7 V1 - SD card is operational again.

haslinghuis, 2023-04-14:
Fixed in #12671

aowi7280, 2023-04-15:
@haslinghuis Yes, it seems to be fixed in the 4.4.2

Squizzy911, 2023-04-30:
please publish 4.4.2 build. milestone seems to be finished. or backport to 4.4.1 and hotfix :)