betaflight/betaflight #12648: CLRACINGF7 No USB Connection after "Apply Custom Defaults" in 4.4.1

c3r63ru5, 2023-04-09:
### Describe the bug

After flashing 4.4.1 the USB connection comes back, but after applying custom defaults USB does not connect any more. Flashing back 4.3.2 in DFU mode also restores "normal" USB functionality.

### To Reproduce

Flash 4.4.1 to a CLRACINGF7 FC and hit "Apply Custom Defaults" after flashing.

### Expected behavior

USB should work after applying custom defaults for the FC.

### Support ID

```txt
This is not gonna work as no CLI is available.
```

### Flight controller

CLRACING F7 V2

gardo403, 2023-04-10:
Same here... flashing back to 4.3 and applying custom defaults brings it back to life.

haslinghuis, 2023-04-10:
4.4.0 or 4.5.0 should work as there seems to be a problem addressing dual gyro's in 4.4.1

Gozzz, 2023-04-13:
Same, also of note I'm unable to get it into bootloader mode to flash back unless I unsolder my Foxeer ELRS lite 2.4 receiver from UART 3. Once I did that I was able to flash back to 4.4.0.

c3r63ru5, 2023-04-13:
> 4.4.0 or 4.5.0 should work as there seems to be a problem addressing dual gyro's in 4.4.1

I can confirm flashing 4.4.0 instead of 4.4.1 does not result in the same behaviour.

gardo403, 2023-04-13:
4.4.0 did work fine... No bootloader mode here. Probably because I'm using non ESP rx

woxulv, 2023-04-15:
I have just spent 1 full day on figuring this out.. Iflight F722 Twing FC - Soft bricked on 4.4.1
Disconnected GPS, 4in1 and DJI Airunit, held boot button and managed to bring up STM32 DFU mode again.
Reflashed with 4.4.0 (after having of course tried multiple times with 4.4.1 with and without custom defaults) and now all works.