betaflight/betaflight #12653: mamba h743 V2

BT203, 2023-04-09:
### Describe the bug

Mamba H743_2022B file missing on the betaflight config.. i had to download directly from mambas site.. but it seems to have issues randomly rebooting when trying to set all my settings on betaflight.. had issues with setting up my o3 air unit canvas mode aswell.

### To Reproduce

log onto betaflight with fc and see random reboots, especially going to CLI.

### Expected behavior

if you can actually type into the CLI it will reboot and you will have to unplug and plug fc back in

### Support ID

```txt
cannont access my cli to paste this
```

### Flight controller

Mamba 4th gen MK4 H743 V2  Target: MAMBA H743_2022B bought from Pyrodrone

### Other components

MAMBA 4-1 60amp esc

### How are the different components wired up (including port information)

used plug from esc to fc. 

### Add any other context about the problem that you think might be relevant here

the quad was flying okay but could not get my on screen display to work. im sure yall are already aware of the issue because the fc target is not available on bf config.

haslinghuis, 2023-04-09:
Please use MAMBAH743

closed 2023-04-09 by haslinghuis

BT203, 2023-04-09:
This file will not open in betaflight config at all. Only works to flash a
new firmware.

haslinghuis, 2023-04-09:
[image]

BT203, 2023-04-10:
Okay yes I understand, but I tried this yesterday, and I was unable to
connect to bf config..  so I had to go to mambas website and download the
h743_2022b and I was able to connect with this.. but having issues like I
said..

BT203, 2023-04-10:
I have both an older mamba h743 and the newer V2 h743. The older board
works perfectly on this firmware. But the new board on that firmware will
not even open in betaflight configurator.. is there something I'm missing
perhaps? Please advise, thank you!!