betaflight/betaflight #12656: SPEEDYBEE F405V3 SD Card doesn't work in 4.4.1 but works fine in 4.3.2

xabc1, 2023-04-10:
### Describe the bug

SPEEDYBEEF405V3 SD Card doesn't work in 4.4.1 but works fine in 4.3.2. Re-flashed the FC with 4.4.1 but still the same.

### To Reproduce

Just flash SPEEDYBEEF405V3 with 4.4.1 and the Onboard SD Card status turned red with Fatal Error.

### Expected behavior

SD Card should work

### Support ID

```txt
0790153e-37ca-45cb-be6f-d1540500ada3
```

### Flight controller

SPEEDYBEEF405V3 bought from Speedybee online store.

### Other components

RX: TBS Crossfire Nano RX (SE)

spatzengr, 2023-04-10:
Same as I'm seeing in #12647

xoga74, 2023-04-19:
Same problem on OMNIBUSF4SD

haslinghuis, 2023-04-19:
Fixed in https://github.com/betaflight/betaflight/pull/12671

closed 2023-04-19 by haslinghuis

haslinghuis, 2023-05-17:
@SpaceMan76 checkout development 4.4.2 pre-release version.

[image]

nicoladelre, 2023-05-30:
with 4.4.2 will it be solved?

haslinghuis, 2023-05-30:
Yes, you should try it out after saving backup