betaflight/betaflight #12658: Skystars H7 dual bmi270

Some-Won-1, 2023-04-10:
### Is your feature request related to a problem? Please describe

Firmware not found.

### Describe the solution you'd like

Requesting a firmware addition in the configurator for Skystars H7 Dual bmi270 FC.

### Describe alternatives you've considered

None

### Other information

It has a different configuration with LED ports at each corner.  I'm not sure if it needs an additional feature.

haslinghuis, 2023-05-12:
SKYSTARSH7HD

Some-Won-1, 2023-05-13:
Still no stable firmware. Fport not working. I need to verify the fport receiver did not get blown. Receiver has the green light though and links with TX. If you need a tester with hardware let me know. I am novas flyer.

haslinghuis, 2023-05-13:
Please provide Support ID (button in CLI) and post in here.

haslinghuis, 2023-05-13:
Duplicate of #12710

closed 2023-05-13 by haslinghuis