betaflight/betaflight #12662: SPRacingH7ZERO target is missing

hydra, 2023-04-10:
### Describe the bug

See title and steps to reproduce.

### To Reproduce

a) Attempt to upgrade an SPRacingH7ZERO from BF 4.3 to BF 4.4 - no firmware is available.

b) Attempt to build firmware for SPRacingH7ZERO from `master` results in:
```
$ make CONFIG=SPRACINGH7ZERO
Makefile:115: *** Config file not found: ./src/config/SPRACINGH7ZERO/config.h.  Stop.
```

### Expected behavior

Working firmware in both cases.

### Support ID

```txt
N/A
```

### Flight controller

SPRacingH7ZERO

### Add any other context about the problem that you think might be relevant here

BF 4.3 had the SPRacingH7EXTREME, H7NANO and H7ZERO which were all closely related targets, unified targets should have been created and published for them all before the legacy targets were deleted from the source code.  There have been recent efforts to re-create the H7EXTREME and H7NANO but the H7ZERO is still missing.

Automated steps to re-create `config.h` files for all unified targets from the unified targets repo obviously do not re-create the H7ZERO (or H7EXTREME and H7NANO) because unified targets were never created in the first place.

DanO83, 2023-05-01:
is there a workaround available to be able to flash the H7EXTREME target?

haslinghuis, 2023-05-01:
H7NANO and H7EXTREME should work (only tested 4.5.0)

DanO83, 2023-05-01:
in bf configurator 10.9.0 the newest version I can choose is 4.3.2 for the H7EXTREME, if i'm choosing H7NANO no version is showing.

when i change to the development branche, version 4.5.0 is available, but no 4.4

haslinghuis, 2023-06-28:
Fixed in #12575

closed 2023-06-28 by haslinghuis