betaflight/betaflight #12665: FC dropping into bootloader randomly

smitty97, 2023-04-10:
### Describe the bug

I have 3 new builds for mayhem, using the Foxeer F722 V4, installed BF 4.4.1

When i plug into configurator, i can connect and read data but then it will switch to bootloader.  sometimes it happens immediately, sometimes it takes 30 seconds.  there is nothing pressing on the button.

I believe it is also happening on battery.  I can plug in and arm, but other times i cannot.  stick commands wont work at that point either, to try to see the menu.

4.4.0 still had the problem.  4.3.2 seems to be ok, but missing many of the hdz setup things i need.
this is happening on 3 identical quads.

### To Reproduce

get foxeer F722 V4, install 4.4.1
plug into bf configurator.  at a random time it will drop into bootloader.

### Expected behavior

it should not drop into bootloader

### Support ID

```txt
with 4.3.2 installed: Id: "2fadb6d7-795a-42fd-b25e-4be35d156460"

with 4.4.1 installed: Id: "14beb588-ce7e-4608-8ada-0728830f83de"
```

### Flight controller

Foxeer F722 V4 from pyro

### Other components

foxeer F4 6A ESC.
IRC ghost atto on uart1
2-wire msp setup for hdzero on uart2
hdzero race v2 vtx powered thru matek micro 6-30v bec using VCC pad of FC as source.

### How are the different components wired up (including port information)

[image]

michael-blue2, 2023-04-12:
I just had this happen while flying. The quad just dropped out of the sky without warning. It was some tall bushes and it took me 30-40min to find it as i couldnt beep or do anything on the quad. My vista just had battery voltage and nothing else. the voltage of the quad was frozen as well. As soon as I removed the battery and replugged everything was fine. but no chance i am trusting this quad again

haslinghuis, 2023-05-13:
Please check:

[image]

SteveCEvans, 2023-05-13:
This should have been fixed by https://github.com/betaflight/betaflight/pull/12760 which has now been merged into master.

closed 2023-06-21 by github-actions[bot]