betaflight/betaflight #12676: VTx low power during disarm not working with VTx switch

truglodite, 2023-04-15:
### Describe the bug

By default low power during disarm works, but however when I configure VTx power switching via transmitter it is no longer honored. Instead, regardless if it is disarmed, the vtx uses the power level based on the VTx power channel switch.

### To Reproduce

Enable "Low power on disarm", and configure a VTX switch in CLI, for example on my Unify Pro32 nano I use:

```
vtx 0 6 0 0 2 900 1600
vtx 1 6 0 0 3 1601 1800
vtx 2 6 0 0 4 1801 2100
```
When the quad is powered on and disarmed, aux7 is at 1500, and VTx stays at 100mW, instead of 25mW.

### Expected behavior

Low power (ie 1, or 25mW) after plugging in pack while disarmed.

### Support ID

```txt
version
# Betaflight / STM32F411 (S411) 4.4.0 Jan 30 2023 / 03:38:33 (4605309d8) MSP API: 1.45

# config: YES
# board: manufacturer_id: BEFH, board_name: BETAFPVF411

# status
MCU F411 Clock=108MHz (PLLP-HSE), Vref=3.30V, Core temp=29degC
Stack size: 2048, Stack address: 0x2001fff0
Configuration: CONFIGURED, size: 3528, max available: 16384
Devices detected: SPI:1, I2C:0
Gyros detected: gyro 1 locked dma
GYRO=BMI270
OSD: MAX7456 (30 x 13)
BUILD KEY: cd6c80a7113a6fea64285a084a746da3 (4.4.0)
System Uptime: 25 seconds, Current Time: 2023-04-15T14:49:46.164+00:00
CPU:38%, cycle time: 309, GYRO rate: 3236, RX rate: 15, System rate: 9
Voltage: 211 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
FLASH: JEDEC ID=0x00ef4017 8M
Arming disable flags: RXLOSS CLI MSP RPMFILTER

# flash_info
Flash sectors=128, sectorSize=65536, pagesPerSector=256, pageSize=256, totalSize=8388608 JEDEC ID=0x00ef4017
Partitions:
  0: FLASHFS   0 127
FlashFS size=8388608, usedSize=8388608

# dump hardware

# version
# Betaflight / STM32F411 (S411) 4.4.0 Jan 30 2023 / 03:38:33 (4605309d8) MSP API: 1.45

# config: YES

# start the command batch
batch start

board_name BETAFPVF411
manufacturer_id BEFH

# resources
resource BEEPER 1 B02
resource MOTOR 1 B04
resource MOTOR 2 B05
resource MOTOR 3 B06
resource MOTOR 4 B07
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource LED_STRIP 1 A08
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 NONE
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 NONE
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 B03
resource SERIAL_TX 12 B10
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 NONE
resource SERIAL_RX 4 NONE
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 NONE
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 B03
resource SERIAL_RX 12 B10
resource INVERTER 1 NONE
resource INVERTER 2 NONE
resource INVERTER 3 NONE
resource INVERTER 4 NONE
resource INVERTER 5 NONE
resource INVERTER 6 NONE
resource INVERTER 7 NONE
resource INVERTER 8 NONE
resource INVERTER 9 NONE
resource INVERTER 10 NONE
resource INVERTER 11 NONE
resource INVERTER 12 NONE
resource I2C_SCL 1 B08
resource I2C_SCL 2 NONE
resource I2C_SCL 3 NONE
resource I2C_SDA 1 B09
resource I2C_SDA 2 NONE
resource I2C_SDA 3 NONE
resource LED 1 C13
resource LED 2 C14
resource LED 3 NONE
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 3 NONE
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MISO 3 NONE
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 B15
resource SPI_MOSI 3 NONE
resource ESCSERIAL 1 NONE
resource ADC_BATT 1 B00
resource ADC_RSSI 1 NONE
resource ADC_CURR 1 B01
resource ADC_EXT 1 NONE
resource PINIO 1 NONE
resource PINIO 2 NONE
resource PINIO 3 NONE
resource PINIO 4 NONE
resource USB_MSC_PIN 1 NONE
resource FLASH_CS 1 A00
resource OSD_CS 1 B12
resource GYRO_EXTI 1 A01
resource GYRO_EXTI 2 NONE
resource GYRO_CS 1 A04
resource GYRO_CS 2 NONE
resource USB_DETECT 1 C15
resource PULLUP 1 NONE
resource PULLUP 2 NONE
resource PULLUP 3 NONE
resource PULLUP 4 NONE
resource PULLDOWN 1 NONE
resource PULLDOWN 2 NONE
resource PULLDOWN 3 NONE
resource PULLDOWN 4 NONE

# timer
timer A03 AF3
# pin A03: TIM9 CH2 (AF3)
timer B04 AF2
# pin B04: TIM3 CH1 (AF2)
timer B05 AF2
# pin B05: TIM3 CH2 (AF2)
timer B06 AF2
# pin B06: TIM4 CH1 (AF2)
timer B07 AF2
# pin B07: TIM4 CH2 (AF2)
timer B03 AF1
# pin B03: TIM2 CH2 (AF1)
timer B10 AF1
# pin B10: TIM2 CH3 (AF1)
timer A00 AF2
# pin A00: TIM5 CH1 (AF2)
timer A02 AF2
# pin A02: TIM5 CH3 (AF2)
timer A08 AF1
# pin A08: TIM1 CH1 (AF1)

# dma
dma SPI_MOSI 1 NONE
dma SPI_MOSI 2 NONE
dma SPI_MOSI 3 NONE
dma SPI_MISO 1 NONE
dma SPI_MISO 2 NONE
dma SPI_MISO 3 NONE
dma SPI_TX 1 NONE
dma SPI_TX 2 NONE
dma SPI_TX 3 NONE
dma SPI_RX 1 NONE
dma SPI_RX 2 NONE
dma SPI_RX 3 NONE
dma ADC 1 1
# ADC 1: DMA2 Stream 4 Channel 0
dma ADC 2 NONE
dma ADC 3 NONE
dma UART_TX 1 NONE
dma UART_TX 2 NONE
dma UART_TX 3 NONE
dma UART_TX 4 NONE
dma UART_TX 5 NONE
dma UART_TX 6 NONE
dma UART_TX 7 NONE
dma UART_TX 8 NONE
dma UART_RX 1 NONE
dma UART_RX 2 NONE
dma UART_RX 3 NONE
dma UART_RX 4 NONE
dma UART_RX 5 NONE
dma UART_RX 6 NONE
dma UART_RX 7 NONE
dma UART_RX 8 NONE
dma pin A03 NONE
dma pin B04 0
# pin B04: DMA1 Stream 4 Channel 5
dma pin B05 0
# pin B05: DMA1 Stream 5 Channel 5
dma pin B06 0
# pin B06: DMA1 Stream 0 Channel 2
dma pin B07 0
# pin B07: DMA1 Stream 3 Channel 2
dma pin B03 0
# pin B03: DMA1 Stream 6 Channel 3
dma pin B10 0
# pin B10: DMA1 Stream 1 Channel 3
dma pin A00 0
# pin A00: DMA1 Stream 2 Channel 6
dma pin A02 0
# pin A02: DMA1 Stream 0 Channel 6
dma pin A08 0
# pin A08: DMA2 Stream 6 Channel 0

# feature
feature -RX_PPM
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -RANGEFINDER
feature -TELEMETRY
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -OSD
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -RX_SPI
feature -ESC_SENSOR
feature -ANTI_GRAVITY
feature RX_SERIAL
feature TELEMETRY
feature OSD
feature AIRMODE
feature ANTI_GRAVITY

# serial
serial 20 1 115200 57600 0 115200
serial 0 64 115200 57600 0 115200
serial 1 2048 115200 57600 0 115200

# master
set gyro_to_use = FIRST
set adc_device = 1
set blackbox_device = SPIFLASH
set dshot_burst = AUTO
set dshot_bitbang = OFF
set dshot_bitbang_timer = AUTO
set current_meter = NONE
set battery_meter = ADC
set vbat_scale = 110
set ibata_scale = 400
set beeper_inversion = ON
set beeper_od = OFF
set beeper_frequency = 0
set system_hse_mhz = 8
set max7456_clock = NOMINAL
set max7456_spi_bus = 2
set max7456_preinit_opu = OFF
set led_inversion = 0
set pinio_config = 1,1,1,1
set pinio_box = 255,255,255,255
set usb_msc_pin_pullup = ON
set flash_spi_bus = 2
set gyro_1_bustype = SPI
set gyro_1_spibus = 1
set gyro_1_i2cBus = 0
set gyro_1_i2c_address = 0
set gyro_1_sensor_align = CW90
set gyro_1_align_roll = 0
set gyro_1_align_pitch = 0
set gyro_1_align_yaw = 900
set gyro_2_bustype = SPI
set gyro_2_spibus = 0
set gyro_2_i2cBus = 0
set gyro_2_i2c_address = 0
set gyro_2_sensor_align = CW0
set gyro_2_align_roll = 0
set gyro_2_align_pitch = 0
set gyro_2_align_yaw = 0
set i2c1_pullup = OFF
set i2c1_clockspeed_khz = 800
set i2c2_pullup = OFF
set i2c2_clockspeed_khz = 800
set i2c3_pullup = OFF
set i2c3_clockspeed_khz = 800
set mco2_on_pc9 = OFF
set scheduler_relax_rx = 1
set scheduler_relax_osd = 1
set box_user_1_name = -
set box_user_2_name = -
set box_user_3_name = -
set box_user_4_name = -

# end the command batch
batch end

#
```

### Flight controller

Betafpv 1s 5A aio serial elrs

### Other components

TBS Unify Pro32 nano, otherwise stock.

### How are the different components wired up (including port information)

NA

### Add any other context about the problem that you think might be relevant here

The same issue is present on all of my kwads. This includes a Meteor 75, Meteor 65pro, and a 7" with a Matek F722se. Otherwise smartaudio is working as expected, and low power on disarm also works if I do not configure a VTx switch. It appears to me that Betaflight doesn't honor the low power on disarm if it VTx control by receiver is configured.

truglodite, 2023-05-06:
I will implement a workaround using another tx switch and some mixing, together with a 4th vtx control entry for 25mW. However It seems the setting should be honored, rather than ignored.

closed 2023-06-12 by github-actions[bot]

goldeneyeonline, 2023-08-11:
I'm experiencing the same behavior on my whoopfly16 analog version ...