betaflight/betaflight #12683: BTFL defaults cannot be applied on T-Motor F7 FC

saimat, 2023-04-16:
### Describe the bug

Applying the BTFL 4.4.1 defaults on T-Motor F7 FC fails because of barometer ressource.

```
# defaults nosave

# resetting to defaults
###ERROR IN resource: INVALID RESOURCE NAME: 'BARO_CS'###
###ERROR IN set: INVALID NAME: baro_spi_device = 3###
```

To my knowledge my T-Motor F7 FC does not have an built in barometer.

### To Reproduce

Load defaults in CLI or load a Backup File from Presets Tab

### Expected behavior

Defaults applied without error

### Support ID

```txt
e2f005d8-36fb-492c-aced-67ebd98e88ba
```

### Flight controller

T-Motor F7

### Other components

RX: Crossfire Nano
VTX: TBS Unify Pro32 HV MMCX
CAM: Runcam Phoenix 2
ESC: T-Motor F55A Pro II
Motors: T-Motor F40 Pro IV

### How are the different components wired up (including port information)

RX: UART1
CAM: UART2
VTX: UART3

haslinghuis, 2023-04-16:
Seems there is BARO defined on the target:

https://github.com/betaflight/unified-targets/blob/master/configs/default/TMTR-TMOTORF7.config

saimat, 2023-04-17:
Yes, but only the F7 PRO FC comes with a baro, the normal F7 FC does not have a baro.

saimat, 2023-04-30:
Hi any news here when will the target be fixed?

haslinghuis, 2023-04-30:
Please try 4.5.0 build

saimat, 2023-05-18:
Hi, with 4.5.0 it works as expected, no more errors.
Will it also be fixed in 4.4 as 4.5 is still in development?

haslinghuis, 2023-05-18:
We are working on it: https://github.com/betaflight/betaflight/pull/12814#issuecomment-1552239476

Until merged you can add `BARO FLASH` to Custom Defines

[image]

haslinghuis, 2023-06-01:
Closing as fixed in 4.4.2 release.

closed 2023-06-01 by haslinghuis