betaflight/betaflight #12686: JHEMCU F4 NOXE v3. BMI270 - work. ICM42688P - not work.

IvanBiv, 2023-04-16:
### Describe the bug

4.4.1 on the board with BMI270 it works, but on the board with ICM42688p it does not work. It just does not detect the accelerometer and gyroscope.
But ICM42688p sensors work fine on 4.3.1.

### To Reproduce

jhemcu f4 noxe v3 with icm42688p
betaflight 4.4.1
And sensors is not detected.

### Expected behavior

jhemcu f4 noxe v3 with icm42688p
betaflight 4.4.1
I want sensors acc and gyro is detected.

### Support ID

```txt
3e734242-bd2f-4e64-bc51-1d8548f90704
```

### Flight controller

jhemcu f4 noxe v3 delux icm 230309

haslinghuis, 2023-04-16:
Does it work on 4.5 ?
Defines already there.

IvanBiv, 2023-04-16:
> Does it work on 4.5 ? Defines already there.

Realy!
On 4.5.0 it work!
It is cool, i will wait a stable version.

Thx!

closed 2023-05-17 by haslinghuis