betaflight/betaflight #12689: Manual switching of the rescue mode on the fly (in flight from the remote control).

VMaximI, 2023-04-17:
### Is your feature request related to a problem? Please describe

Create the ability to switch the rescue mode directly in flight using the switch on the remote control.

### Describe the solution you'd like

Before the flight, we are forced to take into account the surrounding conditions in advance (the presence of trees, hills, water). And already based on these conditions, through the BF menu, set the desired behavior of the quadrocopter at FS.
But there are situations when it is impossible to take into account the flight conditions in advance, for example, when they change during the flight itself.
I'll give you an example:
I took off from the point and fly under the trees, then I fly ashore and fly over the water.
Obviously, in the case of FS, you cannot use "GPS Rescue", because. the quadcopter will get stuck in the trees and the best option here is "Drop". But having taken off into the water, the "Drop" mode will lead to a guaranteed loss of the quadcopter, and with FS, "GPS Rescue" will work fine here.
Having the ability to switch the rescue mode, flying out into the open space, I myself will switch the toggle switch from "Drop" to "GPS Rescue" and in the case of FS I will not lose the quadcopter.

How can this be implemented for the user:
In BF, in the "Modes" section, the user binds the channel to the rescue mode ("Drop"/"GPS Rescue").

### Describe alternatives you've considered

no options

### Other information

This will save many pilots the hassle of changing rescue mode while traveling. It will also make flying among different terrain options safer.

jonmahoney15, 2023-05-29:
I think this will need an OSD element as well saying which mode similar to the flight modes. I will give this feature request a shot. 

I guess it shouldn't allow a GPS rescue mode if there isn't a good sat lock, so it'll also need to display a warning when the user flips to that mode that it wasn't activated?

theGISer, 2026-01-04:
Anything new on this feature request? The ability to change the behaviour of GPS Rescue during flight would be helpful especially for long-range pilots.

Here's my real-world example:
After diving a 2000m peak and flying back home, I lost video and activated GPS rescue. Still didn't regain video. I knew the copter didn't have enough battery to climb to 'Max Altitude', and then head home. So without video, I disengaged GPS Rescue, activated Horizon mode, and lowered throttle to just above my known hovering value and hoped for the best. After a couple of minutes I disarmed. 
From my Tango 2 telemetry, we plotted out the last known GPS coordinates on Google Maps (offline), hiked for two hours in dense forest, and eventually found the quad with the help of the onboard buzzer. Minor damage to the quad. 

The above incident clearly highlights the need for a second GPS Rescue option which could be activated based on pilot's judgement.

I created a YT Short covering this incident:
https://youtube.com/shorts/SmsImZpNEXw?si=3Ke6TU92gclEBsJ7

theGISer, 2026-01-05:
@Quick-Flash 
Yes having all three "_GPS Rescue (max altitude in flight), GPS Rescue (current altitude), and Land in Place_" options available in flight to the pilot would be great. 
Don't you think the "_Drop_" option is similar to disarming so no real need to worry about including it?

As an example, all three GPS options could be mapped on a three way switch, and the pilot could set up audio call-outs on their radio for each switch position when triggered. With a bit of practice and muscle memory, it should be clear which GPS rescue option they've trigged even with no video.

thanks for your time and effort, much appreciated.

VMaximI, 2026-03-02:
The configurator is no longer offline, and implementing this option is more important than ever. I can no longer reconfigure rescue mode from my laptop without internet access.