betaflight/betaflight #12691: No gyro JHEMCU GF16-BMI

dwhacks, 2023-04-18:
### Describe the bug

Flashing 4.4 no gyro found

### To Reproduce

flash target  JHEF/JHEF405(STM32F405), version: 0
no gyro found, adding config.txt from manufacturer, still no gyro found
Flash "Core only" and apply custom defaults, still no gyro found until adding config.txt from manufacturer 

[GF16-F405-BMI.txt](https://github.com/betaflight/betaflight/files/11257194/GF16-F405-BMI.txt)

### Expected behavior

Expected it to find the gyro, I can't be the only one with this board.

### Support ID

```txt
efe28918-8e3e-49d8-ab00-c11f711d96bb
```

### Flight controller

JHEMCU GF16-BMI Stack F405-BMI Flight Controller BMI270 W/OSD AT7456E BLHELI_S 2-4S 13A 4in1 ESC Dshot600, from aliexpress

dwhacks, 2023-04-24:
Here is a support data with 4.3.2 with everything working:

95f25270-fdc3-4171-a736-4fc74d47541a

josemaria-vilaplana, 2023-05-01:
@dwhacks Were can I download the working 4.3.2 cli?
Thx

dwhacks, 2023-05-06:
> @dwhacks Were can I download the working 4.3.2 cli? Thx

I beleive it works with the [GF16-F405-BMI.txt](https://github.com/betaflight/betaflight/files/11257194/GF16-F405-BMI.txt) I have linked earlier in the thread.

haslinghuis, 2023-05-06:
Try:

[image]

dwhacks, 2023-05-07:
> Try:
> 
> [image]

on 4.4.1 this still says "no gyro found" after applying custom defaults. 
After applying my diff all from a previous build it works
[BTFL_cli_20230506_172706.txt](https://github.com/betaflight/betaflight/files/11413633/BTFL_cli_20230506_172706.txt)

dwhacks, 2023-05-07:
also no gyro found on 4.4.0 unless I use that diff all (which i believe is from the stock firmware that came on the board)

Should I not be clicking "apply custom defaults"?

haslinghuis, 2023-05-07:
Please try PRO version with the custom defines?

dwhacks, 2023-05-07:
will not build with custom defines, here's the log:

https://build.betaflight.com/api/builds/6d829cac9bf438c23e09044f3cb2cc3e/log

dwhacks, 2023-05-07:
JHEF405PRO without any custom defines seems to work. Only need to change the board alignment to CW90.

closed 2023-05-07 by haslinghuis

vedmant, 2023-10-03:
I have the same issue, I have "JHEMCU GF16-BMI Stack F405-BMI Flight Controller BMI270 W/OSD AT7456E BLHELI_S 2-4S 13A 4in1 ESC", after flashing new version 4.4.2 there is no gyro and accelerometer. Tried to apply custom defaults and no, result is the same. Loading saved preset not helping as well. Barometer works however.

JPMat78, 2024-04-14:
> I have the same issue, I have "JHEMCU GF16-BMI Stack F405-BMI Flight Controller BMI270 W/OSD AT7456E BLHELI_S 2-4S 13A 4in1 ESC", after flashing new version 4.4.2 there is no gyro and accelerometer. Tried to apply custom defaults and no, result is the same. Loading saved preset not helping as well. Barometer works however.
> 
> Tried to flash JHEF405PRO, hovewer I see very slow response on gyro, when I turn controller position in the betaflight is changing with really large delay.
> 
> Update: actually it starts to work normally after about 10-20 seconds of connection to betaflight. Yeah, each time after reboot it works slow for like 10-30 sec

Same for mre