betaflight/betaflight #12693: VGOODRCF4 no Blackbox on BF 4.4

Williwolle, 2023-04-18:
### Describe the bug

After updating to Betaflight 4.4 under the Blackbox Tab it says that the Flightcontroller doesnt support Blackboxlogging.
(I wasnt able to select the "flash" option under build configuration)

### To Reproduce

Update to Betaflight 4.4. or 4.4.1.

### Expected behavior

Having 16MB of onboard Blackboxstorage like in Betaflight 4.3.

### Support ID

```txt
976f231a-3fe2-411c-8d0a-f831cf4fd067
```

### Flight controller

VGOODRCF4

AndKe, 2023-04-22:
same for GEPRCF411  (tested BF4.4.1 )@haslinghuis   - is this a easy fix for you?

haslinghuis, 2023-04-22:
@AndKe please open a new issue.

http://betaflight.com/docs/wiki/configurator/firmware-flasher-tab#board-defines

AndKe, 2023-04-22:
@haslinghuis created https://github.com/betaflight/betaflight/issues/12711

closed 2023-04-28 by sugaarK