betaflight/betaflight #12703: Matek H743 after update 4.4.1 com port not visible

czislo, 2023-04-20:
### Describe the bug

Matek H743 after update 4.4.1 com port not visible
Tested on MATEK H743 WING and Matek H743 mini

### To Reproduce

Firmware 4.4.0 and 4.5.0 zulu
works normally

### Expected behavior

-

### Support ID

```txt
-
```

### Flight controller

MATEK H743 WING

kevinsumner, 2023-04-20:
I'm looking at this on a couple of my Matek H743 Mini FCs today. Loading the firmware from #12671 doesn't prompt to apply custom defaults, then applying a `diff all` from 4.4.0 causes it to stop responding on USB. I'm trying a `dump all`, but I have some errors about `dma` commands I've asked about over on Discord before I put this in the air:

https://discord.com/channels/868013470023548938/924169080045445120/1098675195633487872

Meanwhile, applying the `dump all` from 4.4.0 to the build from #12671 and accepting the errors from the `dma` commands at least gets the config applied and brings the FC back up on USB properly after it reboots from the `save`.

kevinsumner, 2023-04-20:
Original issue: #12703

haslinghuis, 2023-04-20:
Duplicate of #12703

czislo, 2023-04-20:
> Komu: betaflight/betaflight ***@***.***>
> not visible (Issue #12703)
> "
>
> Please test:
>
> MATEKH743.zip
> (https://github.com/betaflight/betaflight/files/11289637/MATEKH743.zip)
>
> Note: this build is 4.4.0 with all PRs in 4.4.1 except the configuration PRs
>
> (https://github.com/betaflight/betaflight/issues/12703#issuecomment-1516819267)
> "

Hi,

 thank you for your attention to this issue.

I tried the firmware from the attachment and another one in version 4.4.2,

In both cases, the unit will be loaded, but without parameters for the 
board.

Can't see, gyroscope, barometer, etc--

in attachment

---------- Původní e-mail ----------

czislo, 2023-04-20:
#
[image]
[image]
[image]
[image]

haslinghuis, 2023-04-20:
Works for me :upside_down_face: 

[image]

czislo, 2023-04-20:
I don't know why, but it worked after about the 4th update
[image]

haslinghuis, 2023-04-20:
Yes with a non working firmware it might be better to use the boot button to force the FC into DFU mode bypassing normal serial connection needed to force to board into DFU using software methods.

kevinsumner, 2023-04-20:
> @kevinsumner the issue you mention is this issue 🙃

oops! #12647

haslinghuis, 2023-04-20:
Duplicate of #12647

closed 2023-04-20 by haslinghuis

kevinsumner, 2023-04-20:
I confirmed #12671 fixed all issues on at least one of my Matek H743 Mini FCs.

JohnDGodwin, 2023-05-03:
I'm running into this issue on my Matek H743-mini. I can flash 4.4.0 and it will prompt to apply custom defaults, but 4.4.1 currently will not.